iceman1001/proxmark3 #5: Cannot compile nonce2key/crypto1_bs.c

pmorange, 2016-04-22:
Hi,
In my ProxSpace, I totally replaced the pm3 folder with your sources. I then executed the "make clean && make all" command.

Here is the error I encountered :
```
gcc -std=c99 -O3 -mpopcnt -march=native -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall -g -/mingw/include -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/nonce2key/crypto1_bs.o nonce2key/crypto1_bs.c
In file included from nonce2key/crypto1_bs.c:25:
nonce2key/crypto1_bs.h:25: error: alignment of array elements is greater than element size
make[1]: *** [obj/nonce2key/crypto1_bs.o] Error 1
make[1]: Leaving directory `/pm3/client'
make: *** [client/all] Error 2
```

I tried removing the "-march=native"  in the client\makefile as stated in http://www.proxmark.org/forum/viewtopic.php?pid=21093#p21093 and still got the same error.
Also tried to replace it with:  "-mmmx"  or "-msse", but got the same error every time...

I'm compiling on Windows 7 64 bits, Mingw32. 
The main branch compiles well.
If I can give you more information tell me what you need.

iceman1001, 2016-04-22:
thanks for reporting it,   this is indeed strange since I myself is compiling it on Win7 64b / mingw32.

Whats the gcc --version output?

pmorange, 2016-04-22:
It gives me :
gcc.exe (GCC) 4.4.0

iceman1001, 2016-04-22:
I have updated mine to GCC 4.9.3.   
On the default installation, ie proxspace rar, I've have confirmed your problem.

pmorange, 2016-04-22:
Ok it's good you have the same behavior :-)
Shall I update ? Or do you think you can look into it ?
Thanks

iceman1001, 2016-04-22:
The mingw distro included in the proxspace file, is quite old. 
I installed mingw on my computer and can update it accordingly,
I also have a Ubuntu14.04 virtual machine running,  where its easy to clone the github fork and compile.

I don't think I will fiddle around with the mingw-distro provided inside the proxspace package.

pmorange, 2016-04-22:
I'll try and update mingw then, and hope everything will work without having to err too much. 
Thanks for your help.

greatone76, 2016-04-22:
I've got the same error can I get some help on getting a new mingw folder for the proxspace or what exactly I need to due to update the files and or programs.  Sorry for the basic questions.  I tried getting a different mingw folder from gethub and got the same error.

iceman1001, 2016-04-23:
When I fiddled with the mingw enviroment, it was a mess to do.  An easy alternativ is to use ubuntu virtual machine,  just download a ready vmimage (or type of choice for virtual tech)  and clone the fork to it.  A full linux distro is much easier to maintain and there is lots of guides on the net about  it.

greatone76, 2016-04-23:
Being that I have never touched linux, I really prefer to fix the mingw.  I've gone to the website and downloaded the loader ect.  I feel like I'm missing a step in getting pm3 to talk to the new version and not the old version.  

On the flip side any recommended tutorial on setting up a linux virtual machine and cloning from github would be greatly appreciated.  I'm getting the impression if I use linux I would only need your branch pm3 folder and the rest of the stuff I have in windows is just the compiler and such in the ProxSpace Folder?

iceman1001, 2016-04-23:
If you go with a virtual linux, like ubuntu, then you would follow the wiki directions here to get it at a pair with what you got in mingw.   the upside is plenty more ppl uses linux then mingw. 
However, I understand yor concern.   Your new version is located in c:\mingw  and you need to change the "runme.bat" file like this:

rem echo %MYPATH%mingw		/mingw >>msys/etc/fstab
echo c:\mingw			/mingw >>msys/etc/fstab

greatone76, 2016-04-26:
I think I've got my mingw working.  Any help with missing mingwm10.dll and libgcc_s_dw2-1.dll?  I get errors that they are missing and I close the pop up boxes and it continues to compile.  I compiles gives me a second set of error that are the same about missing the above files and then continues to an error that says:

 proxmakr3.c:17:31 fatal error: readline/readline.h: No such file or directory
#include <readline/readline.h>

Again I'm very appreciative of your help so far.

iceman1001, 2016-04-26:
Sorry, but I don't see the error messages you are refering to. If you would post them, it would help out. thanks. 

Use your google-foo to find a guide on how to install libreadline in a mingw environment.

Another tip for ppl using mingw:    install  mingw-get
That is a good tool to have.

iceman1001, 2016-04-28:
As a fast solution for your compiling problems and not having a linux distro installed I have made a docker image,  on  https://hub.docker.com/r/iceman1001/proxmark3/

::Instructions::
1) follow guide here, https://docs.docker.com/windows/step_one/
2) run the "docker quickstart terminal"
3) download and run image with
     docker run -it iceman1001/proxmark3:1.6.0

if all went well, you will have a small prompt saying:
:/opt#

Now you are inside a linux/ubuntu 14.04 bash shell.  
Where you do the same stuff as before.  
but its not COM3 anymore,  its  /dev/ttyACM0  

Let me know how this works for you.

iceman1001, 2016-04-28:
and it was not that simple as I tought. We need to add the proxmark3 device some how.

pmorange, 2016-04-28:
Hi,
Installed Docker and got your image.
Now I can compile, but I haven't tried plugging the PM3 to the computer yet. I guess it'll be recognized in Windows and not in the Docker env, right ? In VMWare and/or VirtualBox, I remember there was an option to get the peripherals' access and view, removing it from Windows while it was used in the Virtual env. I guess that's what you are referring to...

Other questions with the Docker env : 
 - It should be possible to update the PM3's firmware from the Docker env, right ?... I used to do it from Windows and will have to check how to do it from the Docker env. Or maybe I could get the compiled files, put them in Windows and update firmware as I did before, no ? This way I would only run the client executable from the Docker env...
 - If you update your repository, how can we update the sources too in the Docker env ?

iceman1001, 2016-04-28:
The docker image is a complete clone of my fork with git installed.
Just run "git pull  && make clean && make all"  to get the latest stuff.

If the docker container can get access to the pm3 device, then you should be able to both flash och run it from within.   Easy.

pmorange, 2016-04-28:
Excellent, I wanted to say "You're the man, Iceman" :-)

iceman1001, 2016-04-28:
but remember it is a docker container,   where it goes back to its prestine state when shutting down. If I understand it correct.

About usb-devices.  You were right,  I needed to download a virtualbox extension pack and install.
Now the docker host (not client)  has access to the USB devices.   one step forward.

ref: Oracle_VM_VirtualBox_Extension_Pack-5.0.18-106667.vbox-extpack

iceman1001, 2016-04-28:
Ok, it works.  I've updated my docker description.

https://hub.docker.com/r/iceman1001/proxmark3/

pmorange, 2016-04-28:
Great work.
So if I understand correctly, we shall start it from the VirtualBox manager instead of from a Quickstart Terminal, executing the command you specified in the description... 

I shut down and restarted the container from the same running prompt, and indeed all files disappeared... Wonder if there is a way to keep files between sessions or if this is against the purpose of containers...

iceman1001, 2016-04-28:
Once you enabled the USB ports and restarted the host then you do as normal from the Quickstart Terminal,   and with the new command.

greatone76, 2016-05-01:
Alright I've got a computer that is 64 bit and can allow virtualization.  I've got docker up and running.  I've got USB set up.  (I picked 2.0 should have I just done 1.1 or 3.0?)  I have it getting the iceman fork when I just use ""docker run -it iceman1001/proxmark3:1.6.0""  It download fine.  When I try to go cd proxmark3 then make it starts but appears to stop in the middle.  

I'm thinking part of my issue is setting up the driver and setting up the correct firmware on the Proxmark.  I had the base Proxmark firmware on the Proxmark at this point.  I don't believe I have any driver on this computer at all.  Do I need to get the drivers for windows or Linux or both?  How do I get the Linux driver if I need it?  Where do I get the correct firmware for the iceman fork and how do I load it through Linux?  

Again I thank you for your advice and guidance in advanced and I am grateful for all the previous help.

iceman1001, 2016-05-01:
Don't use that command to run the docker container. 
use:
docker run -it --device="/dev/ttyACM0" iceman1001/proxmark3:1.6.0

So you get access to your pm3 device.

iceman1001, 2016-05-01:
You need to install your PM3 device for you installed OS.  It has to be detected and identified correct first.  Then you start the docker container,  where you should have access to it with "/dev/ttyACM0"

greatone76, 2016-05-02:
I've got the windows drive installed and I'm on COM3.  This is the error I get:  (I get the same thing with ttyACM0 as well, but my understanding is that the 0 is for the com port)

$ docker run -it --device="/dev/ttyACM3" iceman1001/proxmark3:1.6.0
C:\Program Files\Docker Toolbox\docker.exe: Error response from daemon: linux runtime spec devices: error gathering device information while adding custom device "/dev/ttyACM3": lstat /dev/ttyACM3: no such file or directory.

iceman1001, 2016-05-02:
Almost there,  the dedicated com port in linux is different than for windows.
COM3 can be ttyACM0..   The wiki , subcategory linux,  will give you a lot of information.

run:   dmesg | tail  

to find which tty got assigned for your pm3 device.   Use that one.  Normally its ttyACM0

greatone76, 2016-05-03:
I got the usb device to link.  I had to use the virtual box and load something, but that is done.  I got the 

$ docker run -it --device="/dev/ttyACM3" iceman1001/proxmark3:1.6.0

to run without error.  
then I do cd Proxmark3 and then make and it starts the compile and just stops after the following lines.  Just hangs with a flashing cursor.  

cipher.o obj/hfsnoop.o obj/parity.o -lgcc
arm-none-eabi-objcopy -O binary -I elf32-littlearm --remove-section .data obj/fullimage.stage1.elf obj/fullimage.nodata.bin
arm-none-eabi-objcopy -O elf32-littlearm -I binary -B arm --rename-section .data=stage1_image obj/fullimage.nodata.bin obj/fullimage.nodata.o
arm-none-eabi-objcopy -O binary -I elf32-littlearm --only-section .data obj/fullimage.stage1.elf obj/fullimage.data.bin
../client/fpga_compress obj/fullimage.data.bin obj/fullimage.data.bin.z

iceman1001, 2016-05-03:
Look like it hanged when the libzib was trying to compress the fullimage.elf.

did you do these steps?

cd proxmark3
make clean
make all

greatone76, 2016-05-04:
Thank you so much I'm up and running.  I'm sorry for all the noob issues.  I'm very glad you helped me through this.  

For my NTAG Emulation I have for the tag dump working to a .bin and the script working to convert to .eml, but when I go to eload the .eml file it says "test.eml not found or locked".  Where does the .eml file end up and how do I reference it correctly or move it to the correct location or unlock it.  

I'm going to post this in the Proxmark forum in the NTAG213 Emulation thread, so you can close this issue log if you want.

closed 2016-05-04 by iceman1001