iceman1001/proxmark3 #7: Compiling for Mac OS X El Capitan

myFairCatsSick, 2016-05-06:
I tried compiling for mac - I installed all dependencies, gcc48, readline and went through many of the known issues, had to place malloc.h in by hand.

I also could not get rid of the errors in ui.c regarding undefined readline references:
### **
etopt.o obj/tea.o obj/prng.o obj/radixsort.o obj/bucketsort.o guidummy.o -L/opt/local/lib -lreadline -lpthread -lm ../liblua/liblua.a  -o proxmark3
Undefined symbols for architecture x86_64:
  "_rl_copy_text", referenced from:
      _PrintAndLog in ui.o
  "_rl_readline_state", referenced from:
      _PrintAndLog in ui.o
  "_rl_replace_line", referenced from:
      _PrintAndLog in ui.o
  "_rl_restore_prompt", referenced from:
      _PrintAndLog in ui.o
  "_rl_save_prompt", referenced from:
      _PrintAndLog in ui.o
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2**

 no matter what I tried - so I ended up commenting them out to get things running.

Let me know if anyone else runs into these and has a better fix?

iceman1001, 2016-05-06:
Can you run these brew commands and post the output?
brew --config
brew update
brew tap iceman1001/proxmark3
brew install --HEAD proxmark3

myFairCatsSick, 2016-05-06:
>>brew --config
HOMEBREW_VERSION: 0.9.9
ORIGIN: https://github.com/Homebrew/brew.git
HEAD: 6f80b4688bce7b391093412def6e65c3b61cbd20
Last commit: 2 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: fa2970dccd936c86dcf2f443365403cc24796747
Core tap last commit: 34 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_REPOSITORY: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: quad-core 64-bit ivybridge
OS X: 10.11.4-x86_64
Xcode: 7.3.1
CLT: 7.3.0.0.1.1457485338
GCC-4.2: build 5666
Clang: 7.3 build 703
MacPorts/Fink: /opt/local/bin/port
X11: 2.7.7 => /opt/X11
System Ruby: 2.0.0-p648
Perl: /usr/bin/perl
Python: /Library/Frameworks/Python.framework/Versions/2.7/bin/python => /Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Ruby: /usr/bin/ruby => /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby
Java: 1.8.0_40, 1.7.0_21, 1.6.0_65-b14-468

>>brew update
Updated Homebrew from 6f80b46 to 1c1c48c.
Updated 4 taps (homebrew/core, homebrew/dupes, homebrew/science, homebrew/versions).
==> New Formulae
byteman              github-markdown-toc  hspell               libosinfo            ocamlbuild           percona-xtrabackup   zplug              
==> Updated Formulae
app-engine-go-32             checkstyle                   homebrew/versions/gnupg21    jfrog-cli-go                 opam                       
app-engine-go-64             crystal-lang                 homebrew/versions/jpeg9      juise                        premake                    
buildapp                     devil                        homebrew/versions/node4-lts  kobalt                       qtfaststart                
byobu                        henplus                      homebrew/versions/node5      lablgtk                      svtplay-dl                 
camlp4                       homebrew/science/fwdpp       hqx                          ocaml                      
camlp5                       homebrew/science/raxml       imgur-screenshot             ocamlsdl    

>>brew tap iceman1001/proxmark3
==> Tapping iceman1001/proxmark3
Cloning into '/usr/local/Library/Taps/iceman1001/homebrew-proxmark3'...
Username ....
Password ....
remote: Repository not found.
fatal: repository 'https://github.com/iceman1001/homebrew-proxmark3/' not found
Error: Failure while executing: git clone https://github.com/iceman1001/homebrew-proxmark3 /usr/local/Library/Taps/iceman1001/homebrew-proxmark3 --config core.autocrlf=false --depth=1

>> brew install --HEAD proxmark3
Error: /Library/Caches/Homebrew/Formula/proxmark3.rb:5: syntax error, unexpected '<'
<!DOCTYPE html>
 ^
/Library/Caches/Homebrew/Formula/proxmark3.rb:6: syntax error, unexpected '<'
<html lang="en" class="">
 ^
/Library/Caches/Homebrew/Formula/proxmark3.rb:6: syntax error, unexpected keyword_class, expecting end-of-input
<html lang="en" class="">
                     ^
Please report this bug:
    https://git.io/brew-troubleshooting
/usr/local/Library/Homebrew/formulary.rb:21:in `module_eval'
/usr/local/Library/Homebrew/formulary.rb:21:in `load_formula'
/usr/local/Library/Homebrew/formulary.rb:38:in `load_formula_from_path'
/usr/local/Library/Homebrew/formulary.rb:87:in `load_file'
/usr/local/Library/Homebrew/formulary.rb:78:in `klass'
/usr/local/Library/Homebrew/formulary.rb:74:in `get_formula'
/usr/local/Library/Homebrew/formulary.rb:211:in `factory'
/usr/local/Library/Homebrew/formulary.rb:356:in `find_with_priority'
/usr/local/Library/Homebrew/extend/ARGV.rb:20:in `block in formulae'
/usr/local/Library/Homebrew/extend/ARGV.rb:16:in `map'
/usr/local/Library/Homebrew/extend/ARGV.rb:16:in `formulae'
/usr/local/Library/Homebrew/cmd/install.rb:95:in `install'
/usr/local/Library/brew.rb:87:in `<main>'

iceman1001, 2016-05-06:
Thanks, 
The brew tap command fails,   it tries to load homebrew-proxmark3..   That is something that needs to be fixed.

myFairCatsSick, 2016-05-06:
I also tried:

brew install https://github.com/iceman1001/proxmark3/blob/master/proxmark3.rb

Which is why you see the errors for the last command as well...not sure something is wrong there as well or I called it wrong.

iceman1001, 2016-05-06:
Ok, I somethings wrong when it comes to _homebrew_.  
Can you try again with the  
**brew tap iceman1001/proxmark3**

myFairCatsSick, 2016-05-06:
brew install --HEAD proxmark3
==> Installing proxmark3 from iceman1001/proxmark3
==> Cloning https://github.com/iceman1001/proxmark3.git
Updating /Library/Caches/Homebrew/proxmark3--git
==> Checking out branch master
==> make -C client/hid-flasher/
Last 15 lines from /Users/supermanruleshereallydoes/Library/Logs/Homebrew/proxmark3/01.make:
2016-05-06 10:53:17 -0500

make
-C
client/hid-flasher/

gcc -std=gnu99 -I. -I../include -I../common -I/opt/local/include -Wall -Wno-unused-function  -g -O3 -c -o obj/flash.o flash.c
In file included from flash.c:15:
./proxusb.h:30:3: error: must use 'struct' tag to refer to type 'usb_dev_handle'
  usb_dev_handle *handle;
  ^
  struct 
1 error generated.
make: *** [obj/flash.o] Error 1

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/iceman1001/homebrew-proxmark3/issues

iceman1001, 2016-05-06:
ok,  thats much better..
Now, in my fork, the old HID-flasher doesnt compile anymore. 
So I pushed an update not to compile it.

Would you mind trying it again?

myFairCatsSick, 2016-05-06:
I don't see any changes - still get the error when I do:

brew tap iceman1001/proxmark3
brew install --HEAD proxmark3

also tried 
brew reinstall --HEAD proxmark3

iceman1001, 2016-05-06:
hm,  so that was not how to comment out in a brewfile.  ok, I removed those two lines instead.
You only need to run the "Brew tap" command, since if it fails, there rest will also fail.
Please, try again

myFairCatsSick, 2016-05-06:
same error still.

iceman1001, 2016-05-06:
same error?  it should not try to compile the hid-flasher...

myFairCatsSick, 2016-05-06:
>>brew tap iceman1001/proxmark3
>>$ brew install --HEAD proxmark3
==> Installing proxmark3 from iceman1001/proxmark3
==> Cloning https://github.com/iceman1001/proxmark3.git
Updating /Library/Caches/Homebrew/proxmark3--git
==> Checking out branch master
==> make -C client/hid-flasher/
Last 15 lines from /Users/sasdfasdfsdfsdfasdf/Library/Logs/Homebrew/proxmark3/01.make:
2016-05-06 11:27:30 -0500

make
-C
client/hid-flasher/

gcc -std=gnu99 -I. -I../include -I../common -I/opt/local/include -Wall -Wno-unused-function  -g -O3 -c -o obj/flash.o flash.c
In file included from flash.c:15:
./proxusb.h:30:3: error: must use 'struct' tag to refer to type 'usb_dev_handle'
  usb_dev_handle *handle;
  ^
  struct 
1 error generated.
make: *** [obj/flash.o] Error 1

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/iceman1001/homebrew-proxmark3/issues

myFairCatsSick, 2016-05-06:
so even if I go into client/proxmark3.rb and remove the hid-flasher make AND install lines, (and fixed malloc.h missing on Mac OSX) I run into the issue when its building of:

`IP -DZLIB_PM3_TUNED  -c -o obj/ui.o ui.c
ui.c:40:19: error: use of undeclared identifier 'rl_readline_state'
        int need_hack = (rl_readline_state & RL_STATE_READCMD) > 0;
                         ^
ui.c:40:39: error: use of undeclared identifier 'RL_STATE_READCMD'
        int need_hack = (rl_readline_state & RL_STATE_READCMD) > 0;
                                             ^
ui.c:44:16: warning: implicit declaration of function 'rl_copy_text' is invalid in C99 [-Wimplicit-function-declaration]
                saved_line = rl_copy_text(0, rl_end);
                             ^
ui.c:44:14: warning: incompatible integer to pointer conversion assigning to 'char *' from 'int' [-Wint-conversion]
                saved_line = rl_copy_text(0, rl_end);
                           ^ ~~~~~~~~~~~~~~~~~~~~~~~
ui.c:45:3: warning: implicit declaration of function 'rl_save_prompt' is invalid in C99 [-Wimplicit-function-declaration]
                rl_save_prompt();
                ^
ui.c:46:3: warning: implicit declaration of function 'rl_replace_line' is invalid in C99 [-Wimplicit-function-declaration]
                rl_replace_line("", 0);
                ^
ui.c:58:3: warning: implicit declaration of function 'rl_restore_prompt' is invalid in C99 [-Wimplicit-function-declaration]
                rl_restore_prompt();`

so I then ran:
>>brew link --force readline

and then I get:
`ld: warning: directory not found for option '-L/opt/local/lib'
Undefined symbols for architecture x86_64:
  "_memalign", referenced from:
      _crack_states_bitsliced in cmdhfmfhard.o
  "_rl_copy_text", referenced from:
      _PrintAndLog in ui.o
  "_rl_readline_state", referenced from:
      _PrintAndLog in ui.o
  "_rl_replace_line", referenced from:
      _PrintAndLog in ui.o
  "_rl_restore_prompt", referenced from:
      _PrintAndLog in ui.o
  "_rl_save_prompt", referenced from:
      _PrintAndLog in ui.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2
`

Which is what I was never able to get past, without commenting that all out.

iceman1001, 2016-05-06:
hm, ok,  I added the path for  Darvin/Mac OSX to malloc,  in the client\makefile.

That should fix your malloc.h issue.

myFairCatsSick, 2016-05-06:
Thanks - that still leaves the full branch not working but helps remove a manual step which is great!

iceman1001, 2016-05-06:
now, back to the _memalign issue..
It seems darvin/mac os x,  doesnt have memalign, but  its  malloc is already memaligned 16bytes.  So,  I pushed a fix.  
Please try again,

myFairCatsSick, 2016-05-06:
I'm still not getting your flasher fixes or this change - not quite sure what is up there.

iceman1001, 2016-05-06:
try untap,  then tap it again?
brew untap iceman1001/proxmark3

myFairCatsSick, 2016-05-06:
ok - now I got the flasher updates and that part doesn't try and compile, and readline errors were gone too - can you make the fix for __APPL__ as well?

iceman1001, 2016-05-06:
ok, at least we are getting somewhere.  I'm running blind here, you know.  Give some credit.
I pushed the APPLE define fix.

myFairCatsSick, 2016-05-06:
ok - so I am only left with these now:

Undefined symbols for architecture x86_64:
  "_rl_copy_text", referenced from:
      _PrintAndLog in ui.o
  "_rl_readline_state", referenced from:
      _PrintAndLog in ui.o
  "_rl_replace_line", referenced from:
      _PrintAndLog in ui.o
  "_rl_restore_prompt", referenced from:
      _PrintAndLog in ui.o
  "_rl_save_prompt", referenced from:
      _PrintAndLog in ui.o
ld: symbol(s) not found for architecture x86_64

iceman1001, 2016-05-06:
One more down,  some few left.
Those relates to your Readline issue actually..
Can you provide with which commands you run and there outputs?   Im looking for if the readline gives some warnings

myFairCatsSick, 2016-05-06:
>>brew reinstall readline
`==> Reinstalling readline
==> Downloading https://homebrew.bintray.com/bottles/readline-6.3.8.el_capitan.bottle.tar.gz
Already downloaded: /Library/Caches/Homebrew/readline-6.3.8.el_capitan.bottle.tar.gz
==> Pouring readline-6.3.8.el_capitan.bottle.tar.gz
==> Caveats
This formula is keg-only, which means it was not symlinked into /usr/local.

OS X provides the BSD libedit library, which shadows libreadline.
In order to prevent conflicts when programs look for libreadline we are
defaulting this GNU Readline installation to keg-only.

Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

    LDFLAGS:  -L/usr/local/opt/readline/lib
    CPPFLAGS: -I/usr/local/opt/readline/include

==> Summary
🍺  /usr/local/Cellar/readline/6.3.8: 46 files, 2.0M
`

then I do:
>>brew link --force readline
`Linking /usr/local/Cellar/readline/6.3.8... 18 symlinks created`

and then make clean/make install - and I get the errors still.

iceman1001, 2016-05-06:
So that sounds like Darvin wants some extra paths in the makefile.  ok. lets try that then.

iceman1001, 2016-05-06:
I've push a fix.
please try again

myFairCatsSick, 2016-05-06:
getting further.....

`n file included from /usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/types.h:63:0,
                 from apps.h:17,
                 from start.c:13:
/usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/_stdint.h:32:20: error: conflicting types for 'uint32_t'
 typedef __uint32_t uint32_t ;
                    ^
In file included from ../include/usb_cmd.h:21:0,
                 from ../include/proxmark3.h:17,
                 from start.c:12:
./stdint.h:21:24: note: previous declaration of 'uint32_t' was here
 typedef unsigned int   uint32_t;
                        ^
make[1]: *** [obj/start.o] Error 1
make: *** [armsrc/all] Error 2
`

iceman1001, 2016-05-06:
hm,  in   armsrc/apps.h   line 17...   can you test to comment it out  <sys/types.h>  and run:

make clean 
make all

iceman1001, 2016-05-06:
I pushed the change,   please try again

myFairCatsSick, 2016-05-06:
`./stdint.h:21:24: note: previous declaration of 'uint32_t' was here
 typedef unsigned int   uint32_t;
                        ^
In file included from start.c:13:0:
apps.h:113:14: error: conflicting types for 'LogTrace'
 bool RAMFUNC LogTrace(const uint8_t *btBytes, uint16_t len, uint32_t timestamp_start, uint32_t timestamp_end, uint8_t *parity, bool readerToTag);
              ^
In file included from apps.h:21:0,
                 from start.c:13:
BigBuf.h:37:21: note: previous declaration of 'LogTrace' was here
 extern bool RAMFUNC LogTrace(const uint8_t *btBytes, uint16_t iLen, uint32_t timestamp_start, uint32_t timestamp_end, uint8_t *parity, bool readerToTag);
                     ^
make[1]: *** [obj/start.o] Error 1
make: *** [armsrc/all] Error 2
`

iceman1001, 2016-05-06:
hm, now it just look strange again.  
how do you compile this?   running the brew install?!

myFairCatsSick, 2016-05-06:
brew install --HEAD proxmark3

iceman1001, 2016-05-06:
I changed the homebrew script,  it should do
make clean 
make all  

please, try again

myFairCatsSick, 2016-05-06:
I tried that manually which didn't help, also tried the update which didn't help.

iceman1001, 2016-05-06:
alright,  What the output?  Where does it go wrong now?

myFairCatsSick, 2016-05-06:
$ brew install --HEAD proxmark3

*==>** Installing proxmark3 from iceman1001/proxmark3*

*==>** Cloning https://github.com/iceman1001/proxmark3.git
<https://github.com/iceman1001/proxmark3.git>*

Cloning into '/Library/Caches/Homebrew/proxmark3--git'...

remote: Counting objects: 612, done.

remote: Compressing objects: 100% (569/569), done.

remote: Total 612 (delta 82), reused 308 (delta 37), pack-reused 0

Receiving objects: 100% (612/612), 6.58 MiB | 3.17 MiB/s, done.

Resolving deltas: 100% (82/82), done.

Checking connectivity... done.

*==>** Checking out branch master*

*==>** make clean*

*==>** make all*

Last 15 lines from
/Users/sdfgsdfgdsfgsdfg/Library/Logs/Homebrew/proxmark3/02.make:

                 from start.c:12:

./stdint.h:21:24: note: previous declaration of 'uint32_t' was here

 typedef unsigned int   uint32_t;

                        ^

In file included from start.c:13:0:

apps.h:113:14: error: conflicting types for 'LogTrace'

 bool RAMFUNC LogTrace(const uint8_t *btBytes, uint16_t len, uint32_t
timestamp_start, uint32_t timestamp_end, uint8_t *parity, bool readerToTag);

              ^

In file included from apps.h:21:0,

                 from start.c:13:

BigBuf.h:37:21: note: previous declaration of 'LogTrace' was here

 extern bool RAMFUNC LogTrace(const uint8_t *btBytes, uint16_t iLen,
uint32_t timestamp_start, uint32_t timestamp_end, uint8_t *parity, bool
readerToTag);

                     ^

make[1]: *** [obj/start.o] Error 1

make: *** [armsrc/all] Error 2

READ THIS: https://git.io/brew-troubleshooting

If reporting this issue please do so at (not Homebrew/brew):

  https://github.com/iceman1001/homebrew-proxmark3/issues

iceman1001, 2016-05-06:
can you make a pastebin of the whole logfile?

myFairCatsSick, 2016-05-06:
it wasn't that big just sent it over to you.

iceman1001, 2016-05-06:
thanks, 
it seems to be a stdint, types  conflict when compiling gcc-arm..

iceman1001, 2016-05-06:
ok,  pushed another fix.  We shouldn't have call to stdio.h on the ARMside..

myFairCatsSick, 2016-05-06:
Kevins-MacBook-Pro:~ Kevin$ brew install --HEAD proxmark3

*==>** Installing proxmark3 from iceman1001/proxmark3*

*==>** Cloning https://github.com/iceman1001/proxmark3.git
<https://github.com/iceman1001/proxmark3.git>*

Updating /Library/Caches/Homebrew/proxmark3--git

*==>** Checking out branch master*

*==>** make clean*

*==>** make all*

Last 15 lines from /Users/Kevin/Library/Logs/Homebrew/proxmark3/02.make:

                 from start.c:12:

./stdint.h:21:24: note: previous declaration of 'uint32_t' was here

 typedef unsigned int   uint32_t;

                        ^

In file included from start.c:13:0:

apps.h:113:14: error: conflicting types for 'LogTrace'

 bool RAMFUNC LogTrace(const uint8_t *btBytes, uint16_t len, uint32_t
timestamp_start, uint32_t timestamp_end, uint8_t *parity, bool readerToTag);

              ^

In file included from apps.h:21:0,

                 from start.c:13:

BigBuf.h:37:21: note: previous declaration of 'LogTrace' was here

 extern bool RAMFUNC LogTrace(const uint8_t *btBytes, uint16_t iLen,
uint32_t timestamp_start, uint32_t timestamp_end, uint8_t *parity, bool
readerToTag);

                     ^

make[1]: *** [obj/start.o] Error 1

make: *** [armsrc/all] Error 2

READ THIS: https://git.io/brew-troubleshooting

If reporting this issue please do so at (not Homebrew/brew):

  https://github.com/iceman1001/homebrew-proxmark3/issues

iceman1001, 2016-05-07:
Seeing only the last 15lines of the log, is less useful to me.

iceman1001, 2016-05-08:
ok,  I pushed another fix,  which should remove the errors you get for Logtrace-function being already declared.

myFairCatsSick, 2016-05-09:
Thanks - I should try and work some of this myself soon, or just try the
docker image - this is somewhat spinning out of control.

iceman1001, 2016-05-09:
I noticed something else too,  you have a very old GCC,  like 4.2.   You need to update it to 4.8 or 4.9.
But why is the HitagS.h message still there,  there shouldn't be any references to stdio.h in it anymore.

You can always try to compile the PM3 Master instead of my fork. That one is more friendly to older GCC versions to get you started.

myFairCatsSick, 2016-05-09:
hmm - I have gcc4.8 so I'm not sure why.  I will force 4.8 and try again if
it reverted back to original.

iceman1001, 2016-05-09:
I somehow didn't push that hitags.h changes. I pushed them, so there should be other errors now.

myFairCatsSick, 2016-05-09:
somehow GCC 4.2 shows up in that log file even when I do gcc -v on command
line and I get 4.8.....maybe macs aren't so great after all!

I will try the docker image I think.

myFairCatsSick, 2016-05-09:
Not sure how/why the 02.make log file shows GCC as version 4.2?????

$ gcc -v

Using built-in specs.

COLLECT_GCC=gcc

Target: x86_64-apple-darwin14.1.0

Configured with: ../configure --build=x86_64-apple-darwin14.1.0
--prefix=/usr/local/Cellar/gcc48/4.8.4
--libdir=/usr/local/Cellar/gcc48/4.8.4/lib/gcc/4.8
--enable-languages=c,c++,objc,obj-c++ --program-suffix=-4.8
--with-gmp=/usr/local/opt/gmp4 --with-mpfr=/usr/local/opt/mpfr2
--with-mpc=/usr/local/opt/libmpc08 --with-cloog=/usr/local/opt/cloog018
--with-isl=/usr/local/opt/isl011 --with-system-zlib
--enable-libstdcxx-time=yes --enable-stage1-checking
--enable-checking=release --enable-lto --disable-werror
--with-pkgversion='Homebrew gcc48 4.8.4' --with-bugurl=
https://github.com/Homebrew/homebrew-versions/issues --enable-plugin
--disable-nls --enable-multilib

Thread model: posix

gcc version 4.8.4 (Homebrew gcc48 4.8.4)

iceman1001, 2016-05-09:
Well,  try the
git pull
make clean 
make all
one more time,  to see where it stops this time.

If you are going the docker container,  read this:  https://hub.docker.com/r/iceman1001/proxmark3/

Don't know how it work on Mac.

iceman1001, 2016-05-09:
The GCC issue seems to be a PATH related issue,   http://stackoverflow.com/questions/28970935/osx-replace-gcc-version-4-2-1-with-4-9-installed-via-homebrew  

Try update your ,bashrc and set your PATH env accordingly

iceman1001, 2016-05-16:
How did it go for you?  I'm closing this one.  If you need, do please open another issue.

closed 2016-05-16 by iceman1001