iceman1001/proxmark3 #9: hf 14a sim x -- can't recover key usign mfkey32

zhovner, 2016-05-27:
I try to use `hf 14a sim t 1 u 1122344 x` and get this:

```
Emulating ISO/IEC 14443 type A tag with 4 byte UID (11 22 33 44 )
Press pm3-button to abort simulation
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 0101010c 7dd47b66 03d3143f cc444d14 82dd5f9f
Enter mfkey32
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 01010116 c87267bc 2f0d4f4e 6f0ed790 5f402c85
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 01010120 fc105d69 4e2855d3 07b515ee cb529591
Time in mf32key: 521320 ticks 

Enter mfkey32_moebius
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 0101012a 82d67f5b 50a00161 b11cc9a6 c49cf766
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 01010134 011f8c4f 3893e58d 89a2e633 10f0dcde
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 0101013e 20cd029e 582b4a9a f4e5693f ffe62a46
Time in mfkey32_moebius: 524057 ticks 

Enter mfkey32
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 01010148 c7e28d00 050543bd 5db8b84b 0ef2daf0
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 01010152 64341e06 8b855314 55e38b2d c0263ffd
Time in mf32key: 513880 ticks 

Enter mfkey32_moebius
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 0101015c df6518a6 d2772509 fb538251 806ed0a9
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 01010166 8a28a6c8 f99fd53f b27a6df0 b521f2e3
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 01010170 bc06d228 cc7f82b6 8872259a 12125bb5
Time in mfkey32_moebius: 517238 ticks 

Enter mfkey32
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 0101017a 49d9369a 212b2877 2ab2fec8 00021419
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 01010184 d400949d 32ea0784 6eec461b c5f8b70f
Time in mf32key: 400049 ticks 

Enter mfkey32_moebius
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 0101018e d106bb80 46d29961 1ec04fb0 04c34eb7
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 01010198 4e5e1f22 debe6837 f0df9a67 a8ed1c99
Time in mfkey32_moebius: 405646 ticks 

Enter mfkey32
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 010101a2 f5f257d8 e4c79b30 01dbcba8 a6b6b7dc
Time in mf32key: 337041 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 328752 ticks 

Enter mfkey32
Time in mf32key: 441795 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 443736 ticks 

Enter mfkey32
Time in mf32key: 391389 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 397668 ticks 

Enter mfkey32
Time in mf32key: 431515 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 434262 ticks 

Enter mfkey32
Time in mf32key: 320496 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 319287 ticks 

Enter mfkey32
Time in mf32key: 375993 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 376867 ticks 

Enter mfkey32
Time in mf32key: 312789 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 315243 ticks 

Enter mfkey32
Time in mf32key: 381093 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 382302 ticks 

Enter mfkey32
Time in mf32key: 399618 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 391661 ticks 

Enter mfkey32
Time in mf32key: 310761 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 309417 ticks 

Enter mfkey32
Time in mf32key: 408467 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 411931 ticks 

Enter mfkey32
Time in mf32key: 387380 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 385217 ticks 

Enter mfkey32
Time in mf32key: 361426 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 360463 ticks 

#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 010101b0 dc26c937 ee7e73ce 37eb0caa da1a74c6
Enter mfkey32
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 010101bc cc8316d0 a9017c4a 563e7ffb 30cfed94
Time in mf32key: 363426 ticks 

Enter mfkey32_moebius
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 010101c6 57185912 8bbcd289 97879455 a6524bb0
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 010101d0 d70e23b5 8d04f480 8089da95 60dfb6a8
Time in mfkey32_moebius: 364397 ticks 

Enter mfkey32
Time in mf32key: 326639 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 324742 ticks 

Enter mfkey32
Time in mf32key: 420310 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 422265 ticks 

Enter mfkey32
Time in mf32key: 409619 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 402283 ticks 

#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 010101db c41a1759 0dc746ae 1c0466d8 3fdda2d5
Enter mfkey32
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 010101e5 ac2905be 549c01dd 879efee1 7758deda
Time in mf32key: 340741 ticks 

Enter mfkey32_moebius
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 010101ef 194b62d2 892797e4 9f969097 0275eda9
#db# Collected two pairs of AR/NR which can be used to extract keys from reader:
#db# ../tools/mfkey/mfkey32 11223344 010101f9 24182fff 5de0e23b 005a966b 411486e5
Time in mfkey32_moebius: 349865 ticks 

Enter mfkey32
Time in mf32key: 632524 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 638289 ticks 

Enter mfkey32
Time in mf32key: 417693 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 422874 ticks 

Enter mfkey32
Time in mf32key: 445529 ticks 

Enter mfkey32_moebius
Time in mfkey32_moebius: 447829 ticks 
```

Why I run mfkey32 directly i get this:

```
$ ../tools/mfkey/mfkey32 11223344 01010148 c7e28d00 050543bd 5db8b84b 0ef2daf0

MIFARE Classic key recovery - based 32 bits of keystream
Recover key from two 32-bit reader authentication answers only!

Recovering key for:
    uid: 11223344
     nt: 01010148
 {nr_0}: c7e28d00
 {ar_0}: 050543bd
 {nr_1}: 5db8b84b
 {ar_1}: 0ef2daf0

LFSR succesors of the tag challenge:
  nt': febac9ae
 nt'': d4b5b5c3

Keystream used to generate {ar} and {at}:
  ks2: fbbf8a13
Time : 484377 ticks 
```

iceman1001, 2016-05-27:
Difference in output is because the "sim -x" command tries to find the key, I changed the output to be more cleaner.

Btw,  mfkey32 and sim -x failed to find the key.

zhovner, 2016-05-27:
My question is why they failed to find key? I try it many times.

iceman1001, 2016-05-27:
why it failed?   Could be that the card has an hardend prng,  then it will fail.
If it as clone like fudan,  it can also fail.  I have observed that mfkey32 fails,  but  mfkey64 works almost everytime.

closed 2016-05-30 by iceman1001

iceman1001, 2016-07-28:
@marshmellow42 did some changes to "sim x" command where it now collects matching nonces, from same authtentication requests to block and keytype.  This makes "sim x" really reliable now. 
He just made a PR to PM3 Master.