iceman1001/proxmark3 #10: Emulating Ultralight EV1

MickaelBergem, 2016-06-07:
I am trying to emulate an Ultralight EV1 (MF0UL1101D) tag with the PM3, latest commit on master.

I can dump the tag with `hf mfu dump` (the default password is used), which gives me a dump with 32 pages (128 bytes):

```
pm3 --> hf mfu dump
TYPE : MIFARE Ultralight EV1 48bytes (MF0UL1101)          
Reading tag memory...          
Error: Authentication Failed UL-EV1/NTAG          

*Special* block data:          

DataType| Data        |   | Ascii          
---------------------------------          
GetVer-1| 00 04 03 01 |   |           
GetVer-2| 01 00 0B 03 |   |           
TBD     | 00 00       |   |           
Tearing |    BD BD BD |   | ���          
Pack    |    00 00    |   |           
TBD     |          00 |   |           
Sig-1   | CE C0 E8 DD |   |           
Sig-2   | EB 5C 09 52 |   |     R          
Sig-3   | FD C9 8C 5A |   | �ɌZ          
Sig-4   | 28 59 3C 62 |   | (Y<b          
Sig-5   | B0 99 06 44 |   | ��D          
Sig-6   | 88 E7 51 C2 |   | �         
Sig-7   | 60 18 BA DB |   | `�         
Sig-8   | F1 BA F6 8B |   | �          

Block#  | Data        |lck| Ascii          
---------------------------------          
00/0x00 | 04 3C 80 30 |   | <�0          
01/0x01 | 9A DB 49 81 |   | ��          
02/0x02 | 89 48 00 00 |   | �H          
03/0x03 | 00 00 00 00 | 0 |           
04/0x04 | 23 EB B2 34 | 0 | #          
05/0x05 | 72 60 41 46 | 0 | r`AF          
06/0x06 | 48 74 08 08 | 0 |           
07/0x07 | 40 EC C3 35 | 0 | @5          
08/0x08 | 59 C8 53 B9 | 0 | Y�          
09/0x09 | 6D 6F 5E 89 | 0 | mo^�          
10/0x0A | 45 70 DC 86 | 0 | Ep܆          
11/0x0B | A4 42 DA CB | 0 | �B          
12/0x0C | 8C E8 B9 B3 | 0 | �蹳          
13/0x0D | 32 DD C5 C4 | 0 | 2         
14/0x0E | 36 E2 1D D5 | 0 | 6         
15/0x0F | 70 A8 AB 39 | 0 | p��9          
16/0x10 | 00 00 00 FF | 0 |           
17/0x11 | 00 05 00 00 | 0 |           
18/0x12 | 00 00 00 00 | 0 |           
19/0x13 | 00 00 00 00 | 0 |           
---------------------------------          
Dumped 32 pages, wrote 128 bytes to 043C809ADB4981.bin
```

I then use the `dumptoemul-mfu` script to convert this bin file to its plain ASCII representation and get the .eml file:

```
00040301
01000b03
fd00bdbd
bd000000
cec0e8dd
eb5c0952
fdc98c5a
28593c62
b0990644
88e751c2
6018badb
f1baf68b
043c8030
9adb4981
89480000
00000000
23ebb234
72604146
48740808
40ecc335
59c853b9
6d6f5e89
4570dc86
a442dacb
8ce8b9b3
32ddc5c4
36e21dd5
70a8ab39
000000ff
00050000
00000000
00000000
```

Unfortunately, running `hf mfu eload u 043C809ADB4981.eml 32` followed by `hf mfu sim t 2 u 043C809ADB4981` doesn't work properly: 
* the emulator does not work on the usual reader (card not recognized)
* the NFC TagInfo application on my NFC-enabled phone doesn't recognize the card properly, but instead says the card is a "Mifare Ultralight X", with... 1232 bytes of memory, with 308 4-bytes-long pages

The actual card has 80 bytes of memory , with only 20 pages. My phone tries to enumerate all the 308 pages to retrieve the content of the memory and hang until I remove the card.

I guess metadatas are not properly understood by the emulator / specific commands are not implemented.

I also tried to only use the 20-pages-long dump (only the data), but I don't get a better result.

Any idea?

iceman1001, 2016-06-13:
The "hf mfu sim" command should be  **hf mfu sim t 7 u 043C809ADB4981**

iceman1001, 2016-06-13:
And I'm supposing you mean this fork's master branch,  otherwise it will not work at all

iceman1001, 2016-06-18:
how did it go with this?

MickaelBergem, 2016-08-05:
Sorry I didn't have time to investigate this issue, and I won't have access to a PM3 until a long time so I'm just going to close this issue.

Ultralight EV1 are increasingly deployed in cashless payment solutions, such as in music festivals or in the Euro 2016 fan zones in Paris, so I hope someone will have time to confirm everything works as expected to emulate such cards ;)

closed 2016-08-05 by MickaelBergem