iceman1001/proxmark3 #11: Compiling under Mac OSX  #include <malloc.h> error

FredTacker, 2016-06-17:
if i try to Compiling the standart iceman1001/proxmark3 reposity with homebrew under mac osx 10.10 the terminal tells me the following error:

cmdhfmfhard.c:33:10: fatal error: 'malloc.h' file not found
#include <malloc.h>
         ^
1 error generated.
make[1]: *** [obj/cmdhfmfhard.o] Error 1
make: *** [client/all] Error 2

could you maybe help me to solve this problem?

iceman1001, 2016-06-17:
can you run  "uname" on a terminal ?

iceman1001, 2016-06-17:
and you can test in cmdhfmfhard.c to change the malloc.h call to this:

#ifndef __APPLE__
	#include <malloc.h>
#endif

iceman1001, 2016-06-20:
how did it got with this one?  If no answer, I'll close it.

FredTacker, 2016-06-20:
i didn't had time to check it out yet sorry

iceman1001, 2016-07-17:
Since I haven't heard more here, I'm closing it.

closed 2016-07-17 by iceman1001