iceman1001/proxmark3 #22: hf mf chk

osysltd, 2016-08-26:
Dear Iceman,

Just noticed, with the latest commit version hf mf chk stopped working for me. Also hf se works from time to time.
I've managed to get pm3 working only on a1ab594e244c1554b2e997efa8813e8ec3b80017

Here is device info (pm3 easy):

> bootrom: /-suspect 2015-04-02 15:12:04
> os: icemanHEAD/v1.1.0-1472-ga1ab594-suspect 2016-08-26 21:17:59
> LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
> HF FPGA image built for 2s30vq100 on 2015/11/ 2 at  9: 8: 8
> uC: AT91SAM7S256 Rev D
> Embedded Processor: ARM7TDMI
> Nonvolatile Program Memory Size: 256K bytes. Used: 216596 bytes (83%). Free: 45548 bytes (17%).
> Second Nonvolatile Program Memory Size: None
> Internal SRAM Size: 64K bytes
> Architecture Identifier: AT91SAM7Sxx Series
> Nonvolatile Program Memory Type: Embedded Flash Memory

iceman1001, 2016-08-26:
I'm guessing you have to older green model of pm3 device?

https://github.com/iceman1001/proxmark3/commit/63a1d80130db41e4d18bce2e6ada85c6a2c569ac

If you change the  tc2 >= 1 line to the earlier tc0 < 0xff0 ?

iceman1001, 2016-08-26:
I've adjusted the timeout, the 'hf mf chk'  should work again.

closed 2016-08-26 by iceman1001