iceman1001/proxmark3 #23: homebrew does not compile on MacOS el capitan

SpaceTeddy, 2016-08-31:
brew install --HEAD proxmark3
==> Installing proxmark3 from iceman1001/proxmark3
==> Tapping nitsky/stm32
Cloning into '/usr/local/Library/Taps/nitsky/homebrew-stm32'...
remote: Counting objects: 7, done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 7 (delta 0), reused 5 (delta 0), pack-reused 0
Unpacking objects: 100% (7/7), done.
Checking connectivity... done.
Tapped 2 formulae (33 files, 24.2K)
==> Cloning https://github.com/iceman1001/proxmark3.git
Updating /Users/xxx/Library/Caches/Homebrew/proxmark3--git
==> Checking out branch master
==> make clean
==> make all
Last 15 lines from /Users/xxx/Library/Logs/Homebrew/proxmark3/02.make:
In file included from iso14443a.h:20:0,
                 from desfire_crypto.h:9,
                 from desfire_crypto.c:28:
../common/cmd.h:41:6: error: conflicting types for 'cmd_send'
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
In file included from ./util.h:18:0,
                 from ./string.h:16,
                 from desfire_crypto.h:4,
                 from desfire_crypto.c:28:
./apps.h:214:6: note: previous declaration of 'cmd_send' was here
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
make[1]: *** [obj/desfire_crypto.o] Error 1
make: *** [armsrc/all] Error 2

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/iceman1001/homebrew-proxmark3/issues

iceman1001, 2016-08-31:
Sorry, I've updated homebrew repo for a long time since I don't use OS X myself.

Try again, I changed it to use the latest release instead v.1.6.5

SpaceTeddy, 2016-08-31:
wow, 
thanks for your quick feedback, but... I got similar errors:

brew install proxmark3
==> Auto-updated Homebrew!
Updated 1 tap (iceman1001/proxmark3).
==> Updated Formulae
iceman1001/proxmark3/proxmark3
==> Installing proxmark3 from iceman1001/proxmark3
==> Downloading https://github.com/iceman1001/proxmark3/archive/v1.6.5.tar.gz
==> Downloading from https://codeload.github.com/iceman1001/proxmark3/tar.gz/v1.6.5
######################################################################## 100,0%
==> make clean
==> make all
Last 15 lines from /Users/xxx/Library/Logs/Homebrew/proxmark3/02.make:
In file included from iso14443a.h:20:0,
                 from desfire_crypto.h:9,
                 from desfire_crypto.c:28:
../common/cmd.h:41:6: error: conflicting types for 'cmd_send'
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
In file included from ./util.h:18:0,
                 from ./string.h:16,
                 from desfire_crypto.h:4,
                 from desfire_crypto.c:28:
./apps.h:214:6: note: previous declaration of 'cmd_send' was here
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
make[1]: *** [obj/desfire_crypto.o] Error 1
make: *** [armsrc/all] Error 2

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/iceman1001/homebrew-proxmark3/issues

iceman1001, 2016-08-31:
can you pull the latest source and compile ?

SpaceTeddy, 2016-09-18:
get the following message now:

In file included from iso14443a.h:20:0,
                 from desfire_crypto.h:9,
                 from desfire_crypto.c:28:
../common/cmd.h:41:6: error: conflicting types for 'cmd_send'
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
In file included from ./util.h:18:0,
                 from ./string.h:16,
                 from desfire_crypto.h:4,
                 from desfire_crypto.c:28:
./apps.h:214:6: note: previous declaration of 'cmd_send' was here
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
make[1]: *** [obj/desfire_crypto.o] Error 1
make: *** [armsrc/all] Error 2

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/iceman1001/homebrew-proxmark3/issues

exploitagency, 2016-10-01:
Shot in the dark here as I dont use OS X but in proxmark3/common/Makefile.common look for -Werror and change it to just -W it is behind CFLAGS arguments.

It warns instead of making all warnings be errors and aborting.

I've seen this fix various firmware compile issues with no affect to function of the firmware when theres a tiny error in code or at least how the compiler interprets it.  maybe homebrew has stricter warnings?  of course the warnings are always there for a reason under certain conditions its possible to have the wrong data assigned to a variable but that should not be the case here

exploitagency, 2016-10-01:
See what happens when you edit proxmark3/common/Makefile.common and change -Werror to -W 

It is one of the args behind CFLAGS

iceman1001, 2017-01-20:
I noticed that *issues* wasn't enabled in homebrew-proxmark3 repo.  No wonder ppl post it here.  *my bad*

And to track,  I've updated the homebrew,  how about you try it out and see if it works.

SpaceTeddy, 2017-01-20:
Hi,
I've tied to install via Homebrew (github works fine), but unfortunately it does not work to install the full package.

> Christians-MBP:proxmark3 Christian$ brew tap iceman1001/proxmark3
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core, iceman1001/proxmark3).
==> Updated Formulae
libgosu                                     rabbitmq                                    statik

Chris-MBP:proxmark3 Christian$ brew install --HEAD proxmark3
==> Installing proxmark3 from iceman1001/proxmark3
==> Installing dependencies for iceman1001/proxmark3/proxmark3: iceman1001/proxmark3/arm-none-eabi-gcc
==> Installing iceman1001/proxmark3/proxmark3 dependency: iceman1001/proxmark3/arm-none-eabi-gcc
==> Downloading https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update/+download/gcc-arm-none-eabi-5_4-2016q3-20160926-mac.ta
==> Downloading from https://launchpadlibrarian.net/287101378/gcc-arm-none-eabi-5_4-2016q3-20160926-mac.tar.bz2
######################################################################## 100,0%
🍺  /usr/local/Cellar/arm-none-eabi-gcc/20160926: 5,502 files, 447.0M, built in 2 minutes 25 seconds
==> Installing iceman1001/proxmark3/proxmark3 --HEAD
==> Cloning https://github.com/iceman1001/proxmark3.git
Cloning into '/Users/Christian/Library/Caches/Homebrew/proxmark3--git'...
remote: Counting objects: 629, done.
remote: Compressing objects: 100% (588/588), done.
remote: Total 629 (delta 83), reused 295 (delta 35), pack-reused 0
Receiving objects: 100% (629/629), 6.64 MiB | 1.50 MiB/s, done.
Resolving deltas: 100% (83/83), done.
==> Checking out branch master
==> make -C client/hid-flasher/
Last 15 lines from /Users/Christian/Library/Logs/Homebrew/proxmark3/01.make:
2017-01-20 18:20:57 +0100

make
-C
client/hid-flasher/

gcc -std=gnu99 -I. -I../include -I../common -I/opt/local/include -Wall -Wno-unused-function  -g -O3 -c -o obj/flash.o flash.c
In file included from flash.c:15:
./proxusb.h:30:3: error: must use 'struct' tag to refer to type 'usb_dev_handle'
  usb_dev_handle *handle;
  ^
  struct 
1 error generated.
make: *** [obj/flash.o] Error 1

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/iceman1001/homebrew-proxmark3/issues

>

iceman1001, 2017-01-20:
hid-flasher doesn't compile for me on Mingw,  but it does compile on Ubuntu.

The old HID-Flasher project needs Libusb do compile. However, when using Homebrew the search paths for include (gcc -I) is not matching up.   I've pushed a fix that disables the making/installing of the old HID-flasher.

And pushed another fix for my fork,  which will be pulled from git.  

Try again?

SpaceTeddy, 2017-01-20:
well done!

==> Installing proxmark3 from iceman1001/proxmark3
==> Installing dependencies for iceman1001/proxmark3/proxmark3: iceman1001/proxmark3/arm-none-eabi-gcc
==> Installing iceman1001/proxmark3/proxmark3 dependency: iceman1001/proxmark3/arm-none-eabi-gcc
==> Downloading https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update/+download/gcc-arm-none-eabi-5_4-2016q3-20160926-mac.ta
Already downloaded: /Users/Christian/Library/Caches/Homebrew/arm-none-eabi-gcc-20160926.tar.bz2
Warning: This keg was marked linked already, continuing anyway
🍺  /usr/local/Cellar/arm-none-eabi-gcc/20160926: 5,502 files, 447.0M, built in 22 seconds
==> Installing iceman1001/proxmark3/proxmark3 --HEAD
==> Cloning https://github.com/iceman1001/proxmark3.git
Updating /Users/Christian/Library/Caches/Homebrew/proxmark3--git
==> Checking out branch master
==> make all clean
==> make
==> Install success!  Only proxmark3-flasher (modern firmware) is available.
==> The latest bootloader and firmware binaries are ready and waiting in the current homebrew Cellar within share/firmware.
🍺  /usr/local/Cellar/proxmark3/HEAD-360a5b1: 11 files, 1M, built in 1 minute

closed 2017-01-20 by SpaceTeddy