iceman1001/proxmark3 #24: compile git rep. on MacOS

SpaceTeddy, 2016-08-31:
In file included from /usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/types.h:63:0,
                 from /usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/strings.h:13,
                 from desfire_crypto.h:5,
                 from desfire_crypto.c:28:
/usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/_stdint.h:31:19: error: conflicting types for 'int32_t'
 typedef __int32_t int32_t ;
                   ^
In file included from ./string.h:15:0,
                 from desfire_crypto.h:4,
                 from desfire_crypto.c:28:
./stdint.h:16:18: note: previous declaration of 'int32_t' was here
 typedef int      int32_t;
                  ^
In file included from /usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/types.h:63:0,
                 from /usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/strings.h:13,
                 from desfire_crypto.h:5,
                 from desfire_crypto.c:28:
/usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/_stdint.h:32:20: error: conflicting types for 'uint32_t'
 typedef __uint32_t uint32_t ;
                    ^
In file included from ./string.h:15:0,
                 from desfire_crypto.h:4,
                 from desfire_crypto.c:28:
./stdint.h:21:24: note: previous declaration of 'uint32_t' was here
 typedef unsigned int   uint32_t;
                        ^
In file included from iso14443a.h:20:0,
                 from desfire_crypto.h:9,
                 from desfire_crypto.c:28:
../common/cmd.h:41:6: error: conflicting types for 'cmd_send'
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
In file included from ./util.h:18:0,
                 from ./string.h:16,
                 from desfire_crypto.h:4,
                 from desfire_crypto.c:28:
./apps.h:214:6: note: previous declaration of 'cmd_send' was here
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
make[1]: *** [obj/desfire_crypto.o] Error 1
make: *** [armsrc/all] Error 2

iceman1001, 2016-08-31:
Thanks for the feedback!
I've no ppl to test out the OS X devenv. Now lets see I've removed a redundant call to cmd.h, that should reduce your error messages a bit,

The readline/string.h I've seen before,

SpaceTeddy, 2016-08-31:
dear iceman,
I updated and compiled again, but unfortunately the same errors. do you know if it can be the root cause that I installed arm-none-eabi-gcc via homebrew?

thanks a lot.

DWITH_HFSNOOP -fno-strict-aliasing -ffunction-sections -fdata-sections -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -I. -Os -mthumb-interwork -o obj/desfire_key.o desfire_key.c 
arm-none-eabi-gcc -c -I../include -I../common -Wall -Werror -pedantic -Wunused -std=c99 -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DON_DEVICE -DWITH_HFSNOOP -fno-strict-aliasing -ffunction-sections -fdata-sections -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -I. -Os -mthumb-interwork -o obj/desfire_crypto.o desfire_crypto.c 
In file included from /usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/types.h:63:0,
                 from /usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/strings.h:13,
                 from desfire_crypto.h:5,
                 from desfire_crypto.c:28:
/usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/_stdint.h:31:19: error: conflicting types for 'int32_t'
 typedef __int32_t int32_t ;
                   ^
In file included from ./string.h:15:0,
                 from desfire_crypto.h:4,
                 from desfire_crypto.c:28:
./stdint.h:16:18: note: previous declaration of 'int32_t' was here
 typedef int      int32_t;
                  ^
In file included from /usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/types.h:63:0,
                 from /usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/strings.h:13,
                 from desfire_crypto.h:5,
                 from desfire_crypto.c:28:
/usr/local/Cellar/arm-none-eabi-gcc/20150921/arm-none-eabi/include/sys/_stdint.h:32:20: error: conflicting types for 'uint32_t'
 typedef __uint32_t uint32_t ;
                    ^
In file included from ./string.h:15:0,
                 from desfire_crypto.h:4,
                 from desfire_crypto.c:28:
./stdint.h:21:24: note: previous declaration of 'uint32_t' was here
 typedef unsigned int   uint32_t;
                        ^
In file included from iso14443a.h:20:0,
                 from desfire_crypto.h:9,
                 from desfire_crypto.c:28:
../common/cmd.h:41:6: error: conflicting types for 'cmd_send'
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
In file included from ./util.h:18:0,
                 from ./string.h:16,
                 from desfire_crypto.h:4,
                 from desfire_crypto.c:28:
./apps.h:214:6: note: previous declaration of 'cmd_send' was here
 bool cmd_send(uint32_t cmd, uint32_t arg0, uint32_t arg1, uint32_t arg2, void* data, size_t len);
      ^
make[1]: *** [obj/desfire_crypto.o] Error 1
make: *** [armsrc/all] Error 2

iceman1001, 2016-08-31:
I've no idea actually.  Running blind so to speak. 
There has been others, 
https://github.com/Proxmark/proxmark3/issues/183
https://github.com/Proxmark/proxmark3/issues/189

who has made it.  Can you follow their instructions?

SpaceTeddy, 2016-08-31:
hi iceman,

I commented out //#include <strings.h> in desfire_crypto.h and it compiles well now.

:)

#ifndef __DESFIRE_CRYPTO_H
#define __DESFIRE_CRYPTO_H

#include <string.h>
//#include <strings.h>
#include <stdarg.h>
#include "printf.h"

#include "iso14443a.h"
#include "../common/desfire.h"
#include "des.h"
//#include "aes.h"

#endif

iceman1001, 2016-08-31:
Thanks, I pushed the removal of the strings.h which seemed to be a problem. 
Could you pull and make a clean compilation?

SpaceTeddy, 2016-08-31:
done and works fine:
.
.
.
Applications/Xcode.app/Contents/Developer/usr/bin/make -C recovery all
arm-none-eabi-objcopy --gap-fill=0xff --pad-to 0x00102000 -O binary ../bootrom/obj/bootrom.elf bootrom.bin
arm-none-eabi-objcopy --gap-fill=0xff -O binary ../armsrc/obj/fullimage.elf fullimage.bin
cat bootrom.bin fullimage.bin > proxmark3_recovery.bin

closed 2016-08-31 by SpaceTeddy

iceman1001, 2016-08-31:
Since you do have OS X,  do you mind testing out the QT5 issues? Is there a way of mailing you?

SpaceTeddy, 2016-09-01:
Yes for sure, I can try my best :)
I'm available at spaceteddy1@gmx.net 
Hope I can help.