iceman1001/proxmark3 #27: FAC/Card No converter broken?

ralik, 2016-09-16:
Hello I am on 5d8246, the Wiegand converter does not seem to work anymore. Is it only me?
```

pm3 --> lf hid wiegand 0 4 410
HID | OEM |  FC |   CN  |  Wiegand  |  HID Formatted          
----+-----+-----+-------+-----------+--------------------          
 -123776832 | 0  | 4  | 410  | 0000000000000000  |  0000000000000000          
 -123776832 | 0  | 4  | 410  | 0000000000000000  |  0000000000000000          
 -123776832 | 0  | 4  | 410  | 0000000000000000  |  0000000000000000          
 -123776832 | 0  | 4  | 410  | 0000000000000000  |  0000000000000000          
 -123776832 | 0  | 4  | 410  | 0000000000000000  |  0000000000000000          
 -123776832 | 0  | 4  | 410  | 0000000000000000  |  0000000000000000          
 -123776832 | 0  | 4  | 410  | 0000000000000000  |  0000000000000000          
----+-----+-----+-------+-----------+--------------------
```

iceman1001, 2016-09-16:
Yeah, I know, I wasn't happy with it so I removed it until I could implement a better version.
Feel free to contribute.

ralik, 2016-09-16:
It was very useful! And always worked for H10301 (so far)

iceman1001, 2017-01-10:
I've started to re-work it. Some changes is pushed.

iceman1001, 2019-09-19:
I took the wiegand conversion from offical repo and merged into RRG/Iceman repo, since this repo is deprecated.
ref
https://github.com/RfidResearchGroup/proxmark3

You find the new command direct
```
pm3->weigand
```

closed 2019-09-19 by iceman1001