iceman1001/proxmark3 #28: lf partially broken

osysltd, 2016-09-16:
Dear Chris,

I've faced the problem when `lf search` works well, however `lf t55x detect` and `lf em4x em410xwrite` make the device (RDV 2.0) to reset.

```
pm3 --> lf em4x em410xwrite 030070AB2A 1
Writing T55x7 tag with UID 0x030070ab2a (clock rate: 64)          
#db# Started writing T55x7 tag ...          
#db# Clock rate: 64          
pm3 --> 
```

Please could you check?
Thank you!

exploitagency, 2016-09-23:
+1

compiled your commit this morning

lf hid clone resets it after a few seconds

proxmark3> lf hid fskdemod
proxmark3> 
proxmark3> #db# TAG ID: 2006485524 (10898) - Format Len: 26bit - FC: 36 - Card: 10898                 
proxmark3> #db# TAG ID: 2006485524 (10898) - Format Len: 26bit - FC: 36 - Card: 10898                 
proxmark3> lf hid clone 2006485777
Cloning tag with ID 2006485777          
proxmark3> 
proxmark3> #db# Stopped

RDV2 - RESET

iceman1001, 2016-09-23:
I know, I know,  the previous calls to spindelayus() wasn't good enough so I started changing them to a more precise timer.  However my changes to lfops.c got pushed to my fork now when I fiddled with "hf legic".  
I've now pushed a fix for this.   The WDT resets and the LF functionality should be back in business.

iceman1001, 2016-09-23:
Still, Thank you for testing and reminding me.

exploitagency, 2016-09-23:
Works great....


> proxmark3> lf t55x detect
> Modulation : FSK2a          
> Bit Rate   : 4 - RF/50          
> Inverted   : Yes          
> Offset     : 1          
> Block0     : 0x00107060          
> proxmark3> 
> proxmark3> lf em4x em410xwrite 030070AB2A 1
> Writing T55x7 tag with UID 0x030070ab2a (clock rate: 64)          
> proxmark3> 
> proxmark3> #db# Started writing T55x7 tag ...                 
> proxmark3> #db# Clock rate: 64                 
> proxmark3> #db# Tag T55x7 written with 0xff80c003c14b969a
> proxmark3> lf em4x em410xread
> proxmark3> 
> proxmark3> lf em4x em410xdemod
> proxmark3> 
> proxmark3> #db# EM TAG ID: 030070ab2a - (43818_112_07383850)                 
> proxmark3> #db# EM TAG ID: 030070ab2a - (43818_112_07383850)            

  

You the man!

iceman1001, 2016-09-23:
Still need to verify a lot of the current default-timings in lfops.c 
I'll close this one now.

closed 2016-09-23 by iceman1001