iceman1001/proxmark3 #29: Undefined references to `crc_update2'

osysltd, 2016-09-21:
Hello Iceman,

Sorry for bothering, but while trying to build the latest trunk getting the following error:

`obj/cmdhflegic.o: In function 'calc4':

/root/pm/ice/client/cmdhflegic.c:709: undefined reference to 'crc_update2'
/root/pm/ice/client/cmdhflegic.c:710: undefined reference to 'crc_update2'
/root/pm/ice/client/cmdhflegic.c:711: undefined reference to 'crc_update2'
/root/pm/ice/client/cmdhflegic.c:716: undefined reference to 'crc_update2'
/root/pm/ice/client/cmdhflegic.c:717: undefined reference to 'crc_update2'
obj/cmdhflegic.o:/root/pm/ice/client/cmdhflegic.c:718: more undefined references to 'crc_update2' follow
collect2: error: ld returned 1 exit status
Makefile:184: recipe for target 'proxmark3' failed
make[1]: *** [proxmark3] Error 1
make[1]: Leaving directory '/root/pm/ice/client'
Makefile:12: recipe for target 'client/all' failed
make: *** [client/all] Error 2
`
Haven't yet a chance to take a more precise look though.

iceman1001, 2016-09-21:
Thanks for letting me know,  I took notice of it yesterday when TravisCI started complaining.
It was only the keys files that were supposed to be pushed but an unfinish change found its way in too.  So I'll fix that when I have some more time.

iceman1001, 2016-09-21:
Not that I wanted to add these changes yet,  but now it compiles at least.
There will be more changes to t55xx soon not to mention legic..

closed 2016-09-21 by iceman1001