iceman1001/proxmark3 #30: hf mf can't select card

osysltd, 2016-09-22:
Dear Iceman,

Sorry to bother you. Im trying to write some data to the sectors using pm3 Easy version.
`hf se` goes well as well as first time` hf mf wrbl`, but the second `hf mf wrbl` says `can't select card`.
I may suppose that the problem is in the timings.

iceman1001, 2016-09-22:
nop, shouldnt be any problems with "hf mf wrbl" at all.
I've heard rumors that the pm3 easy kit,  seem to really picky with placement of tag over HF antenna.

osysltd, 2016-09-22:
After wrbl the device led goes out and start to show up after trying hf se serveral times. Then wrbl works again only once.

iceman1001, 2016-09-22:
hm, strange.  just to make certain you do have the same fullimage.elf flashed as client from same build?

osysltd, 2016-09-22:
Be sure, prior to open issue I'm compiling and flashing the latest commit and efficiently position the card as well. How I can help to investigate the issue?

iceman1001, 2016-09-22:
Lets see,  "hf mf dbg 4"  and run the writeblock again.  the output would be nice.
also the "hf list 14a" output.  
What kind of tag?   Is the access-write ok?  key ok?

osysltd, 2016-09-22:
My bad, indeed pm3 easy _very_ sensitive for the card position, I've found the right place finally.
Very sorry Chris!

closed 2016-09-22 by osysltd