iceman1001/proxmark3 #35: lf sim does not work due to timing

Highpressure, 2016-10-19:
when running your last build with your firmware simulating em4x does not work 
when running main build with main firmware it works fine
when running main your build with main firmware (yep I know), it brings stranges numbers to the display and reader is stuck for a second

I assume its as you said the timing or something mixed up

iceman1001, 2016-10-19:
Looking at it, I'm guessing that the timer wasn't started when LF simulating.  
I've added it and pushed.  Will you pull and flash to test it?

Highpressure, 2016-10-19:
yes, can do it tomorrow. dont have a reader at home.

iceman1001, 2016-10-19:
Nop, wasnt that easy. I'll need to see which functions that depends on the GPIO_SSC_CLK clock.

iceman1001, 2016-10-20:
Using two pm3,  one "lf em em410xsim" and the other "lf search" I can see the data plot that it is sending the tagsignal and the receiver gets it. However it can't decode it at all. Too many errors.

Highpressure, 2016-10-20:
I would not recommend to test with to "fake" readers.
E.g. when I try the hf mf sim commands it works with the acr122u reader but wont progress as there is no answer. In real life when I check it with a real reader it wont work, as it is checking for other keys - multiple keys and so also wont progress. (the next issue i am opening as soon as I can report detailed what my experience is in front of the reader)

please push the new code and I´ll flash and test it.. I´ll drive over to my company to test ;)
if you will be online like 8pm we can do live testing. Could provide you GotoMeeting / TeamViewer session and you can see live whats happening - if you would like

iceman1001, 2016-10-20:
This issue is about **lf sim**  and **not**  hf mf sim. 
two complete different commands,  please keep to the subject.

Highpressure, 2016-10-20:
yep, but the issue with readers behaving different than the readers we connect to a pc.
I thought maybe you get also different results when reading with an proxmark than when reading with a "real reader". So I thought of giving it a try

iceman1001, 2016-10-20:
when it comes to LF, we mimic signal quite exact as we snooped it.  
when it comes to generating LF from tag, we are still quite on par. timewise.
its a complete different story with HF where the underlaying protocol demands much more.

iceman1001, 2017-02-03:
To be able to solve this one, we need to show the output from "lf sim" of a em4 tag.
compared with a proper tag / reader output. 

Is this something you can collect?

Highpressure, 2017-02-04:
Would it be ok to read it with a 2nd proxmark or do you want to have a legit em4 reader?
We have it at work for checkin/checkout. I´m not sure if I can read raw or what is in the backend.
Going to ask. Back at work next week after some time off. Thats also why I didnt answer yet on the other request in the forum.

iceman1001, 2017-02-04:
A legit em4x reader would have been the best option. 
A sniff trace between valid card/reader is needed for further analysing

Highpressure, 2017-02-06:
partitially works.

did
lf em4x em41xdemod 
hold the legit em4x token to see ID

lf em4x em410xsim tagid
went to our time system / check-in/check-out reader "Gantner Electronic - Terminal 1050"
success
had to hold it really close compared to the tokens

went to our door lock - no reaction
there is no brand / name on it, so I cant tell by now.

Gonna create a sniff - more in the forum then

iceman1001, 2017-08-07:
I do belive the **lf sim** works now

closed 2017-08-09 by iceman1001