iceman1001/proxmark3 #42: Legic.lua script is not working

iceman1001, 2016-10-20:
With the remake of legic commands, the existing scripts from @mosci doesn't work anymore.

icsom, 2017-01-10:
I will take a look at that in a couple of days

akileos, 2017-04-30:
Issue still open ? Might be able to have a look

iceman1001, 2017-05-01:
still not working,  have a go at it :)

akileos, 2017-05-04:
Starting point : 

function readFromPM3()
   local tag, bytes, infile
     infile="legic.temp"
     core.console("hf legic reader")
-    core.console("hf legic save "..infile)
+    core.console("hf legic dump o "..infile)
+    infile="legic.temp.bin"

--
function getInputBytes(infile)
 doesn't work yet for me, carrying on.

iceman1001, 2017-05-23:
@akileos  are you ready to merge your changes?

akileos, 2017-05-23:
No sir, all crc calculations are still wrong on my tests and have no clue yet on what to include on CRC() to fix those. Advice welcome as you have one of those badge with exact same issue ;)

iceman1001, 2017-05-23:
do you mean all crc calculations from the legic.lua script is wrong?

akileos, 2017-05-24:
Those were I guess perfectly fine with mosci's segments but don't match mine at least. Need some more time to figure out what's wrong here, not a CRC master like some other folks here

iceman1001, 2017-05-25:
ok...  well,  how about you post your progress on the forum and explain with samples / what you are trying to archive.

iceman1001, 2017-06-05:
With @akileos  latest PR, 
legic.lua now reads **binary** files, instead of ascii/eml hex files.

The script in general looks like it works.  What doesn't work?  Besides a unknown crc-16 algo for some legic-cash which needs to be identified.    Time to close?

akileos, 2017-06-05:
There are still a lot of things that still need to be ironed out.
Token creation : Doesn't works except for SAM
'dlc' / 'd3p' : Only dumps last matching segment
kgh doesn't seem to be working as expected aswell

Maybe I missed more, up to you to keep or close. Any help appreciated.

Explorer1092, 2017-10-12:
has fixed？

iceman1001, 2017-10-12:
if you use @icsom  (mosci) fork,  then the legic.lua script works there fine.

here it still has some issues,

WolfgangMau, 2018-01-17:
just to get it clear.
legic segments can be designed in various ways - you even can design your own segment.
the lua-script just covers a couple of segments, which I could have my hands on.
d3p (dump third party) per example was a cash-segment from a 3rd party, which was used in a cantina - and the script works perfect on that particularly segment. And in fact: on those cards it was always the second (and last) segment.
don't think the script is perfect - it was just perfect for me (mosci); and I shared it in hope it gets extended by others.
all segments I could get my hand on have being described within the comments of the script.
There could be millions of different segments, you can't cover them all, because if I would, I can create 100 different segments a day - The reader just checks the header and crc's.
Per example: the 'd3p'-segment is in fact a cash-segment, but totally different to a legic-cash segment.
It's part of the 3rd party to implement the verification of the data then - and part of the 'hacker' to reverse-engineer this layout.
The script is not working for any kind of legic-segments - but it has some mapping features which makes it more easy to get them 'reverse-engineered'

iceman1001, 2018-01-17:
You made a great script. 

The main reason for this script not working in iceman fork is the "no decision" idea to support eml or bin dumps.  I'm more leaning to support both but not got around to implement it.  Once that is in place, the script generations will work.  

-  add support for bin / eml for  load / save

iceman1001, 2018-08-06:
There is some new improvements for legic prime support coming.  Part of it got merged (@drandreas)

- improved reading distance  0-5cm (keyfob)  2-8cm (card)
-simplified code
- simulation....

akileos, 2018-08-08:
Hello @iceman1001 , may I ask why It was reverted ? Thanks

iceman1001, 2018-08-08:
The old reading distance is still there.
the new pr was a work-in-progress,   a missunderstanding between @drandreas and me.  
He has gotting write / leds / tracelog to work since.   So his refactoring is now better than what we have.

When he feels ok,  he will make new PR.

akileos, 2018-08-09:
@iceman1001 Gave it a test, can now read tags  1-2 cm from reader with @drandreas fork. Was a millimeter positioning job before.

iceman1001, 2018-08-09:
Thats an improvement indeed :)   Once @drandreas will push it,  I merge it.

Only thing with legic.lua script is the acceptans of both binary / eml files...

drandreas, 2018-08-09:
write works now too

iceman1001, 2019-05-14:
The legic script in RRG repo saves bin / eml,   I will close this issue.

closed 2019-05-14 by iceman1001