iceman1001/proxmark3 #45: hf mf sim: CMD_SIMULATE_MIFARE_CARD response sends different to expected arguments

micolous, 2016-10-22:
Poking through the sources:

https://github.com/iceman1001/proxmark3/blob/master/armsrc/iso14443a.c#L2983

Here if the `INTERACTIVE` flag is set for `hf mf sim`, then the ar_nr_responses are sent back to the host.  However, when the client program actually gets the data, it tries to push this data into a `nonces_t` struct:

https://github.com/iceman1001/proxmark3/blob/master/client/cmdhfmf.c#L1502

Then this is pushed into the `mfkey32` attack with bad parameters, when the debug strings suggest using `mfkey32v2` (moebius).

What probably should happen here is that the device should encode and pass back the sector number and key ID in the same format as the `nonces_t` struct, so that the client can start attacking for the key.

I think that the `hf mf sim` codepath is out of sync with the rest of the repository, because `hf 14a sim` appears to be correct.

However, the debug strings for `hf mf sim` appear to indicate correct commands, while `hf 14a sim` doesn't appear to have correct data needed to calculate the key.

Also, the functionality to set the emulator memory doesn't appear to set the key correctly.

I'm working on a patch for this issue.

iceman1001, 2016-10-26:
Great work correcting the errornous "hf mf sim" on the device side.
With FLAG_NR_AR_ATTACK  flag I'm still not too fond of the debug print statements in the end (iso14443a.c).  If the current impl is stable we don't need them.  

That actually goes the same for the debug statments you put back in the nonce2key mfkey functions.
I think it gobbels up the output.   Good for when you debug it.  

Now back to the attack itself.  I've not test this fix. Does the sim work stable with finding the key?

micolous, 2016-10-27:
When using `hf mf sim` with my patch, I was able to recover keys reliably with the PM3 up against a pn532 card reader connected being driven by libnfc:

```
nfc-mfclassic r a /dev/null keyfile.mfd f
```

Where `keyfile.mfd` is a dump of a Mifare Classic 1K, with the key blocks populated.

And on the PM3:

```
hf mf sim u 12345678 i x e
```

However, I've so far been able to only grab one sector key with it, probably to do with the way libnfc is doing authentication.  I think what it's doing is trying to authenticate to the last sector, then gives up when that fails.

Another easy one is to try with an Android phone, which returns the Mifare Application Directory key.  It looks like the Android OS is trying to authenticate with a card before any application on the phone can, and PM3 doesn't say "no" in the right way for Android to treat it as a simple authentication failure (so Android OS is always trying to auth):

```
pm3 --> hf mf sim u 12345678 i x e
 uid:12 34 56 78 , numreads:0, flags:35 (0x23)           
Press pm3-button or send another cmd to abort simulation          
#db# 4B UID: 12345678          
enter reader attack
Recovering key for:
    uid: 12345678
   nt_0: 01010104
 {nr_0}: a19ede1f
 {ar_0}: 996b11f9
   nt_1: 0101010a
 {nr_1}: 725c0a94
 {ar_1}: cd5f7647

LFSR succesors of the tag challenge:
  nt': d62ecf79
 nt'': 4b936a31
#db# Collected two pairs of AR/NR which can be used to extract keyA from reader for sector 0:          
#db# ../tools/mfkey/mfkey32 12345678 01010104 a19ede1f 996b11f9 ad894c7e 40878626          
#db# Collected two pairs of AR/NR which can be used to extract keyB from reader for sector 0:          
#db# ../tools/mfkey/mfkey32v2 12345678 01010104 a19ede1f 996b11f9 0101010a 725c0a94 cd5f7647          
#db# Emulator stopped. Tracing: 1  trace length: 882           
Time in mfkey32_moebius: 229166 ticks  - possible keys 1
          
M-Found KeyA for sector 00: [a0a1a2a3a4a5]          
Setting Emulator Memory Block 03: [A0 A1 A2 A3 A4 A5 FF 0F 80 69 FF FF FF FF FF FF ]          
```

I haven't been able to do the same with `hf 14a sim`, and I'm unsure what is wrong with that.

My thought on debugging symbols on the client is to make them optional in a flag.  I'm pretty sure it's useful for someone who is trying to debug key recovery, and it's definitely useful to make sure the PM3 is sending the values correctly.

The code also appears to need a keypress to return the console, not sure why that is.

iceman1001, 2016-10-27:
Sounds stable for me. I'll try it against a reader I have laying around.

Debugstatements are good for the purpose of debugging, yes indeed. 
In your output you also see what I meant with the statements becomes gobbled up.

I suggest you just comment them out and leave them in the code, for ppl to recompile if they ever will need to debug it again.  Not a very likely scenario.

The reason for "hf 14a sim" not working is because its not built for the rest of the mifare protocoll in its imp.  You can't never set the emulatore memory to let the device pretend further in the sim process and transmit it encrypted.  
So "hf 14a sim" will be useful to get a first key,  to be used with "nested" or "hardnested"

Purpose of "hf 14a sim" is also not to be a mifare imp, thats why we have a "hf mf sim"  :)

micolous, 2016-10-27:
Acknowledged.  If I can get at least 1 key out of it, then that's good enough to be able to attack the Mifare Plus cards.  Being able to do this all in-protocol (rather than debug command responses) is important to me, as I'm currently in the process of porting the client to Java (in order to have a native Android port).

Also having it work properly with the C client is a important first step, before I even attempt to port it to Java :smile:

micolous, 2016-10-27:
PR #49 should address the verbosity issue.

iceman1001, 2016-10-27:
Much appreciated, bro. 
I suggest you change the "-m" into the more commonly used "-v" for verbose output. 
and of course, change the parameter in functions to  "verbose" instead of showmath.

Its a huge job converting it into java.  I once started to do a pure c# client,  then I stoped when I saw the amount of resources needed.  The Proxmark tool gui by gaucho was a better idea at that time.  Just a gui wrapped over the client.

iceman1001, 2016-10-27:
👍  Merged 
and closing.

closed 2016-10-27 by iceman1001