iceman1001/proxmark3 #53: hf mf hardnested -  device fails to answer

iceman1001, 2016-11-15:
hf mf hardnested -  device fails to answer the call to brute_force fails. 
On the client side it just iterates and sends the collecting nonce command but somehow the device doesn't want to do that.  It goes inside the metod MifareAcquireEncryptedNonces() / armsrc/mifarecmd.c
but gets stuck in a loop. 
I wonder if its the  "while(GetCountSspClk() < timeout);" loop thats the cause to this behavior

18688994688, 2016-11-15:
how long it will takes before going to the failed status? I encountered a no response from proxmark error after one hour always during running hardnested (of coz, the key not resolved)

iceman1001, 2016-11-15:
http://pastebin.com/9qSV2epU
Its after the nonces collection when it enters bruteforce mode,   and when it fails,  it should go back and start collecting more nonces again.   Its here when the device doesn't answer anymore.

Its not about how long the nonce collecting takes.

iceman1001, 2016-11-16:
With my latest commits, I get this output below.  
The generate_candidates is called a few times but not all times ok and the client start collecting nonces again. 
The brute_force calls fails and the client start collecting nonces again. 

That is basically what this loop should do.  I do however managed to get the client to crash...

http://pastebin.com/gcuddqaq

iceman1001, 2016-11-16:
ping @matrix

matrix, 2016-11-16:
Hi @iceman1001,
could you retest using my repository?
Let me know

iceman1001, 2016-11-16:
if you have'nt changed anything from yesterday, I have all your stuff.

matrix, 2016-11-16:
Yes probably you have my stuff, but also other mods you commit.

iceman1001, 2016-11-16:
We were on par some days ago,  yr stuff works but still not stable.
Some of the issues is related to when to call the brute_force.
Some is about how the loop after a failed call to brute_force makes the device silent (this thread)
And finally the client crashes,  with yr mods,  and still with my changes to yr mods...  (no thread about it)

matrix, 2016-11-16:
Ok but you do a single test with my repository?
Your code is different from my, also github can't merge in automatic.

iceman1001, 2016-11-16:
Yupp, did that.  Looks good. Since yours must take 5000nonces a step,  you have a high success rate when entering the brute_force mode.  As long as your good_bytes is less than 13, yr version will be faster than piwi's original. 
I haven't gotten a situation when the brute_force failed and it needs to start over with nonce collection. So I can't verify this bug. 
And since I can't get it to drop into brute_force serveral times, I can't trigger the client crash either.  
But it sure looks fine!  Great as alwys.

matrix, 2016-11-17:
Hey @iceman1001 but why you send CMD_FPGA_MAJOR_MODE_OFF before the call the brute_force() function ?

https://github.com/iceman1001/proxmark3/blob/master/client/cmdhfmfhard.c#L888-L890

Remove these lines and I think your bug could be solved :)

iceman1001, 2016-11-18:
I added those lines and the  "always init flag" set,   to see if the device-non-responsive problem would go away.   It didn't help.  As it is now with yr fix to the add_nonce,  the memory doesnt get screwed up and the loop and restart process seems to work again.  I've just not pushed those changes yet.

iceman1001, 2016-11-22:
Ive pushed some changes where I changed some uart.c  timings.  It looks like this fixes my device problem.

closed 2017-04-26 by iceman1001