iceman1001/proxmark3 #54: hf mf hardnested r crashes on pthread_join (Windows)

vasily-z, 2016-12-03:
Client compiled according to Windows guide. System runs Windows 10 (32). Client seems to crush on pthread_join(threads[i], 0); (Line 1716 of client/cmdhfmfhard.c file).
Tested on two machines.

menshiyun, 2016-12-06:
try newer gcc version

iceman1001, 2016-12-07:
Did you compile using GGC version 4.8 or bigger?

vasily-z, 2016-12-07:
Hi! I compiled it with GCC, which comes with QT 5.6.1 (as recommended). I am new to this, but it seems that the version is 4.9.2. I am trying to update GCC, but it not very straightforward on windows...

osysltd, 2016-12-07:
`Thread model: posix, gcc version 4.9.2 (i686-posix-dwarf-rev1, Built by MinGW-W64 project)`
Try to use the following environment and please let us know the results: http://download.qt.io/archive/qt/5.7/5.7.0/qt-opensource-windows-x86-mingw530-5.7.0.exe

menshiyun, 2016-12-07:
You can use mingw-w64-gcc version 6 in msys2.
[This is how I build the PM3 environment.](https://github.com/menshiyun/proxmark3/blob/local/ENV_en.md)(Forgive me for my tough English)
You can change the 4th step to iceman's fork.

vasily-z, 2016-12-07:
@menshiyun thank you! Your solution is simple and it helped.

18688994688, 2016-12-08:
@vasily-z I use @menshiyun 's setup all the way and works perfect :) this guy is a perfect style holic :)