iceman1001/proxmark3 #58: LF Keri - unfinished

iceman1001, 2016-12-15:
Datamapping of Keri is not complete

joef12345, 2019-02-23:
The Encoding for the MS series tags are as follows:

Each Keri tag has a Card Number and Facility ID.

The Card Number can be upto 10 digits

The Facility ID can be between 0 and 31.

With these two number, the Keri Doors software converts them to an 'internal ID' in the older software and 'card number' in the newer doors.net software.  Currently, it is unclear how the software generates the internal ID from the Card Number and Facility ID. The number almost appears to be a hash of some sort from the two numbers.

The Internal ID is what is actually programed into the tag as follows:

The cards transmit 64 bits as follows using PSK1:

000000000000000000000000000001XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX111
^^^^^^^^^^^^^^^^^^^^^^1##########################^^^
Preamble block 29bits of 0s            32bit Internal ID    (First bit always 1)       3Bit of 1s Closing

The 32bit card number is generated as follows: The Internal ID is actually a 32bit unsigned integer. Take the 32bit dec integer and convert that to binary. For example, take 123456789 converted to binary =

00000111 01011011 11001101 00010101

Take the first bit and change it to 1:

10000111 01011011 11001101 00010101

64bit programmed id would be:

0000000000000000000000000000010000111010110111100110100010101111

Programmed into a T5577 tag would be as follows:

Block 0: 603E1040
Block 1: 00000004
Block 2: 3ADE68AF

To read a Keri Tag would be as follows:
LF Read
data rawdemod p1

iceman1001, 2019-02-23:
wow!  
Now I saw both your post in forum and here.  
I added basic support over att RRG repo..
https://github.com/RfidResearchGroup/proxmark3/commit/2dc3bc6af3599f86849d996ae44b7c06713c467f

iceman1001, 2019-02-24:
ok,  I have adjusted **lf keri clone**  to produce the same config block.

iceman1001, 2019-02-24:
and also the **lf keri sim** is available.  Since I can't test it against a valid reader,  it can be considered as work-in-progress.  Would you mind verifying clone & sim commands?

closed 2019-02-24 by iceman1001