iceman1001/proxmark3 #64: REVENG -G  crashes the client

iceman1001, 2016-12-30:
With the update of reveng to 1.4.4,  it now crasches on the client sometimes.

iceman1001, 2017-08-29:
Updated RevEng to v.1.5.2 .   Still breaks the client  reveng -g  on mingw  and doesn't work at all on Ubuntu

iceman1001, 2017-12-04:
Sorry to ask,  but @pwpiwi  these kinds of memory crashes is for you.   Would you mind having a look at it?
Offical pm3 will need to update its reveng also to this version.

pwpiwi, 2017-12-05:
How do you make it crash?

pwpiwi, 2017-12-05:
> a decent long string

How long is "decent"?

iceman1001, 2017-12-05:
this one crashes on  mingw  
**script run e -b 010203040506070809**

iceman1001, 2017-12-05:
on ubuntu 16.04   -- the compilation somehow doesnt load the preset models...
pm3 --> rev -g 010203205060780
no preset models available

bogiton, 2018-02-03:
Sorry to bring this up again, but I just came across this crash myself.
Tried to spot the issue, by diving into the source code, and I think that the crash happens in the mbynam function of the preset.c file where the strlen of the key (model) is called, in line 765.
`	if(!(ukey = malloc((size_t) 1 + strlen(key)))) {`
I guess that the '\0' char is missing from it, thus the strlen call makes it crash.
Couldn't set up a working dev environment to properly debug this, sadly.

iceman1001, 2018-02-03:
if you have ubuntu,  it goes kind of fast for gdb..

iceman1001, 2018-02-03:
got ride of first simple bug,   make **script run e ** exit nicely when no models found.
doesn't take away the underlaying problem.

iceman1001, 2018-02-03:
got ride of second bug,  **script run e** there was an empty string in the first call..

iceman1001, 2018-02-03:
some more bugs,   default value for width if called,   and push lua error if no models found.

still the serious memory bug left...
try calling below a couple of times to trigger it.
script run e -b 010203 -w 1

iceman1001, 2018-02-03:
**script run e -b**  is fixed windows/mingw.  a missing compilation macro  :)

**reveng -g 01020304054065**  now smashes the stack...  :(

iceman1001, 2018-02-03:
ok,  found it and pushed a fix.   It should work just fine now!

closed 2018-02-03 by iceman1001

bogiton, 2018-02-04:
wow! very good job! :+1: