iceman1001/proxmark3 #67: Improve Windows instructions / Add Kali Linux (and Nethunter)

GabrielTK, 2017-01-17:
Hey!
Just a little Tip, why don't you guys use something link the official proxmark3, a downloadable proxspace where you just need to clone the repo and run make? Also, Bash for Windows install support would be nice (and faster).
Also, Instructions for Kali Linux and Nethunter are missing. Im trying right now on Nethunter.
Thank you,
Gabriel

exploitagency, 2017-01-18:
https://github.com/Proxmark/proxmark3/wiki/Kali%20Linux

Standard Kali Linux Rolling Edition based on Debian Testing instructions are in the official Wiki.

Sometimes they modify, rename, or delete packages.  They recently deleted the arm cross compiler then added it back so the instructions were broke for a couple weeks.  Let me know how it goes.

GabrielTK, 2017-01-18:
P -DZLIB_PM3_TUNED  -c -o obj/proxmark3.o proxmark3.c
gcc: error: unrecognized command line option ‘-mpopcnt’
Makefile:202: recipe for target 'obj/proxmark3.o' failed
make[1]: *** [obj/proxmark3.o] Error 1
make[1]: Leaving directory '/root/iceman1001/client'
Makefile:12: recipe for target 'client/all' failed
make: *** [client/all] Error 2
root@kali:~/iceman1001# 
root@kali:~/iceman1001# 

**NOTE: BASH FOR WIN 10 DOES NOT SUPPORT SERIAL**, so it may be impossible

exploitagency, 2017-01-18:
I'll have to look at it when I get time.  (Might be weeks)  IceMan is more likely to help before I get around to testing on  a fresh install.  What OS,  latest Kali?

Run

> sudo apt-get install git build-essential libreadline5 libreadline-dev gcc-arm-none-eabi libusb-0.1-4 libusb-dev libqt4-dev ncurses-dev perl pkg-config

About a week ago I compiled IceMan's latest release on 64bit Kali.

GabrielTK, 2017-01-18:
**FLIXED ON NetHunter: Just Removed the -mpopcnt arg from client/Makefile**

exploitagency, 2017-01-18:
That was going to be my next suggestion.  I just figured you had an odd version of compiler.

GabrielTK, 2017-01-18:
gcc??

exploitagency, 2017-01-18:
Yeah will try to compile latest commit in a minute and see if I get same errorm

GabrielTK, 2017-01-18:
Ok, thx. Also, (just for curiosity) why this fork requires Qt5 on Windows??

exploitagency, 2017-01-18:
@iceman1001

exploitagency, 2017-01-18:
root@kali:~# git clone https://github.com/iceman1001/proxmark3.git
Cloning into 'proxmark3'...
remote: Counting objects: 15354, done.
remote: Compressing objects: 100% (63/63), done.
remote: Total 15354 (delta 33), reused 0 (delta 0), pack-reused 15291
Receiving objects: 100% (15354/15354), 13.75 MiB | 2.48 MiB/s, done.
Resolving deltas: 100% (11786/11786), done.
root@kali:~# cd proxmark3
root@kali~/proxmark3# make
WORKED

GabrielTK, 2017-01-18:
I was compiling on a Nexus 9 device with NetHunter (ARM) rom. I think the problem was the huge list returned by `apt-get upgrade`

exploitagency, 2017-01-18:
Ah... well glad you got it working.  Theres also an Android fork out there.

GabrielTK, 2017-01-18:
Really... :open_mouth Could you please send the link??? Thank you very much!!

exploitagency, 2017-01-18:
https://github.com/search?utf8=✓&q=android+proxmark3

Comes up with 3 repo.  Unsure which one people use.  Most I have spoke to give up or struggle getting it working.  Might work great though, I haven't ever even looked at repo.

GabrielTK, 2017-01-18:
Thank you very much. And... @exploitagency... Just as an observation... I used your guide on my Proxmark3 RDV2 😃

exploitagency, 2017-01-18:
Sweet!  Glad it was useful.  I got to get some sleep though, early start tomorrow.  But will keep an eye out if you need anything.

I think I'd try proxdroid it was updated 1 year ago.  Or keep searching and check under members in that repo to see forks of that fork.

iceman1001, 2017-01-18:
Many questions,
The last proxspace environment, was created 2012?  It brings a MINGW with armcompiler to the Windows OS.  It means compilers and dependecys is old versions.  Its getting harder and harder to find these old tools.   Hence I upgraded to QT5.   The project should look for if you have QT4 or QT5 installed and try compile alternativly.  Bringing support from old to new.  Proxmark3 source is also cross-compiled since so many ppl uses it on different OS/Envs. 
In a way to help out, I added a docker container for ppl who doesn't want to fiddle with old mingw, or linux distros.   Once installed,  its easy to get the latest source and its compiles just fine.   One downside is the plotwindow issue.  For Mac OS ppl, there are a homebrew tap solution,  to make their lifes easier.

Iceman fork is different than PM3 Master.  One breaking change is the bruteforce solver for hardnested attack. It demands a newer GCC,  You need at least version 4.8.   It is also here those extra compiler directives came to the picture.  So removing "**-mpopcount**"  will most definitly influence that part. 

I'm looking forward to hear about how it works on NetHunter / nexus 9 device.    Most functions will work but I'm guessing hardnested will not. 

Feel free to update the offical PM3 wiki with instructions on how to configure/install on Nethunter.

exploitagency, 2017-01-18:
Thanks for the info!  I've been trying to maintain my repo around major releases.  Still need to automate the process but this has been working fine so far.

iceman1001, 2017-01-18:
Me creating issues for the fork is more of a reminder of what needs to be done. A todo list. It doesn't mean I'm gonna do them.

GabrielTK, 2017-01-18:
It works fine on Nexus 9, since it comes with a Kali Linux Chroot. Going to test hardnested now. The instructions are the same as Ubuntu, but must be run inside the Chroot.

closed 2017-01-18 by iceman1001