iceman1001/proxmark3 #68: HF EMV - merging peter fillmores EMV branch

iceman1001, 2017-01-18:
The files are already added to fork. 
However @peterfillmore branch is based on a very old PM3 commit, which this needs to be updated.

I'm guessing that this is a functionality many ppl wants.

iceman1001, 2017-02-05:
It should compile now, but I'm doubtfull its actually working. 

**to compile this,  enable -DWITH_EMV  in armsrc/Makefile**

BE WARNED:
It takes a lot of memory.  There is only 7% left on a  256kb PM3 device.
This is the reason for me not enabling by default.

```
pm3 --> hf emv
help             This help
trans            Perform EMV Reader Transaction
getrng           get random number from terminal
eload            load EMV tag into device
dump             Dump EMV tag values
list             [Deprecated] List ISO7816 history
```

iceman1001, 2017-02-05:
The whole 'hf em sim' part is disabled,  since its looks like code in progress.

exploitagency, 2017-02-06:
When ready let me know and ill add it to  repo

iceman1001, 2017-02-06:
I'm not gonna fiddle with it anymore,  the EMV is 7816-based work over 14a/14b.
Sadly my creditcard does not answer to 14a/14b.   Strangly it does answer to the legic init iv call.

iceman1001, 2017-02-06:
@peterfillmore It compiles on the latest PM3 source at least.

closed 2017-02-06 by iceman1001