iceman1001/proxmark3 #70: Homebrew: Authentication failed for 'https://github.com/iceman1001/homebrew-tap/'

GabrielTK, 2017-01-20:
Hi,
I'm running `brew install proxmark3` after tapping the repo as in the README.md, but I'm getting the following error:
```
Authentication failed for 'https://github.com/iceman1001/homebrew-tap/'
Error: Failure while executing: git clone https://github.com/iceman1001/homebrew-tap /usr/local/Homebrew/Library/Taps/iceman1001/homebrew-tap --depth=1

```
Any ideas??
Thank you,
Gabriel

iceman1001, 2017-01-20:
I wish I did or could help you out.  I don't have a Mac env to test the homebrew tap out.  I saw the @chrisfu repo and modified it to download iceman fork instead.   If you ask him about it and get back with a solution, that would be fine.
Or ask @zhovner , he has also used this homebrew and said it asked for a user/pwd for 'git clone'.

iceman1001, 2017-01-20:
I do see that::   https://github.com/iceman1001/homebrew-tap/  
is wrong.
There is no homebrew-tap repo.

iceman1001, 2017-01-20:
I've pushed a fix,  would you mind testing it out and see if it solves the problem?

iceman1001, 2017-01-20:
I noticed that *issues* wasn't enabled in homebrew-proxmark3 repo.  No wonder ppl post it here.  *my bad*

GabrielTK, 2017-01-20:
Testing now...

GabrielTK, 2017-01-20:
I got a little problem with homebrew. Trying to fix...

iceman1001, 2017-01-20:
Related to  issue https://github.com/iceman1001/proxmark3/issues/23  
This issues is now resolved.   Pull latest from homebrew-proxmark3 repo and follow usual instructions.

closed 2017-01-20 by iceman1001