iceman1001/proxmark3 #71: can not compile since 20.1.

SpaceTeddy, 2017-01-22:
1)
mifareutil.c: In function 'mifare_ultra_auth':
mifareutil.c:280:2: error: implicit declaration of function 'tdes_2key_dec' [-Werror=implicit-function-declaration]
tdes_2key_dec(random_b, enc_random_b, sizeof(random_b), key, IV );
^
mifareutil.c:300:2: error: implicit declaration of function 'tdes_2key_enc' [-Werror=implicit-function-declaration]
tdes_2key_enc(rnd_ab, rnd_ab, sizeof(rnd_ab), key, enc_random_b);
^
cc1: all warnings being treated as errors
make[1]: *** [obj/mifareutil.o] Error 1
make: *** [armsrc/all] Error 2

-> des.h is not activated in mifareutil.h. After activation, error is gone

2)
iso14443b.c:69:3: error: redeclaration of enumerator 'STATE_UNSYNCD'
STATE_UNSYNCD,
^
In file included from mifareutil.h:21:0,
from iso14443b.h:25,
from iso14443b.c:11:
iso14443a.h:66:3: note: previous definition of 'STATE_UNSYNCD' was here
STATE_UNSYNCD,
^
iso14443b.c:102:3: error: redeclaration of enumerator 'DEMOD_UNSYNCD'
DEMOD_UNSYNCD,
^
In file included from mifareutil.h:21:0,
from iso14443b.h:25,
from iso14443b.c:11:
iso14443a.h:36:3: note: previous definition of 'DEMOD_UNSYNCD' was here
DEMOD_UNSYNCD,
^
make[1]: *** [obj/iso14443b.o] Error 1
make: *** [armsrc/all] Error 2

-> don't know how to handle it

System: macOS

Gilles01, 2017-01-22:
Hello,

I have the same issue, i try on kali and a fresh new install ubuntu 14.04.2 LTS.

```
mifareutil.c: In function 'mifare_ultra_auth':
mifareutil.c:280:2: error: implicit declaration of function 'tdes_2key_dec' [-Werror=implicit-function-declaration]
mifareutil.c:300:2: error: implicit declaration of function 'tdes_2key_enc' [-Werror=implicit-function-declaration]
cc1: all warnings being treated as errors
make: *** [obj/mifareutil.o] Error 1
```

iceman1001, 2017-01-24:
Latest commits fixes this.

closed 2017-01-24 by iceman1001