iceman1001/proxmark3 #72: legic advant - [iso15693]

iceman1001, 2017-01-23:
Legic advant implementation for PM3.

- datasheets needed.
- actual cards / dumps?

Given that Grain128a is used in some version "ATC256-MV410", ISO 15693 is used.
http://www.legic.com/en/1093628/atc256-atc1024.html

It would be interesting to see if this attac vector applies.
http://eprint.iacr.org/2012/349.pdf

Mayby our cryptocrunchers @holiman @pwpiwi  find it interesting?

iceman1001, 2017-03-07:
http://ws680.nist.gov/publication/get_pdf.cfm?pub_id=913678

akileos, 2017-04-07:
Hej Iceman,

 Might be able to provide few legic advant cards if of any help to you.

Best,

iceman1001, 2017-04-08:
Sure, having Cards is a great starting point.

iceman1001, 2017-04-21:
I got two keyfobs with Legic Advant (or thats the claim,  thanks @akileos if it was you who send them)

They are not being read currently by PM3 master nor iceman fork..  It could have something to do with ISO15693 support has an issue (is broken).

akileos, 2017-04-21:
No problem, I'll try to snoop traffic once I get a proxmark.

iceman1001, 2017-06-27:
the latest fpga image, fixed the iso15693 support again.  so @akileos  try your tags  w pm3 v3.0.1 offical

sadk, 2017-10-09:
hi @iceman1001
What's new progress recently?

iceman1001, 2017-10-09:
No news.