iceman1001/proxmark3 #73: AppVeyor Build

GabrielTK, 2017-01-24:
Hi.
@iceman1001, just a little tip, why don't you create an appveyor build for windows?? Here is an idea: http://stackoverflow.com/a/26634443/6158547

iceman1001, 2017-01-24:
I did think about it when I added the TravisCI and CoverityScan. At that time I didnt find support for Mingw.
Which your link suggest works now.  If you have done an appveyor.yml file before, feel free to add it to the project.

GabrielTK, 2017-01-27:
Ok, finishing it...

GabrielTK, 2017-01-27:
@iceman1001 Sorry, I actually don't have it

GabrielTK, 2017-01-27:
@iceman1001 If you want, you can host it on my CI. Just add me as a contributor and I'll set it up for you 😄

iceman1001, 2017-01-27:
Sorry, I'm not following :)  I've added a appveyor.yml file to the fork. However I lack the knowledge of configuring it correct.

iceman1001, 2017-04-29:
one step closer.  apparently downloading thirdpart libs is hard on appveyor

iceman1001, 2017-06-05:
Another step closer with @akelios additions to the appveyour.xml ..

iceman1001, 2017-10-23:
With @merlokk 's config it works fine.

closed 2017-10-23 by iceman1001