iceman1001/proxmark3 #77: Standalone modes not working

osysltd, 2017-02-01:
Hello! Just wanted to report that both HF and LF standalone modes are broken. For LF the tag is not being read, for HF - emulation goes with UID with all zeros.

Compiling as advised to enable HF standalone exclude `-DWITH_ISO14443a_StandAlone`

iceman1001, 2017-02-01:
Hm,..  this could be, I never test the standalone modes.  
Do you know which commit that breaks 'em?

iceman1001, 2017-02-01:
LF standalone: Looking at source,  one change is that I turn off LF inside "CmdHIDemodFSK".
row 877, lfops.c   it shouldn't influence since the LF antenna is turned on at the beginning of that function..

iceman1001, 2017-02-01:
I saw that second uidpart gets copied of the first UID part, it could have cleared out your UID.
Pushed a fix for it.  Test it?

osysltd, 2017-02-01:
Thank you, nuch better, HF tag is being emulated however MCT against the device makes it crash, seems due to lots of requests. Any chance to have LF standalone part fixed as well? the card is not being read (on original branch as well btw).

iceman1001, 2017-02-01:
"hf standalone 14a mode",  ref https://github.com/iceman1001/proxmark3/commit/d760c7b3d920636877a30f96fbb876bce798b5a7   New fix,  cleaned up code. Works with 4,7uid for me.

"lf standalone mode",  it only reads HID tags smaller than 44bits.   No em4x or other stuff.  I guess you tried to use it against em4x and of course it doesnt work.  try against a HID tag.

osysltd, 2017-02-01:
Pity that em4x are not supported ;( Thank you for clarification and fixes, now standalone mode works great but I cannot attack device back to pc to make hf list 14a. Is it by design?

iceman1001, 2017-02-01:
Never test standlone much.   I do notice that if I connect to device first with client,  then keep button pressed to enter standalone I can see debugstatments.  
If you figure it out, tell me.
I just added "mifare 4k" again to standalone.

https://legacysecuritygroup.com/index.php/categories/9-rfid/7-proxmark-3-emulating-hid-tags-in-standalone-mode

You can always make your own LF standalone mode, where it does em4x.  Shouldnt be too hard.

Thank you for pointing out the faulty standalone mode.
Im closing this issue now.

closed 2017-02-01 by iceman1001