iceman1001/proxmark3 #79: EMV commands not working

joanbono, 2017-02-08:
Got latest version installed (Uploaded <20hours ago).

`EMV` commands are not working.

~~~~
pm3 --> hf emv trans
#db# unknown command:: 0x0701
pm3 --> hf emv getrng
#db# unknown command:: 0x0706
pm3 --> hf emv eload
File f not found or locked
pm3 --> hf emv dump
#db# unknown command:: 0x0708
Waiting for a response from the proxmark...
Don't forget to cancel its operation first by pressing on the button
Command execute time-out
~~~~

iceman1001, 2017-02-08:
Did you compile with the flag, as instructed in commit?
and did you flash?

osysltd, 2017-02-08:
Guess for EMV stuff menu should be populated if WITH_EMV defined

iceman1001, 2017-02-08:
yeah. I agree.

iceman1001, 2017-02-08:
I did also now notice on ubuntu that with the lfag WITH_EMV it overflows osimage.  
Something that needs to be adjusted in linker scripts.

osysltd, 2017-02-08:
Might this somehow deal with HAS_512..?

iceman1001, 2017-02-08:
That I don't think.  The HAS_512 flag extends availble mem for osimage when used on device.  It doesnt make fulliamge bigger when compiling.  However the EMV commands makes fullimage alot bigger.

joanbono, 2017-02-08:
@iceman1001 you mean de `--HEAD` flag??

iceman1001, 2017-02-08:
Its not the flag I was asking about.

https://github.com/iceman1001/proxmark3/blob/master/armsrc/Makefile#L17

osysltd, 2017-02-10:
@i

joanbono, 2017-02-10:
I'm thinking on use the  EMV commands on a **Proxmark v3 Easy**, which has this specs:

~~~~
uC: AT91SAM7S256 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes. Used: 225858 bytes (86). Free: 36286 bytes (14).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
~~~~ 

The other proxmark I have has `512KB` of Nonvolatile Memory. It's enough (the `256KB` one) for using with EMV?

osysltd, 2017-02-10:
`Nonvolatile Program Memory Size: 256K bytes. Used: 246976 bytes (94%). Free: 15168 bytes ( 6%)`

with the following config:

`
APP_CFLAGS = -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DON_DEVICE -DWITH_HFSNOOP -DWITH_EMV -fno-strict-aliasing -ffunction-sections -fdata-sections
`

iceman1001, 2017-02-10:
@osysltd   I understand your thinking, and sure that would have been sweet. The downside is that most users doesn't know (understand) how to compile and to figure out how to set or not set a flag.  The flag is also static,  since the compiler doesn't ask the device how much memory it has.  

6% free mem on a 256kb device is low and I have no idea what this might have for impact on the device and the code which is running.  No testing has been done. This code is just from Peters fork and made to compile on mine.  Big question is if so much should run on the device instead of the client.  It needs to be refactored one of these days.  

I'm very curious if someone has a contactless card with EMV snd tested this source code.  Did it work? etc etc.

exploitagency, 2017-02-10:
I think a lot of people have the elechouse rdv2 now with 512kb.  I can add precompiled version to my repo once they are working somewhat.  I dont have cards with the right technology yet.

osysltd, 2017-02-10:
Transaction goes well on my emv MasterCard

iceman1001, 2017-02-10:
@osysltd  it does work?!?  cool.  would you mind posting on the forum, showing stuff?

osysltd, 2017-02-13:
@iceman1001 , Wish I could do this but after latest changes building with EMV on `hf se` I've got write timeout. Attaching configs to the post.

iceman1001, 2017-02-13:
@exploitagency  there is a lot of PM3 v3.0 Easy users. Anyway level of experience for this kind of fiddeling is much higher.  I know that ppl wants "one-click" functions but this project is not like that.

iceman1001, 2017-02-13:
@osysltd  'hf search' has no EMV detection.   Where do you get timeouts? It would be helpful if you add debugstatements, traces to show what happens.

osysltd, 2017-02-13:
@iceman1001 I was expecting hf se to show basic tag info.
Currently EMV is not even populated in the menu (with makefile configs attached to previous post).

iceman1001, 2017-02-13:
*pm3 ~$ client/flasher.exe com8 armsrc/obj/fullimage.elf
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0003ed20->0x0003ed20) [R X] @0x94
Error: PHDR is not contained in Flash
Error while loading armsrc/obj/fullimage.elf*

I get this when I compile WITH_EMV and try to flash.  I think I need to remove the LF parts to make room

osysltd, 2017-02-13:
check my config file modification, they suit into 256k well

iceman1001, 2017-02-13:
your armsrc/makefile seems broken,  the app_flags is missing the final \
making the --fn_  calls not applied.   Are you sure your env builds correct?  (make clear & make all)?

osysltd, 2017-02-13:
`killall ModemManager; make clean; git reset --hard; git pull origin master; vim common/Makefile.common; vim armsrc/Makefile; make clean; make -j4 all; ./client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf;
`

[Makefile.common.txt](https://github.com/iceman1001/proxmark3/files/771134/Makefile.common.txt)
[Makefile.armsrc.txt](https://github.com/iceman1001/proxmark3/files/771135/Makefile.armsrc.txt)

No EMV menu is present.

iceman1001, 2017-02-13:
yeah, I see it.  The Makefile is very picky.  
this commit should fix and explain why,  https://github.com/iceman1001/proxmark3/commit/573e8d72c73f12754d5775cca24d7d9380cc2c78

@osysltd  would you mind test again?

osysltd, 2017-02-14:
Building succeeded, even with more free space on the device:
`Nonvolatile Program Memory Size: 256K bytes. Used: 222420 bytes (85%). Free: 39724 bytes (15%)`

[Makefile.armsrc.txt](https://github.com/iceman1001/proxmark3/files/773327/Makefile.armsrc.txt)
[Makefile.common.txt](https://github.com/iceman1001/proxmark3/files/773328/Makefile.common.txt)

iceman1001, 2017-02-14:
Since it works (verified by  @osysltd ) I'm closing this one. 

There is more corrections to be done to get it to work better.

closed 2017-02-14 by iceman1001