iceman1001/proxmark3 #80: LF EM 4305/4x50/4x69 functionality

iceman1001, 2017-02-16:
The old em4x50 / em4x69 functionality is in need for a clean up and refactor,

@marshmellow42 and I have started to remake it.  The goal is to enable PM3 to use EM4305 like it supposed to be with commandset read, write, login which leads us to implement info, dump, restore(?) functionality. Much like the current implementation of T55x7 commands in PM3

To clearify: EM4305 can just like T55x7 mimic other tags.  Somewhat lesser options but still.

iceman1001, 2017-02-24:
@Marshmellow42 has added read/write/dump/info  for  4205/4305/4469/4569  tag. 
I'll haven't added all of it yet since I'm just got my 4305 tags to play with.

iceman1001, 2017-02-27:
Almost done, I still have 'lf em 4x05dump' left.
Sadly I don't get my keyfobs (em4305) to read.  Apparently they need a small and powerful antenna for trigger command mode.  Very strange indeed

iceman1001, 2017-04-21:
So strange how keyfobs needs a much different sized antenna than card sized tags.

marshmellow42, 2017-04-21:
"Maximum power transfer occurs when the NFC tag and NFC poller antenna have the same shape and size. Therefore, in order to achieve the best performance, the choice of NFC tag antenna should be made based on the application and the typical poller NFC antenna for that application. It could be a good idea to test the NFC tag with many readers."
from
https://devzone.nordicsemi.com/blogs/957/nfc-tag-antenna-tuning/

the same is true of LF.

closed 2020-04-30 by iceman1001