iceman1001/proxmark3 #84: Viking clone broke in latest update..

phiberx, 2017-02-28:
pm3 --> lf viking clone 010203
Cloning - ID: 00010203, Raw: F20000000102035A
pm3 --> lf se
Reading 30000 bytes from device memory

Data fetched
Samples @ 8 bits/smpl, decimation 1:1
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

EM410x  pattern found

EM TAG ID      : 0000000000

Possible de-scramble patterns
Unique TAG ID  : 0000000000
HoneyWell IdentKey {
DEZ 8          : 00000000
DEZ 10         : 0000000000
DEZ 5.5        : 00000.00000
DEZ 3.5A       : 000.00000
DEZ 3.5B       : 000.00000
DEZ 3.5C       : 000.00000
DEZ 14/IK2     : 00000000000000
DEZ 15/IK3     : 000000000000000
DEZ 20/ZK      : 00000000000000000000
}
Other          : 00000_000_00000000
Pattern Paxton : 1326592 [0x143E00]
Pattern 1      : 0 [0x0]
Pattern Sebury : 0 0 0  [0x0 0x0 0x0]

Valid EM410x ID Found!
pm3 -->

iceman1001, 2017-02-28:
Ok,  I've pushed a fix.

closed 2017-02-28 by iceman1001

phiberx, 2017-02-28:
Thanks you are awesome!