iceman1001/proxmark3 #96: FW compilation error

mastahofdesastah, 2017-04-02:
Hi Iceman

Trying to compile the FW on the latest update and i get this error:

/emvcmd.o obj/parity.o obj/usb_cdc.o obj/cmd.o -lgcc
obj/emvcmd.o: In function `EMVSelectPPSE':
emvcmd.c:(.text.EMVSelectPPSE+0x4): undefined reference to `emv_selectPPSE'
collect2: error: ld returned 1 exit status
Makefile:104: die Regel für Ziel „obj/fullimage.stage1.elf“ scheiterte
make: *** [obj/fullimage.stage1.elf] Fehler 1

Any idea?

iceman1001, 2017-04-04:
Interesting,  do you think you can pastebin the compilation output from (make clean && make all) and post it here?

mastahofdesastah, 2017-04-05:
Switched from Kali to Ubuntu and have still the same error
https://pastebin.com/THHN4vBZ
and happens only if i uncomment
#APP_CFLAGS += -DWITH_EMV
#COMMON_FLAGS += -DWITH_EMV

iceman1001, 2017-04-05:
Looks like it breaks.  And since its commented out TravisCI does test build it.  Which would explain why it does warn.
I'll have a look into it in a week or two.  Being out of office for awhile

mastahofdesastah, 2017-04-07:
Someone else can test it? Maybe something is wrong with my compiler

osysltd, 2017-04-07:
Got the same error:

```
obj/emvcmd.o: In function `EMVSelectPPSE':
emvcmd.c:(.text.EMVSelectPPSE+0x4): undefined reference to `emv_selectPPSE'
collect2: error: ld returned 1 exit status
```

Make configuration parameters:
                                                                                                                                           
```
# uncomment these two; to enable EMV                                                                                                                  
APP_CFLAGS += -DWITH_EMV                                                                                                                              
COMMON_FLAGS += -DWITH_EMV                                                                                                                            
                                                                                                                                                      
# uncomment these two; to fix 256 vs 512kb PM3 devices                                                                                                
# flashing bootrom -b is needed                                                                                                                       
#APP_CFLAGS += -DHAS_512_FLASH                                                                                                                        
#COMMON_FLAGS += -DHAS_512_FLASH                                                                                                                      
                                                                                                                                                      

APP_CFLAGS = -DWITH_CRC \ 
                         -DON_DEVICE \ 
                         -DWITH_ISO15693 \ 
                         -DWITH_ISO14443b \ 
                         -DWITH_ISO14443a \ 
                         -DWITH_HFSNOOP \ 
                         -DWITH_ISO14443a_StandAlone \ 
                         -fno-strict-aliasing -ffunction-sections -fdata-sections
### IMPORTANT -  move the commented variable below this line
#                        -DWITH_LCD \
```

SpaceTeddy, 2017-07-01:
hi all,
I have the same behaviour on OSX:
arm-none-eabi-gcc -nostartfiles -nodefaultlibs -Wl,-gc-sections -n -Wl,-T,ldscript,-Map,obj/fullimage.stage1.map -o obj/fullimage.stage1.elf obj/version.o obj/fpga_all.o obj/start.o obj/protocols.o obj/fonts.o obj/LCD.o obj/iso15693.o obj/iso15693tools.o obj/lfops.o obj/hitag2.o obj/hitagS.o obj/lfsampling.o obj/pcf7931.o obj/lfdemod.o obj/inflate.o obj/inffast.o obj/inftrees.o obj/adler32.o obj/zutil.o obj/legicrf.o obj/legic_prng.o obj/appmain.o obj/printf.o obj/util.o obj/string.o obj/BigBuf.o obj/ticks.o obj/random.o obj/hfsnoop.o obj/fpgaloader.o obj/iso14443a.o obj/mifareutil.o obj/mifarecmd.o obj/mifaresniff.o obj/epa.o obj/iso14443b.o obj/crapto1.o obj/crypto1.o obj/des.o obj/aes.o obj/desfire_key.o obj/desfire_crypto.o obj/mifaredesfire.o obj/iso14443crc.o obj/crc.o obj/crc16.o obj/crc32.o obj/iclass.o obj/optimized_cipher.o obj/tlv.o obj/emvdataels.o obj/emvutil.o obj/emvcmd.o obj/parity.o obj/usb_cdc.o obj/cmd.o -lgcc
obj/emvcmd.o: In function `EMVSelectPPSE':
emvcmd.c:(.text+0x234): undefined reference to `emv_selectPPSE'
collect2: error: ld returned 1 exit status
make[1]: *** [obj/fullimage.stage1.elf] Error 1
make: *** [armsrc/all] Error 2

is there already a way to fix this?

thx

SpaceTeddy, 2017-09-17:
any news here?

iceman1001, 2017-09-17:
not really, since I don't run OSX anywhere.   Haven't seen much in TravisCI either but thats because EMV is disabled. 
I pushed a minor fix today. Making some functions extern.  Test it?

SpaceTeddy, 2017-09-18:
still the same error:
obj/emvcmd.o: In function `EMVSelectPPSE':
emvcmd.c:(.text+0x234): undefined reference to `emv_selectPPSE'
collect2: error: ld returned 1 exit status
make[1]: *** [obj/fullimage.stage1.elf] Error 1
make: *** [armsrc/all] Error 2

but it occours not only in OSX. Also in my Ubuntu 16.04 Virtual Box.

iceman1001, 2017-09-18:
fixed: https://github.com/iceman1001/proxmark3/commit/3839ce006d9063bd0feade048348465318e01dcf

closed 2017-09-18 by iceman1001

mastahofdesastah, 2017-09-19:
Hi Iceman thx for working on it.

Despite it compiles, i get "unknown command:: 0x070X errors" on all commands except "transaction". And this command seems to be broke, since it returns no data. It was working in earlier stages...