iceman1001/proxmark3 #98: Lua hard script

MaBi6411, 2017-04-07:
I can't use lua _hard script_. It hasn't responses for every block: [https://pastebin.com/Fr1gasZ7](url).
But when I use single command (_hardnested_) it works well: [https://pastebin.com/qjjceq06](url).
Which is the problem?

iceman1001, 2017-04-08:
Great question. 
I also noticed this when I created the script. Thought it was related to memleaks when running the cmd several Times in a row

MaBi6411, 2017-04-08:
Oh yes. It happens with _core.hardnested_ function...
I have tried to edit script using string command with _core.console_ (_local cmdHardNested = "hf mf hardnested 0 A "..key.." "..tostring(trgblockno).." A"_) and loop "works" without reporting values in final table.

Why does hardnested command seem to ignore key arguments?
`pm3 --> hf mf hardnested 0 A a0a1a2a3a4a5 4 A
--target block no:  4, target key type:A, known target key: 0x000000000000 (not
set), file action: none, Slow: No, Tests: 0`

iceman1001, 2017-04-08:
Indeed. I needed to implement the core.hardnested to get the found key. The loop with core.console will work. 

The key is in the details,  the output refers to a targetkey when you know it. Not the key used to run hardnested.  Try calling hardnested with that option and you will see the output matches.

iceman1001, 2017-04-08:
I'll have a look at it in a week or two. I'm out of office.

MaBi6411, 2017-04-13:
iceman, there's no rush. I can only say thank you

iceman1001, 2017-06-12:
This version of the script is outdated,  since @piwi released an improved version of the hardnested attack implementation.

closed 2017-06-12 by iceman1001