iceman1001/proxmark3 #99: How to flash Iceman-Fork

ikarus23, 2017-04-10:
I'm feeling kind of stupid, but how do I flash this fork on my PM3? I was not able to find a hint in the Readme or the Issue Tracker. Did I miss something? I'm getting the error message:
```
$ ./client/flasher /dev/ttyACM3 armsrc/obj/fullimage.elf
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00030d20->0x00030d20) [R X] @0x94
1: V 0x00200000 P 0x00132d20 (0x000019e4->0x000019e4) [RW ] @0x30db4
Note: Extending previous segment from 0x30d20 to 0x32704 bytes

Waiting for Proxmark to appear on /dev/ttyACM3. Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to reappear on /dev/ttyACM3...... Found.
Error: Unexpected reply 0x00fe NACK (expected ACK)
```
There is the same unexpected reply for trying to flash the bootloader.

osysltd, 2017-04-10:
Hi Ikarus,
try `killall ModemManager`

ikarus23, 2017-04-10:
Hi @osysltd, sorry, I forgot to mention an important thing: Everything is working fine with the "original" PM3 repository. This issue only appears at the Iceman-Fork.

I double checked: without `ModemManager` running I still have the same issue.

iceman1001, 2017-04-10:
Try not to flash bootrom. 
Or you have to comment out the HAS_512 etc lines in makefile. 

There have been some issues w this. 
Also thread on forum. 

Best regards
Christian Herrmann

Company: www.icesql.net
Mobile: 0722-208110

ikarus23, 2017-04-10:
Both of my PM3 have the latest bootrom. I also tried to flash only the `fullimage.elf` Futhermore, do I have to comment out these `HAS_512` lines even when I'm using a PM3 with 512k memory?

ikarus23, 2017-04-10:
OK, found the [forum thread](http://www.proxmark.org/forum/viewtopic.php?id=4668). I will try these fixes/workarounds.

ikarus23, 2017-04-10:
Ok, without the `HAS_512` everything works like a charm.

closed 2017-04-10 by ikarus23