iceman1001/proxmark3 #100: Crash on hf legic info

ikarus23, 2017-04-11:
The client (and/or the PM3?) crashes on executing `hf legic info`:
(Some data has been replaced by XX for privacy reasons)
```
pm3 --> hf legic reader
 UID : 41 XX XX XX           
TYPE : MIM1024 card (1002 bytes)          
pm3 --> hf legic info
TYPE : MIM1024 card (1002 bytes)          

 ##  |  0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F          
-----+------------------------------------------------------------------------------------------------          
[00] | XX...
[01] | XX...

CDF: System Area          
------------------------------------------------------          
MCD: 41, MSN: XX XX XX, MCC: b7 OK          
DCF: 60000 (60 ea), Token Type=IM-S (OLE=0)          
WRP=15, WRC=1, RD=1, SSC=ff          
Remaining Header Area          
00 00 00 XX XX XX XX XX XX XX XX XX 00           

ADF: User Area          
------------------------------------------------------          
Segment 01 
raw header | XX... 
Segment len: 22,  Flag: 0x4 (valid:1, last:0), WRP: 10, WRC: 03, RD: 0, CRC: 0x11 (OK)          
WRC protected area:   (I 27 | K 0| WRC 3)          

row  | data          
-----+------------------------------------------------          
[00] | XX XX XX 
Remaining write protected area:  (I 30 | K 30 | WRC 3 | WRP 10  WRP_LEN 7)          

row  | data          
-----+------------------------------------------------          
[00] | XX XX XX XX XX XX XX
Remaining segment payload:  (I 37 | K 37 | Remain LEN 7)          

row  | data          
-----+------------------------------------------------          
[00] | B7 B7 B7 B7 B7 B7 B7 
-----+------------------------------------------------
          
Segment 02 
raw header | 0xB7 0xB7 0xB7 0xB7 
Segment len: 1975,  Flag: 0xB (valid:0, last:1), WRP: 183, WRC: 03, RD: 1, CRC: 0xB7 (fail)          
WRC protected area:   (I 49 | K 44| WRC 3)          

row  | data          
-----+------------------------------------------------          
[00] | B7 B7 B7 
Remaining write protected area:  (I 52 | K 52 | WRC 3 | WRP 183  WRP_LEN 180)          

row  | data          
-----+------------------------------------------------          
[00] | B7 B7 B7 B7 B7 B7 B7 B7 B7 B7 B7 B7 B7 B7 B7 B7 
...
[110] | XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX
[111] | XX XX XX XX XX XX XX XX XX XX XX
-----+------------------------------------------------
          
[1]    2028 bus error (core dumped)  ./client/proxmark3 /dev/ttyACM1
```

ikarus23, 2017-04-11:
More information:
* It does not crash for all tags. For some contents/tags it works.
* `hf legic info` of the "official" PM3 repository displays different information (maybe some length fields are interpreted in a different way?)

ikarus23, 2017-04-11:
This issue is already known from the PM3 forum: http://www.proxmark.org/forum/viewtopic.php?id=3889

iceman1001, 2017-04-11:
yeah,  the info command doesnt work yet,  see thread for reasons,
but the main problem was a way of identifying the tag to know how much memory should be allocated on client. This command can be fixed with some attention

iceman1001, 2017-04-15:
@ikarus23  will you fix this one?  Looking forward for a PR

ikarus23, 2017-04-15:
I will look into it. But no promises.

iceman1001, 2017-04-15:
for this command to work, it need to get the current tagtype, you know how large tagmemory the info command needs to reserve and iterate.  thats why I started the **hf legic reader**  that usbcmd should return a usbcmd with tagtype.  So **hf legic info**  needs to call this one first.

then if you want an offline mode (ie load a tag dump) , it can be based on filesize. 
With this piece of information you have all you need to fix the loops inside **hf legic info** for it not to crash anymore.

ikarus23, 2017-04-17:
Ok, here is something else broken. I think it's `hf legic reader`. The "official" repository reads segment 0 of the tag as follows:
```
Segment 00: raw header=16 40 0a 30, flag=4 (valid=1, last=0), len=0022, WRP=10, WRC=03, RD=0, CRC=af
WRC protected area:
fb 0a 01
Remaining write protected area:
02 02 00 67 58 01 6d
Remaining segment payload:
00 00 00 00 00 00 00
```
As far as I can tell, this is correct. The "iceman-fork" decodes the data as
```
Segment 01 
raw header | 0x16 0x40 0x0A 0x30 
Segment len: 22,  Flag: 0x4 (valid:1, last:0), WRP: 10, WRC: 03, RD: 0, CRC: 0xAF (OK)          

WRC protected area:   (I 27 | K 0| WRC 3)          
row  | data          
-----+------------------------------------------------          
[00] | FB 0A 01 

Remaining write protected area:  (I 30 | K 30 | WRC 3 | WRP 10  WRP_LEN 7)          
row  | data          
-----+------------------------------------------------          
[00] | 02 02 00 67 CE CE CE 

Remaining segment payload:  (I 37 | K 37 | Remain LEN 7)          
row  | data          
-----+------------------------------------------------          
[00] | CE CE CE CE CE CE CE 
-----+------------------------------------------------
```
It seams the decoding breaks in the middle of "Remaining write protected area". However, at a first glimpse, I can't see any error in the decoding code. Therefore I believe the `hf legic reader` must be broken somewhere.

iceman1001, 2017-04-19:
hf legic reader works as it should.
hf legic info only reads 22 bytes of data, but downloads 1024bytes. 
I'll push some fixes for it soon.

iceman1001, 2017-04-21:
I have fixed the problems,  it was how device EML was used and not being downloaded as it should.

ikarus23, 2017-04-21:
Thanks. Could you push the fix to the master?

iceman1001, 2017-07-31:
this one should be fixed now.  
closing.

closed 2017-07-31 by iceman1001