iceman1001/proxmark3 #113: Spanish ID 3.0

joanbono, 2017-06-20:
Hi all!

I'm from Spain and I own a new **Spanish ID Card**, which uses NFC.

So reading the specifications, it uses a `CAN` (Card Access Number) to read the owner information (Name, Surname, Photo, Birth Date and so on). So there's a photo of the card:

[image]

In the photo, at the right-bottom corner, the `CAN` appears (`123456`). I was testing with my PM3 and I was wondering if there is a command to extract this information.

***

### `hf search` 

I got really confused when I did the `hf search` command. The output gives my a different `UID` each time 😕 

~~~~
pm3 --> hf search

 UID    : E4 35 02 07
 ATQB   : E1 F3 5E 11 77 81 A1
 CHIPID : 00
      App Data: E1 F3 5E 11
      Protocol: 77 81 A1
      Bit Rate: 212 kbit/s PICC -> PCD supported
      Bit Rate: 424 kbit/s PICC -> PCD supported
      Bit Rate: 847 kbit/s PICC -> PCD supported
      Bit Rate: 212 kbit/s PICC <- PCD supported
      Bit Rate: 424 kbit/s PICC <- PCD supported
      Bit Rate: 847 kbit/s PICC <- PCD supported
Max Frame Size: 256 bytes
 Protocol Type: Protocol is compliant with ISO/IEC 14443-4
Frame Wait Integer: 10 - 32768 ETUs | 309248 us
 App Data Code: Application is Proprietary
 Frame Options: NAD is not supported
 Frame Options: CID is supported
Tag :
  Max Buf Length: 0 (MBLI) chained frames not supported
  CDI : 0

Valid ISO14443-B Tag Found - Quiting Search
~~~~ 

And next time:

~~~~
pm3 --> hf search

 UID    : EA B9 5D 95
 ATQB   : E1 F3 5E 11 77 81 A1
 CHIPID : 00
      App Data: E1 F3 5E 11
      Protocol: 77 81 A1
      Bit Rate: 212 kbit/s PICC -> PCD supported
      Bit Rate: 424 kbit/s PICC -> PCD supported
      Bit Rate: 847 kbit/s PICC -> PCD supported
      Bit Rate: 212 kbit/s PICC <- PCD supported
      Bit Rate: 424 kbit/s PICC <- PCD supported
      Bit Rate: 847 kbit/s PICC <- PCD supported
Max Frame Size: 256 bytes
 Protocol Type: Protocol is compliant with ISO/IEC 14443-4
Frame Wait Integer: 10 - 32768 ETUs | 309248 us
 App Data Code: Application is Proprietary
 Frame Options: NAD is not supported
 Frame Options: CID is supported
Tag :
  Max Buf Length: 0 (MBLI) chained frames not supported
  CDI : 0

Valid ISO14443-B Tag Found - Quiting Search
~~~~

So it will be really nice to implement something to "attack" the card, or even a Lua script. I'll work on it, but I need some help and orientation.

Regards!! 😃

iceman1001, 2017-06-20:
try **hf 14b** commands but use the PM3 Master instead of icemanfork.  You might be able to read the tag data.

joanbono, 2017-06-20:
No answer using the `proxmark3-master` firmware. The output is

~~~~
proxmark3> hf 14b reader
no 14443B tag found
~~~~

Reflashing to the `icemanfork` the output is:

~~~~
pm3 --> hf 14b reader
 UID    : 16 D3 DC 10
 ATQB   : E1 F3 5E 11 77 81 A1
 CHIPID : 00
      App Data: E1 F3 5E 11
      Protocol: 77 81 A1
      Bit Rate: 212 kbit/s PICC -> PCD supported
      Bit Rate: 424 kbit/s PICC -> PCD supported
      Bit Rate: 847 kbit/s PICC -> PCD supported
      Bit Rate: 212 kbit/s PICC <- PCD supported
      Bit Rate: 424 kbit/s PICC <- PCD supported
      Bit Rate: 847 kbit/s PICC <- PCD supported
Max Frame Size: 256 bytes
 Protocol Type: Protocol is compliant with ISO/IEC 14443-4
Frame Wait Integer: 10 - 32768 ETUs | 309248 us
 App Data Code: Application is Proprietary
 Frame Options: NAD is not supported
 Frame Options: CID is supported
Tag :
  Max Buf Length: 0 (MBLI) chained frames not supported
  CDI : 0
~~~~ 

So, it's quite confusing because the `hf search` in the `proxmark3-master`:

~~~~
proxmark3> hf search
no known/supported 13.56 MHz tags found
~~~~

But it works like I charm when I insert a blank chinese card instead the Spanish ID.
The `hf search` in the `icemanfork` works as shown in the first message.

iceman1001, 2017-06-20:
did you try reading your tag?

joanbono, 2017-06-20:
Sure! With the master fork, and says that there's no tag: **hf 14b reader**.

iceman1001, 2017-06-20:
No, I didn't mean the "hf 14b reader"..  there is some other read commands which tries to read tag memory

osysltd, 2017-06-21:
@joanbono you can try to send raw commands to the tag according to 

> So reading the specifications, it uses a CAN (Card Access Number) to read the owner information (Name, Surname, Photo, Birth Date and so on).

By the way, just out of curiosity, please can you share CAN specification?

joanbono, 2017-06-21:
Hi @iceman1001 , I don't get which command are you refering to... Sorry

***

@osysltd I've tried:

~~~~
pm3 --> hf 14b raw 394496
pm3 --> hf list 14b
Recorded Activity (TraceLen = 12 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |       4732 | Rdr |39  44  96                                                       | !crc| ?
~~~~

About the `CAN`, from the [Spanish Police Webpage](https://www.dnie.es/PortalDNIe/PRF1_Cons02.action?pag=REF_103&id_menu=[1]):

> The `CAN` is a random six number digit which changes in every card. It serves as proof of possession of the DNI 3.0 when used with NFC technology.

There are 46 million people living in Spain, and there are only 1 million of possible `CAN` numbers...

iceman1001, 2017-06-21:
as a suggestion,  you should read up on iso14443b protocol,   then learn how to use the  "hf 14b raw" to send the right command with your CAN number.  I'm curious of this CAN specification.

Somewhere in the specifications is the process of getting the data and how to process the data hidden. 
Find it!

joanbono, 2017-07-04:
**UPDATE** 

I'm waiting answer from the *tech-department*, which has all the information about this card. I asked them for papers and implementations, and also how to take the stored information using the `CAN`.

iceman1001, 2017-07-16:
Did your tech-department come back with some relevant information for you?

joanbono, 2017-07-16:
Still waiting response. I'll ping them again tomorrow in the morning.

I'll keep you updated here.

osysltd, 2017-07-16:
@joanbono if you have an ability to proceed with some transactions using your card with proxmark nearby, you can try to sniff communication and get more details meanwhile

joanbono, 2017-07-17:
@osysltd I'll try, but it's an official document which only has use in webpages using the Smart-card chip. NFC is not fully implemented at the moment to be used to complete official documentation.

joanbono, 2017-08-29:
I've been contacted by the Spanish police.

They say that they can't give me any information about the ID which is not published on their website.

I think I'm going to abort this mission, and perform a "black box" audit to the card 😆 

Stay tuned.

iceman1001, 2017-08-29:
Try to find the manufacturer. Also search for academic papers.

iceman1001, 2017-10-29:
How does the **hf search**  and **hf 14b** commands work now with @pwpiwi 's fixes`?

Time to close?

joanbono, 2017-12-01:
Hi, yes, it works, but with `hf search`:

~~~
pm3 --> hf search
Valid ISO14443-A Tag Found - Quiting Search
~~~

And then:

~~~
pm3 --> hf 14a reader
iso14443a card select failed
~~~

Finally:

~~~
pm3 --> hf 14b reader
 UID    : 43 E6 26 09
 ATQB   : E1 F3 5E 11 77 81 A1
 CHIPID : 00
      App Data: E1 F3 5E 11
      Protocol: 77 81 A1
      Bit Rate: 212 kbit/s PICC -> PCD supported
      Bit Rate: 424 kbit/s PICC -> PCD supported
      Bit Rate: 847 kbit/s PICC -> PCD supported
      Bit Rate: 212 kbit/s PICC <- PCD supported
      Bit Rate: 424 kbit/s PICC <- PCD supported
      Bit Rate: 847 kbit/s PICC <- PCD supported
Max Frame Size: 256 bytes
 Protocol Type: Protocol is compliant with ISO/IEC 14443-4
Frame Wait Integer: 10 - 32768 ETUs | 309248 us
 App Data Code: Application is Proprietary
 Frame Options: NAD is not supported
 Frame Options: CID is supported
Tag :
  Max Buf Length: 0 (MBLI) chained frames not supported
  CDI : 0
~~~

So, the card is not well detected by `hf search`.

iceman1001, 2018-09-18:
@joanbono   ping.  Maybe I suggest you try out the RRG repo

joanbono, 2018-09-19:
Hi @iceman1001 !

I will, my pm3 Rv4 is in my hometown and I'm living in Czech Republic, waiting my parent's visit in two weeks. They will bring the device here! Can't wait!

iceman1001, 2018-09-19:
yeah,  there is some interesting stuff going on over at official pm3,  with fpga firmware enhancements. 
which 14b, 15 etc benefits from