iceman1001/proxmark3 #114: [enhancement] Scan proxmark3.log for HID cards and try them against reader

samturner3, 2017-06-20:
### User Story: 
Today I wanted to get back into a building that I had previously used  ```lf search``` on a valid HID card for. I went to my logs but I had not enabled date logging so it was impossible to know which result was the correct one.

### Solution:
A script that scans through proxmark3.log looking for ```"HID Prox TAG ID:"``` and trying all the card numbers found similar to ```lf hid brute```

### Work done so far:
[https://github.com/samturner3/proxmark3/tree/HIDBruteFromLog](url)

created ```pullHID.c``` that extracts the FC and card number from proxmark3.log.
[image]
[image]

### To Do
Fix up line 83 to further separate / get rid of '- Card'
Implement into a command to run the actual attempt (```sendTry(fmtlen, fc, cn, delay, bs)```)

## About
Any help would be great / feedback if you think this would be useful? I have hardly ever coded in c. ALso help would be appreciated as I don't have too much spare time these days.

iceman1001, 2017-06-20:
hm,  why do you use c?   Why no make a simple lua script?  or python? 
The functionality is 
- read textfile
- grep "hid prox tag id"  (or custom filter text)
- print found rows
- close file

What not even use grep? 
** grep "key found" *.log **

iceman1001, 2017-06-20:
Getting the intersting unique row values from the log file.   Put it in a shell-file.

>  grep "HID Prox TAG ID" *.log | awk '{ printf "Raw %4s | FC %4s %4s\n",$5, $13,$16 }' | sort | uniq

if you want to add this into a file with value pm3 commands for hid sim?  or something else,  then you modify the above accordingly.

if you want to run it from inside pm3-client,  just make a lua script which runs the shell file,  then opens your newly created file and sends row by row to core.console.   

you only need some imagination for these things.

iceman1001, 2017-07-04:
close?

closed 2017-07-16 by iceman1001