iceman1001/proxmark3 #118: How to simulate Mifare Ultralight EV1?

BlackPaaanda, 2017-08-02:
Hi, iceman

I've dumped the data of a ultralight EV1 (48 bytes) tag (hopefully it is the data) using "hf mfu dump" and i got 32*7 characters table by using "pm3_mfd2eml.py" function. However, when i try to use "hf mfu eload" to load this file to pm3's emulator memory, it said: ............File reading error. Did i do something wrong or how actually should i simulate the ultralight EV1 tag?

iceman1001, 2017-08-02:
I know, its a known quirk   
The dump file needs to be 255 blocks... 
just add the extra needed empty blocks and it should Eload without problems.

BlackPaaanda, 2017-08-02:
Does this means that the data table should be 32*255 with all extra blocks be empty?

BlackPaaanda, 2017-08-03:
Hi iceman,

As i tried to modify the dump file, the proxmark3 can simulate the MF0UL11 tag.
However, i checked the .eml data, found that the first 4 blocks are all 0. Number starts to change at block 4 and it matches the card finger prints. What does that mean?

iceman1001, 2017-08-04:
that has to do with the dump-file-format adds "special" data the first 48bytes. Afterwards comes the normal datablock.  I'm unsure if the eload/simulation takes this in consideration.  it should but I'ven't looked at it for a long time.

iceman1001, 2017-08-09:
close ?

BlackPaaanda, 2017-08-09:
Thanks a lot iceman. You can close this

closed 2017-08-09 by iceman1001