iceman1001/proxmark3 #123: GCC version detection issue in client/Makefile

winguru, 2017-08-23:
Iceman,

It appears that some versions of GCC only report the major version when specifying -dumpversion.
This output is used in the following Makefile to test for versions of GCC >= 4.9.0 to disable AVX512 support:
https://github.com/iceman1001/proxmark3/blob/6598e12752aceeb0a02a22a1c8e9c17e5552cdf6/client/Makefile#L200

This was causing a compile error in my Kali build environment.

Newer versions of GCC support the "-dumpfullversion" to return the major and minor build numbers (e.g. 7.1.0), or you can use the "--version" to show the full info.

Here is an alternate way of getting the full version info into the GCC_GTEQ_490 variable:
<code>GCC_GTEQ_490 := $(shell expr `gcc --version | awk '/gcc /{print $$NF;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/'` \>= 40900)</code>

(Sorry, Markdown is removing the backticks around the "expr" statement above)

Let me know if you'd like more info, clarification, or anything else to help with this...

Thank you for all of you hard work on this project!!

--Geoff

iceman1001, 2017-08-23:
Hi  Geoff a.k.a Winguru!
Great feedback, this might explain some issues about crashing 'hardnested' I've gotten lately.

Why don't you make a PR?

iceman1001, 2017-08-23:
it looks like GitHub changed some more,   
/gcc /{print $$NF;}
should be:
/gcc/{print $NF;}

winguru, 2017-08-23:
Iceman,

I'm making the PR now... and yes, you would be correct about awk and $NF, but because of the Makefile variable expansion, to make a literal '$', you need to use '$$'.  Weird, I know...

--Geoff

iceman1001, 2017-08-23:
aha, I see,  I was testing it on the commandline..   The extra space /gcc /  needs to be removed ..
I did a push already :)...

iceman1001, 2017-08-23:
hm, setback...  it fails in mingw.

winguru, 2017-08-23:
Let me see if I can create a more generic approach to detecting the GCC version.  I don't normally use MinGW but I can quickly set up a test environment,

iceman1001, 2017-08-23:
Don't! 
I found the missing char,  it works fine now.  Try pull and test it

winguru, 2017-08-23:
Yup works fine!  I see Markdown stripped out the "\\" in the "\\>=" expression check.

Thanks again!!

closed 2017-08-23 by iceman1001