iceman1001/proxmark3 #124: Problem when "hf mf cload"

BlackPaaanda, 2017-08-28:
Hi iceman,

I found an issue when using command "hf mf cload". I've done everything before this command correctly and now i got a .eml file contains everything of a mifare classic 1K card. However, when i try to write a magic card, it says "File content error. Block data must include 32 HEX symbols". What's the problem of this?

Cheers.

iceman1001, 2017-08-28:
too little information to be able to answer your question.
Have you looked at you .eml file and verfied that it has correct amount of lines?
The message says you have a line of data which is not 32 hex symbols...

BlackPaaanda, 2017-08-28:
I successfully loaded the file to the emulator memory and use "hf mf cload
e" to write the magic card. I think this means the file is correct. The
only problem is I can't do "hf mf cload" from the file directly.

iceman1001, 2017-08-28:
Still too little information to go on. It would help to understand if you,

- mention which commands did you run when you encountered this problem
- are you running the latest source from github
- what does your .eml look like?  (number of rows etc)
- which commands did you run to solved your problem

_Just saying that it doesn't work,  is not useful in understanding a potential problem._

BlackPaaanda, 2017-08-28:
Hi,

I have the .eml file for a standard mifare 1K card, which has 64 rows with
32 HEX symbols per row. For example, the file name is "George.eml".  I
tried to use

 <hf mf cload George>

to write the data to a magic card. However, it said
<File content error. Block data must include 32 HEX symbols>.

Then I used

<hf mf eload George>

to load the file to the memory and it worked. Then i used

 <hf mf cload e>

to write the magic card and succeed. I just wonder why the command cannot
use the file directly.

The source is the latest from your github.

Appreciate if you can answer my question and thanks for you patience.

iceman1001, 2017-08-28:
Ok,  I think I found the problem.  I've pushed a fix,  try pulling latest code and test.

BlackPaaanda, 2017-08-28:
Hi,

I used <git pull> command and updated the code. However, it still says
"File content error. Block data must include 32 HEX symbols"

iceman1001, 2017-08-28:
...and you did compile it aswell...  hm.

BlackPaaanda, 2017-08-28:
Ok....I installed the new client from your source. Now if i run the <hf mf
cload> command, sometimes it says "Can't set magic card bloack 0" sometimes
it says "File content error. Block data must include 32 HEX symbols". But
never succeed......I am pretty sure that the card i used is magic card. Now
i'm confused.....

iceman1001, 2017-08-28:
"can't set magic card block0",  well,   you need to have a magic card gen1 on the antenna when running.

the file content error,  shouldn't say anymore.  Your .eml file seems be problematic.

iceman1001, 2017-08-28:
ok, I tried it on ubuntu and have same issues.  Something is wrong.  I'm guessing its the newline \n\r chars spooking.

dibops, 2017-08-28:
Fix for hf mf cload.

client/cmdhfmf.c
Change line 2146 from "char buf[33] = {0x00};  // 32+1 null terminator" to "char buf[256] = {0x00};"

tested and confirmed working.

iceman1001, 2017-08-28:
Thanks but the char buffer doesn't need to be 256. The reason for this bug is that I changed it from 64 down to 32,  I forgot about newline chars also take space when I did this. 

I pushed fixes and made the output a bit more responsive.

dibops, 2017-08-28:
Confirmed fixed.

closed 2017-08-28 by iceman1001