iceman1001/proxmark3 #125: HF search detection issue

idaoudi, 2017-08-31:
Hi Iceman !

So I have this doorlock which works with an RFID tag https://www.thequicklock.com/product-doorlock.php

It seems your fork can't detect it with the `hf search` command, this is the result of the command using your fork's firmware with your fork's software (the result is the same using your firmware and master software):
```
pm3 --> hf search
          
UART:: write time-out
Sending bytes to proxmark failed          
UART:: write time-out
Sending bytes to proxmark failed          

no known/supported 13.56 MHz tags found
```

However the master firmware does detect a valid ISO15693 tag, here is the output with the latest commit:
```
proxmark3> hf search
          
Tag UID : E004015036F15CD3          
Tag Info: NXP(Philips); IC SL2 ICS20/ICS21(SLI) ICS2002/ICS2102(SLIX)          

Valid ISO15693 Tag Found - Quiting Search
```

Any idea ?

iceman1001, 2017-08-31:
Interesting, the ISo15693 detection is bad.  
Have you tried running the **hf 15** commands manually?

idaoudi, 2017-08-31:
Still doesnt detect the tag:
```
pm3 --> hf 15 cmd inquiry
Response to short, just 0 bytes. No tag?
          
pm3 --> hf 15 cmd sysinfo -2 u
timeout while waiting for reply.          
pm3 --> hf 15 cmd sysinfo -2 *
No Tag found          
timeout while waiting for reply.          
pm3 --> hf 15 cmd sysinfo u
timeout while waiting for reply.          
pm3 --> hf 15 cmd sysinfo *
No Tag found          
timeout while waiting for reply.  
```

And still works with master firmware:
```
proxmark3> hf 15 cmd sysinfo -2 u
0F D3 5C F1 36 50 01 04 E0 00 00 1B 03 01 
UID = E004015036F15CD3
NXP(Philips); IC SL2 ICS20/ICS21(SLI) ICS2002/ICS2102(SLIX)
DSFID supported, set to 00
AFI supported, set to 000
Tag provides info on memory layout (vendor dependent)
 4 (or 3) bytes/page x 28 pages 
IC reference given: 01
```

iceman1001, 2017-08-31:
**Antenna, decent voltage**
# HF antenna: 15.17 V @    13.56 MHz

**running latest**
`
Proxmark3 RFID instrument

 [ ARM ]
 bootrom: iceman/master/v1.7.1-140-g7810dac6-dirty-unclean 2017-08-22 11:19:21
      os: iceman/master/v1.7.1-174-g4d354f75-dirty-unclean 2017-08-31 12:31:56
 [ FPGA ]
 LF image built for 2s30vq100 on 2015/03/06 at 07:38:04
 HF image built for 2s30vq100 on 2017/05/17 at 17:48:26

 [ Hardware ]
  --= uC: AT91SAM7S256 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 216450 bytes (83%) Free: 45694 bytes (17%
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory`

pm3 --> hf search

Tag UID : E007C11056F77E36
Tag Info: Texas Instrument; Tag-it HF-I Standard; 8x32bit

Valid ISO15693 Tag Found - Quiting Search

pm3 --> hf 15 cmd sysinfo -2
missing addr
timeout while waiting for reply.

pm3 --> hf 15 cmd sysinfo *
Detected UID E007C11056F77E36
timeout while waiting for reply.

pm3 --> hf 15 cmd inquiry
UID=E007C11056F77E36
Tag Info: Texas Instrument; Tag-it HF-I Standard; 8x32bit

It still works for me.  A bit itchy about positions.

iceman1001, 2017-08-31:
I've pushed some minor changes.  Mostly when things are init and moved big_buff allocations to after inits.

Try and see if it got any better.
And if you could paste (hw version,  hw tune)  output,  so I know what your setup looks like.

idaoudi, 2017-08-31:
Still nothing after a `git pull` and `make clean && make all`, flashing bootrom & os, and trying the search and cmd with different positions..

Here is what you asked:
```
pm3 --> hw ver
[[[ Cached information ]]]
          

Proxmark3 RFID instrument
          
 [ ARM ]
 bootrom: iceman/master/v1.1.0-2227-geec5780b 2017-08-31 13:49:09
      os: iceman/master/v1.1.0-2227-geec5780b 2017-08-31 13:49:12
 [ FPGA ]
 LF image built for 2s30vq100 on 2015/03/06 at 07:38:04
 HF image built for 2s30vq100 on 2017/05/17 at 17:48:26
          
 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 216927 bytes (41%) Free: 307361 bytes (59%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          
          

pm3 --> hw tune

Measuring antenna characteristics, please wait......          
# LF antenna: 44,55 V @   125.00 kHz          
# LF antenna: 19,94 V @   134.00 kHz          
# LF optimal: 45,51 V @   123,71 kHz          
# HF antenna: 30,66 V @    13.56 MHz          
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

```

iceman1001, 2017-08-31:
That looks ok. 

Have you tried different positions / distances over the antenna & tag?

idaoudi, 2017-08-31:
Yes I think I have tried every possible position
I checked with `hf tune` and it goes from 30V~ to 21V~ when I bring the tag closer, so I think it does see something

And with the original master's firmware I just put the tag on the antenna and it is detected right away, with the same hardware

iceman1001, 2017-09-04:
ok, I've given **hf 15**  on both client side and device side some love. 

Would you @idaoudi  mind testing it out now?
https://github.com/iceman1001/proxmark3/commit/ec07e2e0067adf4948250ef9e4b8cc1e00f8b905

iceman1001, 2017-09-05:
And now it starts to look quite nice.  But I still miss some testing by @idaoudi 

I did a recap of the changes here;  https://www.youtube.com/watch?v=f_vHhmFXDTA

idaoudi, 2017-09-07:
I tested with your latest commit, unfortunately the tag still isn't detected. Here is a trace with debugging enabled:

2 successive `hf search`:
```
pm3 --> hf search
          
UART:: write time-out
Sending bytes to proxmark failed          
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
#db# RECV          
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
#db# ice: demod bytes 0          
#db# error, uneven octet! (extra bits!) mask 02          
timeout while waiting for reply.          
#db# RECV          
UART:: write time-out
Sending bytes to proxmark failed          

no known/supported 13.56 MHz tags found
          
pm3 --> hf search
          
UART:: write time-out
Sending bytes to proxmark failed          
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
#db# RECV          
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
timeout while waiting for reply.          
#db# RECV          
UART:: write time-out
Sending bytes to proxmark failed          

no known/supported 13.56 MHz tags found          

```

hf 15 info & hf 15 read:
```
pm3 --> hf 15 info u
#db# SEND          
#db# .+&.     02 2b 26 a3          
#db# Iso15693InitReader Exit          
#db# RECV          
iso15693 card doesn't answer to systeminfo command          

pm3 --> hf 15 info *
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
#db# RECV          
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
#db# RECV          
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
timeout while waiting for reply.          
No tag found          
#db# RECV          

pm3 --> hf 15 info -2 u
#db# SEND          
#db# .+&.     02 2b 26 a3          
#db# Iso15693InitReader Exit          
iso15693 card doesn't answer to systeminfo command          
#db# RECV          
          
pm3 --> hf 15 info -2 *
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
#db# RECV          
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
#db# RECV          
#db# SEND          
#db# &....    26 01 00 f6 0a          
#db# Iso15693InitReader Exit          
timeout while waiting for reply.          
No tag found          
#db# RECV 

pm3 --> hf 15 read * 0
timeout while waiting for reply.          
No tag found          

pm3 --> hf 15 read * 1
timeout while waiting for reply.          
No tag found          

pm3 --> hf 15 read u 0
iso15693 card select failed          

```

I also tested the new sim command, with the UID I obtained using the master firmware, and here is a trace:
```
pm3 --> hf 15 sim E004015036F15CD3
Starting simulating UID E0 04 01 50 36 F1 5C D3           
#db# ISO-15963 Simulating uid: E004015036F15CD3          
#db# 1 octets read from reader command: aa 0 0 0 0 0 0 0 0          
#db# 2 octets read from reader command: 72 54 0 0 0 0 0 0 0          
#db# 1 octets read from reader command: f5 54 0 0 0 0 0 0 0          
#db# 2 octets read from reader command: 72 54 0 0 0 0 0 0 0          
#db# 2 octets read from reader command: 62 55 0 0 0 0 0 0 0          
#db# 2 octets read from reader command: 72 54 0 0 0 0 0 0 0          
#db# 1 octets read from reader command: ba 54 0 0 0 0 0 0 0          
#db# 1 octets read from reader command: 62 54 0 0 0 0 0 0 0          
#db# 1 octets read from reader command: 62 54 0 0 0 0 0 0 0          
#db# 2 octets read from reader command: 72 54 0 0 0 0 0 0 0          
pm3 --> hf list 15
Recorded Activity (TraceLen = 104 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
ISO15693 - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|    
```

I don't know if it's a normal behaviour, but the read data from reader don't appear in `hf list 15̀`
Also the lock doesn't open with just that, so I suppose the UID is not enough, but it has nothing to do with the issue I suppose

iceman1001, 2017-09-07:
Nice,  your output from **hw status** would also be nice to have. 

This message is strange
```
UART:: write time-out
Sending bytes to proxmark failed    
```

Try running without debug on,  (hf 15 debug 0),  since I can clearly see the pm3 is sending the command.
```
hf 15 debug 0
hf 15 reader 
hf list 15
```

The simulation,
did you try simulating against a reader or against another proxmark3?
and don't use debug when doing sim,  that will screw up the timings very much.
At least we are seeing some kind of data in that case.

idaoudi, 2017-09-07:
Here it is:
```
pm3 --> hw status
#db# Memory          
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing          
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Fgpa          
#db#   mode....................HF          
#db# LF Sampling config:           
#db#   [q] divisor..............95 (125 KHz)          
#db#   [b] bps..................8          
#db#   [d] decimation...........1          
#db#   [a] averaging............Yes          
#db#   [t] trigger threshold....0          
#db# USB Speed:          
#db#   Sending USB packets to client...          
#db#   Time elapsed............1500ms          
#db#   Bytes transferred.......86016          
#db#   USB Transfer Speed PM3 -> Client = 57344 Bytes/s          
#db# Various          
#db#   MF_DBGLEVEL.............2          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308  

pm3 --> hf 15 reader
timeout while waiting for reply.          
No Tag found.          
pm3 --> hf 15 list
Recorded Activity (TraceLen = 23 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
ISO15693 - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          0 |      58432 | Rdr |26  01  00  f6  0a                                               |  ok | INVENTORY          

```

I tried the simulation against the reader on the lock itself, here is the result without debug:
```
pm3 --> hf 15 sim E004015036F15CD3
Starting simulating UID E0 04 01 50 36 F1 5C D3           
#db# ISO-15963 Simulating uid: E004015036F15CD3          
#db# 2 octets read from reader command: 62 54 0 0 0 0 0 0 0          
#db# 2 octets read from reader command: 72 54 0 0 0 0 0 0 0          
#db# 1 octets read from reader command: f5 54 0 0 0 0 0 0 0          
#db# 1 octets read from reader command: 62 54 0 0 0 0 0 0 0          
#db# 1 octets read from reader command: 54 54 0 0 0 0 0 0 0          
#db# 3 octets read from reader command: 62 54 55 0 0 0 0 0 0          
#db# 1 octets read from reader command: aa 54 55 0 0 0 0 0 0          
#db# 3 octets read from reader command: 62 54 55 0 0 0 0 0 0          
#db# 1 octets read from reader command: f7 54 55 0 0 0 0 0 0          
#db# 1 octets read from reader command: aa 54 55 0 0 0 0 0 0          
pm3 --> hf list 15
Recorded Activity (TraceLen = 592 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
ISO15693 - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          0 |        256 | Rdr |01  01  01  01  01  01  01  01  01  01  01  01  00  01  01  01   |     |           
            |            |     |01  01  01  01  01  00  01  00  01  00  01  01  01  01  01  01   |     |           
            |            |     |01  01  01  01  01  01  01  01  01  01  01  01  01  01  01  01   |     |           
            |            |     |01  01  01  01  01  01  01  01  01  01  00  01  01  01  01  01   |     |           
            |            |     |01  01  01  01  01  01  01  00  01  01  01  01  01  01  01  01   |     |           
            |            |     |01  01  01  01  01  01  00  01  01  01  01  01  01  01  01  01   |     |           
            |            |     |01  01  01  01  01  00  01  01  01  01  01  01  01  01  01  01   |     |           
            |            |     |01  01  01  00  01  01  01  00  01  01  00  01  01  01  01  00   |     |           
            |            |     |01  01  01  01  01  01  01  01  01  01  01  01  01  01  01  00   |     |           
            |            |     |01  01  01  01  01  01  01  01  01  01  00  01  00  00  00  01   |     |           
            |            |     |01  01  01  01  01  00  01  01  01  00  01  01  01  01  01  01   |     |           
            |            |     |01  01  01  01  00  01  01  01  01  01  01  01  01  01  01  01   |     |           
            |            |     |01  01  01  01  01  01  01  01  01  01  01  01  01  00  01  01   |     |           
            |            |     |01  01  01  01  01  01  01  01  01  01  01  01  01  01  01  01   |     |           
            |            |     |01  01  01  01  01  01  01  01  01  01  01  01  01  01  01  01   |     |           
            |            |     |01  01  01  01  01  01  01  01  01  01  01  01  01  01  01  01   |     | INVENTORY       
```

iceman1001, 2017-09-07:
ok, 
you have a quite slow usb transfer...  You should look that up on your com-port setting.
the simulation seems to miss the start and get the end of reader communication..  ie the switching to reading signal seems too slow.. however the debug statement and slow usb will also influence that...

The card doesn't answer to the pm3 reader ( inventory command) either.  Well, this is also odd since you say it works on pm3 offical..

ceres-c, 2017-09-23:
Hello, I might have similar problems with ski tags. With your fork some are working and some are not, especially the ones made by Skidata are working and those based on NXP's Icode2 are not.

I'm using a chinese Proxmark3 Easy

**hw version** output
```
[[[ Cached information ]]]
          

Proxmark3 RFID instrument
          
 [ ARM ]
 bootrom: iceman/master/v1.1.0-2275-g72e53b09-dirty-unclean 2017-09-22 15:25:34
      os: iceman/master/v1.1.0-2275-g72e53b09-dirty-unclean 2017-09-22 15:25:35
 [ FPGA ]
 LF image built for 2s30vq100 on 2015/03/06 at 07:38:04
 HF image built for 2s30vq100 on 2017/05/17 at 17:48:26
          
 [ Hardware ]           
  --= uC: AT91SAM7S256 Rev D          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 217105 bytes (83%) Free: 45039 bytes (17%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```

**hw tune** output
```
Measuring antenna characteristics, please wait......          
# LF antenna: 39.74 V @   125.00 kHz          
# LF antenna: 33.41 V @   134.00 kHz          
# LF optimal: 40.15 V @   126.32 kHz          
# HF antenna: 29.99 V @    13.56 MHz          
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
```
As of @idaoudi, testing hw tune with and without the not working card gives different voltages, so it is indeed _seen_ by the proxmark.

The **hf search** command gives me this output with the skidata tag
```
UART:: write time-out
Sending bytes to proxmark failed          
 UID  : E0 16 24 66 06 BF 23 8D          
 TYPE : EM-Marin SA (Skidata); EM4233 [IC id = 09] 23,5pF CustomerID-102          

Valid ISO15693 Tag Found - Quiting Search
```
and **hf search** gives me this output with the NXP card
```
UART:: write time-out
Sending bytes to proxmark failed          
timeout while waiting for reply.          
UART:: write time-out
Sending bytes to proxmark failed          

no known/supported 13.56 MHz tags found
```
**hf 15 ---** commands such as read or info do not work on NXP tags

Thanks and sorry for the time it took you to read this wall of text...

iceman1001, 2017-09-25:
Thank you for taking the time to write and post your feedback. 
All clues to were a problem lays is welcomed at this moment.  

If it is possible to test and get the tracelog from:
PM3 Offical:  
**hf 15 cmd sysinfo 2**    
**hf list 15**
vs 
iceman fork
**hf 15 info 2**
**hf list 15**

I may have mixed up the commands, due to me not using offical version very much but you get the idea.

ceres-c, 2017-09-25:
_Iceman fork_ **hf 15 info 2**
```
Using UID 00 00 00 00 00 00 00 00          
iso15693 card doesn't answer to systeminfo command
```

_Iceman fork_ **hf list 15**
```
Recorded Activity (TraceLen = 31 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
ISO15693 - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          0 |      50224 | Rdr |22  2b  00  00  00  00  00  00  00  00  77  c9                   |  ok | GET_SYSTEM_INFO
```
_Official_ **hf 15 cmd sysinfo 2**
Using UID 0000000000000000          
timeout: no answer

_Official_ **hf list 15**
Command not found, list is supported only for 14a and 14b

Also, I could ship you one of those not working NXP tags, if you want. I have plenty of them :-)

iceman1001, 2017-09-25:
ok, lets see.  Not sure what you used the "2" for.  But don't use the "slower mode" -2, 
try the 
**hf 15 info u**   (iceman)
**hf list 15**
or 
**hf 15 cmd sysinfo u**  (offical)

Offical pm3 doesnt support tracelogging of ISO15693 commands...

iceman1001, 2017-09-25:
Feel free to send one of those not working NXP tags if you can spare one.

ceres-c, 2017-09-25:
I have used the 2 switch as you wrote it in your previous comment, I actually was not sure about it but I decided to run it without asking too much...

_Iceman_ **hf 15 info u**
```
Recorded Activity (TraceLen = 0 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
ISO15693 - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
```
_Iceman_ **hf list 15**
```
Recorded Activity (TraceLen = 0 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
ISO15693 - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------| 
```

_Official_ **hf 15 cmd sysinfo u**
```
0F 71 AE 12 50 00 02 04 E0 02 00 27 03 02 
UID = E00402005012AE71
NXP(Philips); IC SL2 ICS53/ICS54(SLI-S) ICS5302/ICS5402(SLIX-S)
DSFID supported, set to 02
AFI supported, set to 000
Tag provides info on memory layout (vendor dependent)
 4 (or 3) bytes/page x 40 pages 
IC reference given: 02
```

Write me in private your address, maybe via keybase or something like that, thanks

iceman1001, 2017-09-27:
Try the latest source.  I pushed a fix,  where waits for the tag to power up a bit before.
it increased reading distance quite alot on my pm3

ceres-c, 2017-09-27:
Nope, still not found
**hf search**
```
UART:: write time-out
Sending bytes to proxmark failed          
timeout while waiting for reply.          
UART:: write time-out
Sending bytes to proxmark failed          

no known/supported 13.56 MHz tags found
```

**hf 15 info u**
`iso15693 card doesn't answer to systeminfo command`

I believe it might be due to those UART timeouts...

iceman1001, 2017-09-27:
what distance do you have between tag and antenna?
..and the output from your  **hf status**

ceres-c, 2017-09-27:
The tag is right over the antenna, they are in contact.
**hw status**
```
#db# Memory          
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing          
#db#   tracing ................1          
#db#   traceLen ...............22          
#db# Fgpa          
#db#   mode....................LF          
#db# LF Sampling config:           
#db#   [q] divisor..............95 (125 KHz)          
#db#   [b] bps..................8          
#db#   [d] decimation...........1          
#db#   [a] averaging............Yes          
#db#   [t] trigger threshold....0          
#db# USB Speed:          
#db#   Sending USB packets to client...          
#db#   Time elapsed............1500ms          
#db#   Bytes transferred.......769536          
#db#   USB Transfer Speed PM3 -> Client = 513024 Bytes/s          
#db# Various          
#db#   MF_DBGLEVEL.............2          
#db#   ToSendMax...............23          
#db#   ToSendBit...............4          
#db#   ToSend BUFFERSIZE.......2308

```

iceman1001, 2017-10-23:
@pwpiwi latest fixes for 14B https://github.com/Proxmark/proxmark3/pull/438  should help this issue out.
The sensitivity for both 14b and 15 has increased.  I belive the fix is in the new fpga bit files on iceman fork aswell.  Have'nt gotten it confirmed.   But go ahead and test it.

iceman1001, 2017-10-24:
I got great reading distance from @pwpiwi fixes.   Its not in my fork yet, but if you take the fpga_hf.bit fil from pm3 offical and copy into iceman fork,  it works fine.

http://www.proxmark.org/forum/viewtopic.php?pid=30009#p30009

@ID-ao   @ceres-c   I belive I will close this issue now.  An updated fpga_hf.bit will come some.

closed 2017-10-24 by iceman1001