iceman1001/proxmark3 #126: How build firmware proxmark3 on android ?

maxben14, 2017-10-08:
How build firmware proxmark3 on android ?

iceman1001, 2017-10-08:
I have no idea.  There is an android section on the forum which deals with some old proxdroid version. It is outdated.

closed 2017-10-08 by iceman1001

maxben14, 2017-10-08:
@iceman1001 , i want do by this instruction https://github.com/marcv81/proxdroid , but in github marc81 old version firmware. I want do replace folder proxmark on your sorce from your repository. Is it will work and compile with new version sorce ?

osysltd, 2017-10-08:
@maxben14 Use https://github.com/angelsl/proxmark3-android many thanks to @angelsl

maxben14, 2017-10-08:
@osysltd , can write me how compile this apk or give link on build apk.

osysltd, 2017-10-08:
RTFM

maxben14, 2017-10-08:
@osysltd , what read ? not wiki in github.

osysltd, 2017-10-08:
@maxben14 RTFM on how to build Android apps, stop unnecessary flood please

maxben14, 2017-10-08:
@osysltd , ok, give me link please, Is it difficult ?

brantzau, 2017-10-08:
Damn, I can't hold anymore.

Can't you just do your own research, even a little tiny bit, you'll see the
manual.

maxben14, 2017-10-08:
What you say ?

iceman1001, 2017-10-08:
@maxben14   I still do not know.

maxben14, 2017-10-08:
@osysltd , i try build with gradle project but this project not all. https://github.com/angelsl/misc-proxmark3/tree/396772c93aa39600b9a14c58781583abbfcb05d2  Folder proxmark is empty in project @angelsl. I try copy folder proxmark from github @iceman1001  but when i do ./gradle build
i get this error:

 CMake Error at C:/proxmark3-android-master/clib/CMakeLists.txt:122 (add_library):
    Cannot find source file:

      proxmark3/client/cmdhfemv.c
What is cmdhfemv.c file ? I see folder https://github.com/Proxmark/proxmark3/tree/master/client here, but this file not found.

iceman1001, 2017-10-08:
cmdhfemv.c / .h is not in pm3 offical.  its in iceman fork.

maxben14, 2017-10-08:
@iceman1001 , ok, i copy file from iceman fork, but now this error:
 CMake Error at C:/proxmark3-android-master/clib/CMakeLists.txt:123 (add_library):
    Cannot find source file:

      `proxmark3/client/sleep.c`
I see iceman and offical forks but i found only sleep.h but sleep.c not found.

joanbono, 2017-10-09:
Hi @maxben14 ,

Use the released **apk** instead of compiling it.

https://github.com/angelsl/proxmark3-android/releases

The direct download link is the following: https://github.com/angelsl/proxmark3-android/releases/download/v0.0.1/proxmark3.apk

It may not work at all, it's a PoC, as you can see, but try directly the apk.

maxben14, 2017-10-09:
@joanbono , thank you for apk file, but apk don't work, i try send command use apk, but i get error: pm3dev_relay_shutdown: write(outfd) for RELAYDCMD EXIT: Broken pipe.

joanbono, 2017-10-09:
Hi @maxben14 

You should open an issue there (https://github.com/angelsl/proxmark3-android ) instead on the iceman's fork

maxben14, 2017-10-09:
@joanbono , i try open issue in fork angelsl, but i see error` You can't perform that action at this time.`