iceman1001/proxmark3 #128: unable to enumerate USB device

osysltd, 2017-10-12:
Dear Iceman,

I've build the latest commit just now and after flashing I cannot get device to work (pm3 Easy).
```
[80798.961813] usb 1-2: new full-speed USB device number 30 using xhci_hcd
[80798.962082] usb 1-2: Device not responding to setup address.
[80799.170317] usb 1-2: Device not responding to setup address.
[80799.377631] usb 1-2: device not accepting address 30, error -71
[80799.377712] usb usb1-port2: unable to enumerate USB device
```

iceman1001, 2017-10-12:
On which OS?

osysltd, 2017-10-12:
`Linux laptop 4.9.0-kali4-amd64 #1 SMP Debian 4.9.30-2kali1 (2017-06-22) x86_64 GNU/Linux`

iceman1001, 2017-10-12:
I pushed some reverts. Can you try it out?   You might need jtag programming it..

iceman1001, 2017-10-13:
The pm3 client has stable usb communications nows. 
now the question is why the device when entering "flashmode",  doesn't get proper USB recognision.

iceman1001, 2017-10-13:
https://github.com/iceman1001/proxmark3/issues/129

iceman1001, 2017-10-20:
This should be fixed now

closed 2017-10-21 by iceman1001