iceman1001/proxmark3 #129: Flasher broken after flashing bootrom

idaoudi, 2017-10-13:
Hi iceman,

In issue #125, after you said:

> Try the latest source. I pushed a fix, where waits for the tag to power up a bit before.
it increased reading distance quite alot on my pm3

I updated the bootrom with the latest commits at the time (d3282e766a2eb012da9635dc58d130f08852f005) . It flashed correctly, but then I couldn't update the os. 
I actually can't flash anything anymore, the flasher finds /dev/ttyACM0, it enters bootloader.
After a few seconds, the proxmark makes a "click" sound; at the same time /dev/ttyACM0 reappear for only less than a second, disappears and goes to /dev/ttyACM1.
At the same time, the flasher tries to flash but then nothing happens (because /dev/ttyACM0 is not here anymore I suppose ?)

This happens on Kali Linux.

I also tried on Windows 7 with no luck either. The flasher first enters bootloader and wait for Proxmark to reappear on com7.
Nothing happens then, it just wait indefinitely for the proxmark to reappear on com7. The proxmark disappears from com7 in the device manager, and Windows tells me a USB device is not recognized:
```
pm3 ~$ ./client/flasher.exe com7 -b bootrom/obj/bootrom.elf
Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00001090->0x00001090) [RWX] @0x298

Waiting for Proxmark to appear on com7. Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to reappear on com7................................................................................
................................................................................................... 
```
After a few minutes, I press the button to abort, and after a few seconds, the flasher enters "Flashing" step, but doesn't flash. 
At the same time, the unknown device disappears and after a few other seconds, the proxmark reappears on COM7 in Windows device manager:
```
............................................................. Found.
Note: Your bootloader does not understand the new START_FLASH command
      It is recommended that you update your bootloader

Flashing...
Writing segments for file: bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks]
```

The software still works correctly, I can use `./proxmark /dev/ttyACM0 [or com7]` on Kali or Windows.

Here is the output of `hw ver` and `hw status` on Windows :
```
pm3 --> hw ver
[[[ Cached information ]]]

Proxmark3 RFID instrument

 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-11-gd3282e76 2017-09-29 10:50:26
      os: iceman/master/v1.1.0-2267-ga5ffc567 2017-09-18 11:18:49
 [ FPGA ]
 LF image built for 2s30vq100 on 2015/03/06 at 07:38:04
 HF image built for 2s30vq100 on 2017/05/17 at 17:48:26

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 217097 bytes (41%) Free: 307191 bytes (59%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Fgpa
#db#   mode....................HF
#db# LF Sampling config:
#db#   [q] divisor..............95 (125 KHz)
#db#   [b] bps..................8
#db#   [d] decimation...........1
#db#   [a] averaging............Yes
#db#   [t] trigger threshold....0
#db# USB Speed:
#db#   Sending USB packets to client...
#db#   Time elapsed............1501ms
#db#   Bytes transferred.......364032
#db#   USB Transfer Speed PM3 -> Client = 242526 Bytes/s
#db# Various
#db#   MF_DBGLEVEL.............2
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
```

So, is my device bricked or is it a software/firmware issue somehow ?

iceman1001, 2017-10-13:
Thanks for the feedback.  I'm been fiddling with usb communications 
well.. its not brick if you can use it... 

The reason for all this current problems,  is my research into  WCID,  or  using Microsoft OS feature descriptors to get the PM3 device to be identified by Windows 8, 10 without the need for a signed driver file. 
As it is now,  some registry keys needs to be deleted when the device is NOT connected.   You can either uninstall the device in devicemanager or manually delete the registry keys.

When done,  once the device reconnects it will be identified under  **Universial Serial Bus devices**  as  **PM3 Device** without any extra installning.   Yeah!    

Still problems exists,  the flashmode (bootrom)  doesn't make the device identify as normal.  I'm looking at it now to get the device to be identified as usual when in flashmode.   This is where I belive you have issues on windows today.  
Another issues is that this new WCID installning doesn't assign a COMPORT automatically..  

So your device should work on Linux :)  but not on Windows.  

Try pullning latest fixes from today,  and  flash it again.  In worse case scenario you will need to JTAG program it.

ID-ao, 2017-10-13:
No, still no luck with latest fixes.. 
I tried on Kali, Win7 and Ubuntu, I can still use the software but can't flash bootrom/os

iceman1001, 2017-10-13:
Try jtag it with recovery/proxmark3_recovery.bin file.  

What happens on linux when you try to flash?  the dmesg -tail?

jenningsreeve, 2017-10-15:
As of yesterday I have the exact same problem.
I don't have a clue about JTAG.
Does this mean my brand new Proxmark3 RDV2 is stuck with firmware version?.
WTF are you thinking, releasing the version without even testing the basic flashing process.

joanbono, 2017-10-15:
@jenningsreeve as you can see [here](https://github.com/iceman1001/proxmark3#this-fork-is-highly-experimental-and-bleeding-edge), **This fork is HIGHLY experimental and bleeding edge**.

Which means that this fork comes with no warranties. 

The only thing you can say to iceman is "Thanks for sharing your fork".

jenningsreeve, 2017-10-15:
That doen't quite cover it.
Iceman has made changes to the bootloader without testing the result, or even warning taht is is not tested.
Thanks for your helpful reply

brantzau, 2017-10-16:
I reckon everything on GitHub is open source, feel free not to use it.

Developers are sacrificing their own time for none profit purpose. Just
hobbies.

Let's value their efforts by choosing words wisely.

joanbono, 2017-10-16:
@jenningsreeve neither is yours. 

Try to improve the code you didn’t read beforr flashing your device instead of coming here to cry for your broken Proxmark3.

Have a nice day 😄

iceman1001, 2017-10-16:
Aouch,  bummer,  sorry to hear.   However I do belive the latest release should be stable.
its only the latest source code that is screwing things up.   And well, yes,  its work in progress.  
Hence the big warning sign on the description,  "HIGHLY EXPERIMENTAL".

Now, I do get that you are frustrated and want to ventilate it here.  Do take time to breathe and understand that buying/borrowing a segger jtag clone is not a big deal.   Nor is it a hard process to learn either.   You will now have the change to learn new things,  which are needed when dealing with a device like proxmark3.   Sooner or later you brick it.  The PM3 is not an consumer product.  Nor is there a support, warrenty.  Its a hobby project as stated many times and to your luck its also open-source.

If you don't like this fork,  please stop using it.   You will only become frustrated and upset. I don't want you to become that. 

Back to the problem and a possible solution. The device should be recoqnised as a WINUSB device.  You should be able to look at it on the devicemanager and update its driver.  Point it to the old pm3 drivers and it should become a com port again.   *untested, no garantue it works*

jenningsreeve, 2017-10-16:
Iceman

I apologise for venting my frustration after finding the problem.

I found no "HIGHLY EXPERIMENTAL" warning in my following of the Window Client GUI information. 
I read all posts from Gaucho, Asper and Gator96100 regarding the GUI, no warnings!.
You posted the Gaucho version in April, no warnings and no errors reported.
I downloaded from Gator96100 post "Compiled Windows clients - always up to date" which gives no warning of "HIGHLY EXPERIMENTAL".

I can assure you, I read every post about boot problems and tried every possible option of drivers, firmware and Windows settings before posting.

I am more than happy to learn something that I am interested in, unfortulately the development of the PM3 code is not my field, I have chosen to help further it's application "RFID".
Glenn

jenningsreeve, 2017-10-16:
I also extend my sincere apologies to all users that were upset by tone of my post.
It will not happen again.
Glenn

jenningsreeve, 2017-10-16:
Iceman
If I have to learn JTAG to replace the faulty bootrom, would it be feasible for me to then develop an automated recovery boot disc that connects to the JTAG port and re-flashes the bootloader?.
Glenn

iceman1001, 2017-10-16:
Apology accepted,  
The warning sign is in my fork readme,  which is shown automatically by GitHub underneath the folders and source code structure.    Meaning if you ever cloned my fork, you would have had a hard time missing it.   Somehow I get the feeling that you didn't clone my fork, but downloaded it in a binary release...  Which doesn't makes sense,  since its the last commits that screws it up.   
The ProxmarkGui is nothing to do with my fork.  Using it is on your own risk,  it also needs a firmware to flash if you are using it with those batch files.   Still makes me belive you downloaded a pre-compiled binary.  

When we speak of releases;  https://github.com/iceman1001/proxmark3/releases  this is what is meant. 
Those releases should be stable.  I wouldn't release one if it wasn't stable.

I, with all other, always recommend to flash / compile the PM3 offical latest source.   To make sure you have a good working device and understand some of the basics of the device. 
Once you learn that,  you can dive into the forks. ie the enhanced firmwares.    You should also been aware of what this means.  

Now, back to your problem, 
The idea of an automatic recovery boot disc would be interesting,  if you get the dependencies of a jtag-programmer device solved.   I use a segger clone,  there is also hydrabus, buspirate etc which all can connect over jtag.   I love the idea.  Everything that makes it easier for users to recovery is worth gold as you just realised.  There is many posts on the forum talking about the different pin-layouts of the different hardware revisions of proxmark3.

iceman1001, 2017-10-16:
And ok,  you downloaded Gator9600's pre-compiled binaries.  check.   Yes,  then you would not have seen any warnings.

jenningsreeve, 2017-10-16:
Does anyone have experience with SAM-BA?
Could I use this to reload PM3RDV2 bootrom data?
https://startingelectronics.org/articles/atmel-ARM/SAM-BA-USB-CDC-driver-install/

iceman1001, 2017-10-17:
Now that was interesting to read.  So there is a built in function in at91sam7s to overwrite the bootrom making it a cdc acm.  Which will enable a flash of proxmark3-recovery.bin file without a jtag.
Instead of just re-writing bootrom, I would go with the complete image to ensure you have the same related code on the device.

No, I have no experience with it.

joanbono, 2017-10-17:
I have the same issue as the others above.

I have two Bus Pirates (`v3.6` and `v4.0`) at home, and I have ordered a J-Link JTAG. 

I'll try to automate the re-flashing process and will do a PR if you want.

BTW, the SAM-BA option looks interesting for people without a JTAG, and looks faster also.

iceman1001, 2017-10-17:
You should do fine with the buspirate.  
The needed config file is found here: https://github.com/Proxmark/proxmark3/tree/master/tools

ceres-c, 2017-10-17:
Hello there,
since last update my proxmark is stuck on "Waiting for Proxmark to reappear on /dev/ttyACM0" too.
I'll have to borrow a buspirate, try to flash with it and let you know.

joanbono, 2017-10-17:
Good news, my **Proxmark3** is alive!

I've used a **Bus Pirate v3.0** and a MacBook Pro with the modified config file mentioned by @iceman1001 before.

If anybody needs help, feel free to ping me:

I have added a brief guide [here](https://joanbono.github.io/PoC/Flashing_Proxmark3.html).

Basically, in one terminal:

~~~bash
jbono@MacBook [~/proxmark3]> openocd -f tools/at91sam7s512-buspirate.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
  http://openocd.org/doc/doxygen/bugs.html
  Warn : Adapter driver 'buspirate' did not declare which transports it allows; assuming legacy JTAG-only
  Info : only one transport option; autoselect 'jtag'
  adapter speed: 1000 kHz
[...]
~~~ 

And in the other one:

~~~bash
jbono@MacBook [~/proxmark3]> telnet localhost 4444
Trying ::1...
telnet: connect to address ::1: Connection refused
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger

>
> halt
> flash erase_sector 0 0 15
erased sectors 0 through 15 on flash bank 0 in 0.351705s
> flash write_image ./armsrc/obj/fullimage.elf
wrote 190924 bytes from file ./armsrc/obj/fullimage.elf in 281.332581s (0.663 KiB/s)
> flash write_image ./bootrom/obj/bootrom.elf
wrote 3776 bytes from file ./bootrom/obj/bootrom.elf in 6.327095s (0.583 KiB/s)
>
~~~

@iceman1001 Maybe it's possible to "automate" all this using a `expect` script.

PD: I've tried the **SAM-BA** but I cannot download the software (I got stucked waiting for the confirmation e-mail).

iceman1001, 2017-10-17:
Nice one! 
There have been many posts on un-bricking ,  you can also read about it on the forum,  or on wiki,  https://github.com/Proxmark/proxmark3/wiki/Debricking-Proxmark3   which writes about un-bricking with buspirate as @joanbono  did.

Sorry,  but what is a "expect" script?  
If you can make a script for recovery instead of these manual steps for buspirate, go for it.  
I welcome more great utils/scripts that makes the whole proxmark3 experience nicer.

joanbono, 2017-10-17:
Hi @iceman1001 

`expect` is a UNIX program: https://www.lifewire.com/linus-unix-command-expect-2201096

The point is to execute a script just like: 

~~~bash
./proxmark_reflashing $BusPirate_Port
~~~

And then it will do all the re-flashing steps without _human_ interaction.

I'll work on it if you find it useful.

iceman1001, 2017-10-17:
@joanbono    I welcome it.   Any script which make things easier for users, go for it!

iceman1001, 2017-10-20:
This should be fixed now.

ceres-c, 2017-10-20:
It works, confirmed 👍

iceman1001, 2017-10-23:
Still not working?  It works for me on my windows.  Haven't tested on my virtual machines (ubuntu)

@joanbono  did you finish that expect script of yours?

iceman1001, 2017-10-23:
device didn't get enough power?    ` device descriptor read/64, error -110`

joanbono, 2017-10-24:
@iceman1001 it's almost finished in my local repository (not pushed yet).

I'll finish the tests and push the changes.

iceman1001, 2017-10-24:
Perfect,  I look forward to it @joanbono   
meanwhile I close this one.  Its working again.

closed 2017-10-24 by iceman1001

joanbono, 2017-12-01:
Hi @iceman1001 , the `Re-Flasher` is working. 

You can check it at my repo [`pm3RecoveryKit`](https://github.com/joanbono/pm3RecoveryKit).

Cheers! 😃