iceman1001/proxmark3 #136: Some basic helper scripts

winguru, 2017-10-23:
Iceman,

I'm not sure if you (or anyone else) would find these useful, but I wrote a couple of very simple BASH scripts to update the Proxmark3 firmware and to launch the application.  They both do a simple detection to find the first /dev/ttyACM device present.  On rare occasions, I've seen the device jump to a higher numbered interface (typically during or after a firmware upgrade), so it's nice to have it find the device automatically.

Feel free to use them (or not) as you see fit, or let me know if you'd like any sort of additional checks/enhancements.

Thanks!!

The branch with the scripts is located here:
https://github.com/winguru/proxmark3/tree/basic-helper-scripts

iceman1001, 2017-10-23:
Interesting scripts.   you make a PR. 

You know, on linux,  if you run the "make udev" you get the updated blacklist,  but you also get aliases.
- /dev/ttyACM0 
- /dev/pm3-0
So you should be able to identify not only a ttyACM device but a proxmark device :)

winguru, 2017-10-23:
Thanks for the heads up!  I've honestly not used the "make udev" feature, but that should be simple enough to check for.

closed 2017-10-23 by iceman1001