iceman1001/proxmark3 #138: LF Indala, viking and pyramid broken

phiberx, 2017-10-29:
pm3 --> hw ver
[[[ Cached information ]]]

Proxmark3 RFID instrument

 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-114-g0d2b6e80-dirty-unclean 2017-10-29 01:53:30
      os: iceman/master/ice_v3.1.0-114-g0d2b6e80-dirty-unclean 2017-10-29 01:53:39
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/10/25 at 19:51:09

 [ Hardware ]
  --= uC: AT91SAM7S256 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 230483 bytes (88%) Free: 31661 bytes (12%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

pm3 --> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

HID Prox TAG ID: 000123456 (6699) - Format Len: 37bit - FC: 1 - Card: 72235

Valid HID Prox ID Found!
pm3 --> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

Valid Indala ID Found!
pm3 --> lf viking clone 01020304
Cloning - ID: 01020304, Raw: F20000010203045E
pm3 --> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

Valid Indala ID Found!
pm3 --> lf hid clone 02030401
Cloning tag with ID 002030401
pm3 --> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

HID Prox TAG ID: 002030401 (33280) - Format Len: 37bit - FC: 32 - Card: 98816

Valid HID Prox ID Found!
pm3 --> lf pyr clone 123 12233
Preparing to clone Farpointe/Pyramid to T55x7 with Facility Code: 123, Card Number: 12233
Blk | Data
----+------------
 00 | 0x00107080
 01 | 0x00010101
 02 | 0x01010101
 03 | 0x0101016E
 04 | 0xB37F2679
pm3 --> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

No Known Tags Found!

pm3 --> lf indala clone 01020304
Cloning 64bit tag with UID 001020304
pm3 --> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

Valid Hitag Found!

pm3 --> hw tune

Measuring antenna characteristics, please wait......
# LF antenna: 19.39 V @   125.00 kHz
# LF antenna: 27.64 V @   134.00 kHz
# LF optimal: 27.64 V @   133.33 kHz
# HF antenna: 23.04 V @    13.56 MHz
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

iceman1001, 2017-10-29:
All PSK...  yep,  fits the pattern with what I tried to fix somewhere in july/august.  
but then I got sidetracked and left it in the middle with kind of broken status.  
I belive I even tried to explain it on the forum somewhere.  Needless to say **feel free to fix**

iceman1001, 2017-10-30:
Or maybe not,  since today when I pushed a better signal_noise detection the **LF SEARCH**  crashes alot.
I'm patching up as I go.   The most crashes happens when you don't have a tag at all on the antenna.

Right now the hitag detection crash both client, keeps the client hanging which results in a WDT...
That is not one stable client.

alexjx, 2017-11-09:
I got a HID tag crash today... then I found the DetectASKClock function was not correctly expecting the input clock to be 0, which crashes with SIGFPE.
After filtering the input, it seems working..
However, I'm not experts in ASK demod of any sort.... so I dont know if this is a proper fix.
But it works for me...

`
diff --git a/common/lfdemod.c b/common/lfdemod.c
index d4c07d6..6f79126 100644
--- a/common/lfdemod.c
+++ b/common/lfdemod.c
@@ -533,7 +533,9 @@ int DetectASKClock(uint8_t dest[], size_t size, int *clock, int maxErr) {
                }
        }
        // test clock if given as cmd parameter
+       if (*clock != 0) {
                clk[0] = *clock;
+       }
 
        uint16_t ii;
        uint8_t clkCnt, tol = 0;

`

iceman1001, 2017-11-09:
I did some changes to AskClock last week.  Trying to remove the many check HIGH/LOW loops.
Now I need to verify that all calls work again  :(

samturner3, 2018-01-04:
Yeh on latest, I did a `lf search` on an EM tag, however is identified as an indala tag. 

```
>lf search
Checking for known tags:
Valid Indala ID Found!

```
```
>lf indala read
(nothing)
```

```
[34mProxmark3 RFID instrument[0m

 [ ARM ]
 bootrom: iceman/master/ 2018-01-03 15:50:56
      os: iceman/master/ 2018-01-03 15:50:59
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/11/10 at 19:24:16

 [ Hardware ] 
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 235803 bytes (45) Free: 288485 bytes (55)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

# LF antenna: 43.73 V @   125.00 kHz
# LF antenna: 19.39 V @   134.00 kHz
# LF optimal: 45.92 V @   123.71 kHz
# HF antenna: 30.77 V @    13.56 MHz
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

```
However when I switched to master branch, the EM tag was identified correctly.

iceman1001, 2018-01-04:
ok, I also see esc-codes doesn't work on OSX terminals...

Tom5ive, 2018-01-09:
Just wanted to add that em410x seems to be broken too. Issuing "lf search" with a t55 card emulating em410x gets recognised as an indala ID.

iceman1001, 2018-09-18:
@TomHarkness   do you have some indala etc card to test out on RRG repo?

iceman1001, 2019-02-24:
I found a bug with false positive identifcations of indala ,  like in 'lf search',   pushed fix to RRG repo.
@TomHarkness  you can test if this is better
Also found a bug for jablotron clone command... fix is also out.

closed 2019-02-24 by iceman1001