iceman1001/proxmark3 #139: hardnested / "Can't select card (UID)" / "AcquireNonces: Auth1 error"

rainscope, 2017-10-29:
Running the latest iceman version (`59fbf1e3`), `hardnested` fails to work with a stream of "AcquireNonces: Can't select card (UID)", followed by "Authentication failed. Card timeout." and "AcquireNonces: Auth1 error".

The same command works (with the card in the same location) on both the latest iceman release ("minor Sweet Lemon") and the latest Proxmark3 git version (`2031e3f`).

I am running OS X 10.12.4.

## Log

```
pm3 --> hf mf hardnested 3 B FFFFFFFFFFFF 2 A
--target block no:  2, target key type:A, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 1103 million (2^30.0) keys/s     | 140737488355328 |   35h
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |   35h
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
[snip]
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
Waiting for a response from the proxmark...
Don't forget to cancel its operation first by pressing on the button
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (UID)
#db# AcquireNonces: Can't select card (ALL)
Error: No response from Proxmark.

#db# Authentication failed. Card timeout.
pm3 --> #db# AcquireNonces: Auth1 error
#db# Authentication failed. Card timeout.
#db# AcquireNonces: Auth1 error
#db# Authentication failed. Card timeout.
#db# AcquireNonces: Auth1 error
#db# Authentication failed. Card timeout.
#db# AcquireNonces: Auth1 error
#db# Authentication failed. Card timeout.
#db# AcquireNonces: Auth1 error
[PM3 Button pressed]
```

## Revision Information
```
 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-103-g1f8b08cf 2017-10-23 21:08:08
      os: iceman/master/ice_v3.1.0-127-g59fbf1e3 2017-10-29 16:25:17
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/10/25 at 19:51:09

 [ Hardware ]
  --= uC: AT91SAM7S256 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 231461 bytes (88) Free: 30683 bytes (12)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```

iceman1001, 2017-10-30:
its a bug somewhere in the 14A code.   It seem to need longer timeouts...

iceman1001, 2017-10-30:
Pull latest source, compile/flash,   and try running with the slow parameter.

iceman1001, 2017-10-30:
its should be fixed now.

closed 2017-10-30 by iceman1001

rainscope, 2017-10-31:
That fixed it, thanks! Slow option wasn't needed.