iceman1001/proxmark3 #143: hf search not fining legic

scaery, 2017-11-24:
Hi, current legic detection with "hf search" seems to be broken... indeed "hf legic reader" works fine. Did anyone encounter the same issue testing Legic cards?

iceman1001, 2017-12-01:
...some more details?
The legic commands is notorious to have the tag right on the antenna.  
**hf search**   This is the downside with the search command.  All protocols like 14a, 14b, 15, iclass, legic has different reading distances making this command almost useless as it is today.

Feel free to contribute a better reading distance for Legic!

scaery, 2017-12-19:
Okay, the problem was an antenna issue. With the original RyscCorp RFID Antenna REV A the card really wont be detected by hf search. Building my own antenna fixed it. But anyways, i get same bad results from Proxmark RVD 2. That was a learning by doing experiment so using self build antennas is much more convenient for detecting legic cards.

[closed]

closed 2017-12-19 by scaery

iceman1001, 2017-12-19:
A better antenna helps :)
And a better demod for legic would be even better.