iceman1001/proxmark3 #146: crc fail when dumping EM-Marin SA (Skidata); EM4233

hammerzeit, 2017-12-10:
Hi, I'm having an issue dumping an EM4233 card. It's an old skidata ski pass. 
I can read individual pages, but dumping gives me crc errors. 
(UID anonymized)

```pm3 --> hf 15 dump
Reading memory from tag UID E0 XX XX XX XX XX XX XX          
crc fail          
crc fail          
crc fail          
crc fail          
crc fail          

block#   | data         |lck| ascii          
---------+--------------+---+----------          

Saved 0 blocks to text file -4.eml          
Saved 0 bytes to binary file -4.bin```
Also, I can only read sectors if I select the card while reading, not the pre-selected card. 

This does not work:

```pm3 --> hf 15 select
Detected UID E0 XX XX XX XX XX XX XX   
Card is selected          
pm3 --> hf 15 read s 0
iso15693 card select failed```

This does:

```pm3 --> hf 15 read * 0
Detected UID E0 XX XX XX XX XX XX XX     
          
block #  0  |lck| ascii          
------------+---+------          
XX XX XX XX | 0 | XXXX```

I think the dumping may not be working due to the card select issue? I've tried two different cards, with the same results.

My Proxmark RDV2 with the latest Iceman firmware:
```pm3 --> hw ver
[[[ Cached information ]]]
          

Proxmark3 RFID instrument
          
 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-272-g71fa4614 2017-12-10 13:22:38
      os: iceman/master/ice_v3.1.0-272-g71fa4614 2017-12-10 13:22:40
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/11/10 at 19:24:16
          
 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 230025 bytes (44%) Free: 294263 bytes (56%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory```

The card:
```pm3 --> hf search
          
 UID  : E0 XX XX XX XX XX XX XX          
 TYPE : EM-Marin SA (Skidata); EM4233 [IC id = 09] 23,5pF CustomerID-XXX```

```pm3 --> hf 15 info *
Detected UID E0 XX XX XX XX XX XX XX         
  UID  : E0 XX XX XX XX XX XX XX         
  TYPE : EM-Marin SA (Skidata); EM4233 [IC id = 09] 23,5pF CustomerID-XX          
  SYSINFO : 00 0F XX XX XX XX XX XX XX E0 02 00 33 03 02           
     - DSFID supported        [0x02]          
     - AFI   supported        [0x00]          
     - IC reference supported [0x02]          
     - Tag provides info on memory layout (vendor dependent)          
           4 (or 3) bytes/blocks x 52 blocks```

iceman1001, 2017-12-10:
Thanks for letting me know.
Did you try different distances when reading the tag?  Not just direct on the antenna.

The iso-15 cmd implementations is different from all the rest of pm3.  Clearly done by someone else and how they thought you should work with the tag and proxmark.

The 'selected'  command I guess is supposed to be part of a  "raw" command,  when you send raw data to it.  But the 15 read selects by itself,  hence no need for "selecting" before.   Could very well be the reason for the read to fail.

Just to make sure, you didnt select before the **hf 15 dump** ?

hammerzeit, 2017-12-10:
Thanks for your reply Iceman.

I have tried different distances. I get best results if I put a small empty cardboard box in between, which gives me a distance of about 1cm. But the distance doesn't really matter for the results. (Except of course if the card is directly on the antenna or too far away).

The hf 15 read command has 4 different modes of reading. 
- By UID, where you enter the UID of your card in the command
- Selected, which reads from selected tags
- Unaddressed
- Scan for tag

So in fact you do need to select the card beforehand if you use `hf 15 read s`. At least that's how I understand it. In any case, the selected read mode does not work, even if you select the card beforehand using `hf 15 select`.

Should I report this bug to the main repo?

Thanks and happy holidays

iceman1001, 2017-12-10:
Well,  does the offical pm3 have the same problem?

Strange that you can read indiviual blocks and not the dump.
What is the output from  "Hf list 15"  afterwards?

hammerzeit, 2017-12-11:
I'll reflash and try the official f/w,  but for now:

after `hf 15 dump`:
```
pm3 --> hf 15 list
Recorded Activity (TraceLen = 39 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
ISO15693 - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          0 |      51776 | Rdr |22  20  XX  XX  XX  XX  XX  XX  XX  e0  00  fb  4f               |  ok | READBLOCK          
       8832 |      16480 | Tag |00! 82! 08  35! 60! b1! b6                                       |  ok |           

```

After `pm3 --> hf 15 read * 0`:

```
pm3 --> hf 15 list
Recorded Activity (TraceLen = 40 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
ISO15693 - Timings are not as accurate          
          
      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          0 |      51808 | Rdr |62  20  XX  XX  XX  XX  XX  XX  XX  e0  00  fe  82               |  ok | READBLOCK          
       8864 |      16384 | Tag |00! 00! 82! 08  35! 60! 49  8e!                                  |  ok |           

```

hammerzeit, 2017-12-11:
Official firmware is much less featured with ISO15x:

``` 
bootrom: master/v3.0.1-213-gbc3b2f7-suspect 2017-12-09 13:45:04
os: master/v3.0.1-213-gbc3b2f7-suspect 2017-12-09 13:45:05
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59
```
`hf 15`:
```
proxmark3> hf 15
help             This help          
demod            Demodulate ISO15693 from tag          
read             Read HF tag (ISO 15693)          
record           Record Samples (ISO 15693)          
reader           Act like an ISO15693 reader          
sim              Fake an ISO15693 tag          
cmd              Send direct commands to ISO15693 tag          
findafi          Brute force AFI of an ISO15693 tag          
dumpmemory       Read all memory pages of an ISO15693 tag   
```
The `dumpmemory` command does seem to work, I get the same data back from the card like I do reading individual blocks using iceman. Using dumpmemory on official firmware it does error out with `Tag returned Error 15: Unknown error. ` when trying to read block 34 so it stops after block 33. Using iceman I can read up to and including block 51 fine using `hf 15 read * <blocknr>`

`hf 15` on iceman f/w:
```pm3 --> hf 15
help             This help          
debug            Turn debugging on/off          
demod            Demodulate ISO15693 from tag          
dump             Read all memory pages of an ISO15693 tag, save to file          
findafi          Brute force AFI of an ISO15693 tag          
info             Tag information          
list             [Deprecated] List ISO15693 history          
raw              Send raw hex data to tag          
reader           Act like an ISO15693 reader          
record           Record Samples (ISO15693)          
restore          Restore from file to all memory pages of an ISO15693 tag          
sim              Fake an ISO15693 tag          
samples          Acquire Samples as Reader (enables carrier, sends inquiry)          
select           Select an tag with a specific UID for further commands          
read             Read a block          
write            Write a block          
readmulti        Reads multiple Blocks          
```
So I do prefer your firmware :)

iceman1001, 2017-12-11:
its a work in progress.  I have no time to finish up the iso15 changes. 
Feel free to find a solution to the dump command.  Either its the demod or it is actually a bad dump command implementation.

iceman1001, 2017-12-21:
Ok,  I found the bug,  and pushed a fix for it.   Just pull latest source and compile/flash.

The dump should work again,  I also added a help text for it...

Let me know, and we can close this one.

closed 2017-12-24 by iceman1001

hammerzeit, 2017-12-24:
Thanks for your effort, the dump now completes, and the data looks correct (as far as I can tell).

Cheers, and happy holidays again.

iceman1001, 2017-12-24:
Great,  remember,  nothing says thank you as much as a donation!

Merry Christmas! 
i'm signing off, AFK.