iceman1001/proxmark3 #153: Need help. Proxmark 3 is not flashing

PenGenKiddy, 2018-01-04:
```
 ~> dmesg
 [208742.393451] usb 1-1.2: new full-speed USB device number 22 using ehci-pci
 [208747.684960] usb 1-1.2: device descriptor read/64, error -110
 [208752.982103] usb 1-1.2: New USB device found, idVendor=9ac4, idProduct=4b8f
 [208752.982108] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
 [208752.982110] usb 1-1.2: Product: PM3
 [208752.982112] usb 1-1.2: Manufacturer: proxmark.org
 [208752.982578] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
```
After this, i run:
```
 # ./flasher /dev/ttyACM0 ../armsrc/obj/fullimage.elf 
 Loading ELF file '../armsrc/obj/fullimage.elf'...
 Loading usable ELF segments:
 0: V 0x00102000 P 0x00102000 (0x00035948->0x00035948) [R X] @0x94
 1: V 0x00200000 P 0x00137948 (0x00001a1c->0x00001a1c) [RW ] @0x359dc
 Note: Extending previous segment from 0x35948 to 0x37364 bytes
 
 .Waiting for Proxmark to appear on /dev/ttyACM0 Found.
 Entering bootloader...
 (Press and release the button only to abort)
 Waiting for Proxmark to reappear on /dev/ttyACM0......
```
In this time:
```
  ~> dmesg
 [208742.393451] usb 1-1.2: new full-speed USB device number 22 using ehci-pci
 [208747.684960] usb 1-1.2: device descriptor read/64, error -110
 [208752.982103] usb 1-1.2: New USB device found, idVendor=9ac4, idProduct=4b8f
 [208752.982108] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
 [208752.982110] usb 1-1.2: Product: PM3
 [208752.982112] usb 1-1.2: Manufacturer: proxmark.org
 [208752.982578] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
 [208765.716614] usb 1-1.2: USB disconnect, device number 22
 [208765.732939] cdc_acm 1-1.2:1.1: urb 0 failed submission with -19
 [208766.451227] usb 1-1.2: new full-speed USB device number 23 using ehci-pci
 [208766.947273] usb 1-1.2: device not accepting address 23, error -32
 [208767.027181] usb 1-1.2: new full-speed USB device number 24 using ehci-pci
 [208767.523215] usb 1-1.2: device not accepting address 24, error -32
 [208767.603189] usb 1-1.2: new full-speed USB device number 25 using ehci-pci
 [208768.019172] usb 1-1.2: device not accepting address 25, error -32
 [208768.099080] usb 1-1.2: new full-speed USB device number 26 using ehci-pci
 [208768.515044] usb 1-1.2: device not accepting address 26, error -32
 [208768.515278] usb 1-1-port2: unable to enumerate USB device
```

I have BusPirate V4 if I need to.
But ProxMark3 do not want to connect even to JTag

iceman1001, 2018-01-04:
That doesnt sound good.  Never seen this one before. 
A bit too little information.  Like OS,  USB2.0 / USB3.0 port,  since it sounds more like a OS problem but then your pm3 easy could be faulty aswell..

PenGenKiddy, 2018-01-05:
Ubuntu Server 17.10 + XFCE
`Linux laptop 4.10.0-42-generic #46-Ubuntu SMP Mon Dec 4 14:38:01 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`
I tryed USB2 and USB3 ports
```
 ~> sudo lspci -v
[sudo] password for pengenkiddy: 
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 1566
	Subsystem: Dell Device 06bf
	Flags: bus master, fast devsel, latency 0

00:01.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Mullins [Radeon R4/R5 Graphics] (rev 45) (prog-if 00 [VGA controller])
	Subsystem: Dell Mullins [Radeon R4/R5 Graphics]
	Flags: bus master, fast devsel, latency 0, IRQ 39
	Memory at e0000000 (64-bit, prefetchable) [size=256M]
	Memory at f0000000 (64-bit, prefetchable) [size=8M]
	I/O ports at f000 [size=256]
	Memory at feb00000 (32-bit, non-prefetchable) [size=256K]
	Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: [48] Vendor Specific Information: Len=08 <?>
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [270] #19
	Capabilities: [2b0] Address Translation Service (ATS)
	Capabilities: [2c0] #13
	Capabilities: [2d0] #1b
	Kernel driver in use: radeon
	Kernel modules: radeon, amdgpu

00:01.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Kabini HDMI/DP Audio
	Subsystem: Dell Kabini HDMI/DP Audio
	Flags: bus master, fast devsel, latency 0, IRQ 43
	Memory at feb64000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [48] Vendor Specific Information: Len=08 <?>
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel

00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 156b
	DeviceName:  Onboard IGD
	Flags: fast devsel

00:02.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 16h Processor Functions 5:1 (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 25
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 00001000-00001fff
	Memory behind bridge: f0a00000-f0bfffff
	Prefetchable memory behind bridge: 00000000f0c00000-00000000f0dfffff
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Root Port (Slot+), MSI 00
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [b0] Subsystem: Dell Family 16h Processor Functions 5:1
	Capabilities: [b8] HyperTransport: MSI Mapping Enable+ Fixed+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Kernel driver in use: pcieport
	Kernel modules: shpchp

00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 16h Processor Functions 5:1 (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 27
	Bus: primary=00, secondary=02, subordinate=02, sec-latency=0
	I/O behind bridge: 0000e000-0000efff
	Memory behind bridge: fea00000-feafffff
	Prefetchable memory behind bridge: 00000000f0800000-00000000f08fffff
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Root Port (Slot+), MSI 00
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
[... 722 lines cut ...]
       configuration: autonegotiation=off broadcast=yes driver=veth driverversion=1.0 duplex=full link=yes multicast=yes port=twisted pair speed=10Gbit/s
  *-network:3
       description: Ethernet interface
       physical id: 5
       logical name: docker0
       serial: 02:42:c8:d8:f2:f1
       capabilities: ethernet physical
       configuration: broadcast=yes driver=bridge driverversion=2.3 firmware=N/A ip=172.17.0.1 link=no multicast=yes
  *-network:4
       description: Ethernet interface
       physical id: 6
       logical name: veth99df4ce
       serial: 86:b5:98:4a:91:14
       size: 10Gbit/s
       capabilities: ethernet physical
       configuration: autonegotiation=off broadcast=yes driver=veth driverversion=1.0 duplex=full link=yes multicast=yes port=twisted pair speed=10Gbit/s
  *-network:5
       description: Ethernet interface
       physical id: 7
       logical name: vethb59c7ca
       serial: f2:5f:54:fd:61:a3
       size: 10Gbit/s
       capabilities: ethernet physical
       configuration: autonegotiation=off broadcast=yes driver=veth driverversion=1.0 duplex=full link=yes multicast=yes port=twisted pair speed=10Gbit/s
```

iceman1001, 2018-01-05:
Next time, please use pastebin.com for lengthy outputs.  

Did you set the udev rules (blacklisting the modemmanager etc)  
`sudo make udev`

WolfgangMau, 2018-01-16:
here also ...
started with my fork installed -> worked
flashed marshmellows fork -> worked
flashed your fork did get flashed, but did not even read a tag
try to flash anything els will not work anymore ... it stays forever in 
`Waiting for Proxmark to reappear on /dev/tty.usbmodem881..................`
since your fork was flashed the devicename changed
bevore it was always /dev/tty.usbmodem1411 or /dev/tty.usbmodem1412
now its /dev/tty.usbmodem881

any idea?

iceman1001, 2018-01-16:
yes,  its because iceman fork sets the serial number on the usb identifiction.  
So you need to restart your command with the new enumeration.

WolfgangMau, 2018-01-16:
sorry?
what should I do?
;-)

WolfgangMau, 2018-01-16:
../../bin/proxmark3-flasher /dev/tty.usbmodem881 fullimage.elf
Loading ELF file 'fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00031650->0x00031650) [R X] @0x94
1: V 0x00200000 P 0x00133650 (0x00001a14->0x00001a14) [RW ] @0x316e4
Note: Extending previous segment from 0x31650 to 0x33064 bytes

Waiting for Proxmark to appear on /dev/tty.usbmodem881. Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to reappear on /dev/tty.usbmodem881

then nothing else happens exept new dots comming up ;-)

iceman1001, 2018-01-16:
Did you set the udev rules (blacklisting the modemmanager etc) ?
I don't have OSX , so I don't have any real answers when it comes to OSX. Just the normal ones.

WolfgangMau, 2018-01-16:
I'm on OSX

iceman1001, 2018-01-16:
That part I got.

WolfgangMau, 2018-01-16:
ok, I got a non working pm3 - fair enough ... so we both got at least something :-)

iceman1001, 2018-01-16:
I don't say you can't use the pm3 anymore.  There are ppl who used iceman fork on OSX and they all managed to flash it.
Question is which usb enumeration your computer got when you put it in bootload mode?  
That enumeration is what you should use.   
I.e.
bin/proxmark3-flasher /dev/tty.usbmodem881 fullimage.elf
device enter bootloader.
ctrl-c  to break the flasher since tty.usbmodem881 isn't there.
find which enumeration the device has now.
restart flasher with new enumeration:
bin/proxmark3-flasher /dev/tty.XXXXXX fullimage.elf

WolfgangMau, 2018-01-16:
perfect - works!
thanks ... again ... and again ;-)

WolfgangMau, 2018-01-16:
thus I asked 'what should I do'?  didn't understand what you meant

iceman1001, 2018-01-16:
Good that it works.   The take-over of someones issue is also not recommended.
However spoonfeeding is not helpful in the long run.  You didn't learn anything now.

What the status for original poster @PenGenKiddy ?

WolfgangMau, 2018-01-16:
no reason to get rude ;-)
the problem is pretty much describing mine - so absolutely no 'take-over' happens here.
the problem seems to be very specific to your fork - and to share means to care (that's what you told me on former legic-prime times)
btw...  I learned at least that your fork is very 'special'

iceman1001, 2018-01-16:
:)  Sorry if you feel offended.   I did expect more from you since the legic-primes times ;)  You are better than this.

My fork is different,  that didn't change either from the legic-prime ages ;)   I still have the timing issues with legic prime in the fork.

osysltd, 2018-01-17:
@WolfgangMau this is a known problem if smth. like NetworkManager is interferring with the port

WolfgangMau, 2018-01-17:
@osysltd 
I got it working/flashing once I understood what @iceman1001 means with
'*So you need to restart your command with the new enumeration*'
I just didn't understood at first ;-)
I'm a simple guy - so sometimes I need it in a more simple way to get it - absolutely my fault! 
'the device get's renamed while switching to bootloader' would be the simple explanation to that (for me).

iceman1001, 2018-01-17:
Maybe we should add a Wiki to iceman-fork....    Where things like this could have been written down, instead of hidden in forum posts and closed/open issues.     Or you add it to the PM3 offical wiki,  under MacOS,   a segement about going to/from other firmwares to pm3 offical...

Sorry, I took a too deep look at the usb-communications last year and it has marked me for life. Things like _new enumeration_ in the usb context makes perfect sense to me.

iceman1001, 2018-02-02:
@PenGenKiddy   has your problem got solved?

PenGenKiddy, 2018-02-02:
@iceman1001 No. Some days ago i installed Arch Linux.
Now: https://pastebin.com/J6thY8AM

iceman1001, 2018-02-02:
There was another thread on the forum yesterday about a bad Easy v3.0..  Having trouble flashing firmware.  Only worked with the firmware from the seller.   Where did you buy it?  

And you say the jtag doesn't even work,  which suggests to me that your device seems to be crappy.

PenGenKiddy, 2018-02-02:
@iceman1001 I did not buy it, but my good friend.
Bought at store.ryscc.com

iceman1001, 2018-02-02:
Contact them and explain the situation.  I can't help you.
I'm closing this.

PenGenKiddy, 2018-02-02:
@iceman1001 JTAG working, but today I do not have access to the JTAG device

iceman1001, 2018-02-02:
Well,  since you posted this in the first post..  Try to flash with jtag and see if your device comes to life again.

> I have BusPirate V4 if I need to.
> But ProxMark3 do not want to connect even to JTag

closed 2018-02-02 by iceman1001