iceman1001/proxmark3 #154: Standalones mods not working

Th3ju, 2018-01-10:
Hi,

First, thank you for this fork ! 

I've some problems to make some standalone mods working. 
With HF_YOUNG, no problems, all is working as expected.

If I try to compile with colin mod (-DWITH_HF_COLIN in Makefile). Compilation seems to be correct, no problems to flash the PM3 but nothing append when I press the pm3 button. (If I try directly from COLIN github, it's working, the button launch the mod)

Is there anything else to do than replacing -DWITH_HF_YOUNG by -DWITH_HF_COLIN in the Makefile file ? Did I miss something ? 

Thanks

TJ

iceman1001, 2018-01-10:
ok, you change in the makefile,   did you also change inside appmain.c  https://github.com/iceman1001/proxmark3/blob/master/armsrc/appmain.c#L1184  as stated on the readme?

The particular colin standalone mode might not be finished,  @cjbrigato was finishing up his versions of it in his repo.  I'm not sure that the version here works.

Th3ju, 2018-01-10:
OK, thank you,

Effectively, I didn't change the appmain file... It's OK now. 

I'll use @cjbrigato repo to have the last up to date mod.

Thank you !

closed 2018-01-10 by Th3ju

osysltd, 2018-01-10:
> as stated on the readme

@iceman1001 would you please share the reference to readme how to switch between mods correctly?

iceman1001, 2018-01-10:
There is two places which needs adaptions. It is mentioned here:
https://github.com/iceman1001/proxmark3/blob/master/armsrc/Standalone/readme.md

- /armsrc/makefile 
- /armsrc/appmain.c
( https://github.com/iceman1001/proxmark3/blob/master/armsrc/appmain.c#L1184 )

Not super educational :(  Feel free to update the instructions so they become clear.