iceman1001/proxmark3 #161: macOS Sierra fails to compile from homebrew

n0ncetonic, 2018-02-01:
Probably due to the CRC changes I'm about to take a look at it now but wanted to let you know things are broken. Here's my attempt to update.

```
$ brew upgrade --fetch-HEAD iceman1001/proxmark3/proxmark3
==> Upgrading 1 outdated package, with result:
iceman1001/proxmark3/proxmark3 HEAD-e0373212a38e063bf7f5a68b448c3de83ff63891
==> Upgrading iceman1001/proxmark3/proxmark3
==> Cloning https://github.com/iceman1001/proxmark3.git
Updating /Users/user/Library/Caches/Homebrew/proxmark3--git
==> Checking out branch master
==> make clean
==> make all
Last 15 lines from /Users/user/Library/Logs/Homebrew/proxmark3/02.make:
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                          ^
../common/iso15693tools.c:10:5: error: conflicting types for '__builtin___sprintf_chk'
int sprintf(char *str, const char *format, ...);
    ^
/usr/include/secure/_stdio.h:47:3: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
  ^
../common/iso15693tools.c:10:5: note: '__builtin___sprintf_chk' is a builtin with type 'int (char *, int, unsigned long, const char *, ...)'
/usr/include/secure/_stdio.h:47:3: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
  ^
3 errors generated.
make[1]: *** [obj/iso15693tools.o] Error 1
make: *** [client/all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/iceman1001/homebrew-proxmark3/issues

```

iceman1001, 2018-02-01:
try now.

n0ncetonic, 2018-02-02:
Compiles now, thanks :)

closed 2018-02-02 by n0ncetonic